vshymanskyy/TinyGSM #417: Socket error on client GsmClientTest, disconnecting

JJGR93, 2020-07-02:
[X ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[X ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[X ] Question or request for help

### What are you working with?

Modem: SIM7000E
Main processor board: MEGA
TinyGSM version: Latest
Code: MqttClient Example

### Scenario, steps to reproduce
Pubbing data in short intervals from Windows client.

### Expected result
Receiving data from Windows client to Arduino.

### Actual result
Actually, it works great, but if I pub messages in a short interval of time, the Serial output indicates:
"### Unhandled: : 0,0,"TCP","XX.XX.XX.XX","1883","CONNEC"
And then, it disconnect and reconnect to MQTT Broker.

SRGDamia1, 2020-07-07:
Depending on your baud rates, you may need to adjust ```TINY_GSM_YIELD_MS``` some to help keep messages from being chopped up.  If you have it unset or set to 0, try bumping up to 1 or 2 to see if that helps.  See https://github.com/vshymanskyy/TinyGSM/issues/409

SRGDamia1, 2020-08-07:
Did using the yield work for you?