vshymanskyy/TinyGSM #418: Httpsclient - Response status code: -2

fabioce, 2020-07-05:
[ ] I have read the Troubleshooting section of the ReadMe

[ ] Bug or issue with library functionality (ie, sending or receiving data over HTTPS)

### What are you working with?

Modem: SIM800
Main processor board: ESP32 TTGO T-Call
TinyGSM version:  latest (0.10.6)
Code: I'm trying to use your example httpsclient.ino

### Actual result
Response status code: -2

I don't have problem with HTTP, only to HTTPS

SRGDamia1, 2020-07-07:
That's a somewhat odd error to get.  From the HttpClient library -2 is an `[HTTP_ERROR_API](https://github.com/arduino-libraries/ArduinoHttpClient/blob/7f36561e0bced5458516066ae09636e119cae0ed/src/HttpClient.h#L16)` (This call was made when the HttpClient class wasn't expecting it to be called.  Usually indicates your code is using the class incorrectly.)  So something weird is going on.  Are you using the HttpsClient example exactly?

fabioce, 2020-07-08:
You are right, I lost to use http.get(resource);
but now in http it work well while in https it give me failed to connect.
I'm using your url https://vsh.pp.ua/TinyGSM/logo.txt

SRGDamia1, 2020-07-08:
Check which version of the firmware you have on the SIM800l (```getModemInfo()``` or ```ATI```).  From what I'm reading, different runs of the TTGO have different firmware versions.

https://github.com/vshymanskyy/TinyGSM/issues/29#issuecomment-328802556

If you have an old firmware, you'll have to figure out how to get the newer firmware files; as far as I know simcom doesn't post them.  Even with the newer firmware, I can't promise SSL will work.  Others have had issues with it:  https://github.com/vshymanskyy/TinyGSM/issues/378

fabioce, 2020-07-09:
Hi,
my getModemInfo() is here:
Modem: SIM800 R14.18

fabioce, 2020-07-09:
Hello I'm trying your code AT_Debug and it answer me :

[6034] Trying baud rate 115200 ...
[16044] Trying baud rate 57600 ...
[26054] Trying baud rate 38400 ...
[36064] Trying baud rate 19200 ...
[46074] Trying baud rate 9600 ...
[56084] Trying baud rate 74400 ...
[66094] Trying baud rate 74880 ...
[76104] Trying baud rate 14400 ...
[86114] Trying baud rate 28800 ...
***********************************************************
 Module does not respond!
   Check your Serial wiring
   Check the module is correctly powered and turned on
***********************************************************

SRGDamia1, 2020-07-09:
I'd set your baud rate instead of autobauding.  Default for the SIM800 is 9600.  Copy over whatever you needed to get communication working in the HTTPS sketch over to the AT_Debug.

fabioce, 2020-07-10:
Well now I set baud rate to 9600 and the serialmonitor sent me:
10:15:02.025 -> ***********************************************************
10:15:02.025 ->  You can now send AT commands
10:15:02.025 ->  Enter "AT" (without quotes), and you should see "OK"
10:15:02.025 ->  If it doesn't work, select "Both NL & CR" in Serial Monitor
10:15:02.025 -> ***********************************************************

But everythink I send ( AT etc) the serialmonitor don't answer nothing.

fabioce, 2020-07-10:
I changed AT_Debug code like this:

`// Select your modem:
#define TINY_GSM_MODEM_SIM800

// Set serial for debug console (to the Serial Monitor, speed 115200)
#define SerialMon Serial

#define SerialAT Serial2

#define TINY_GSM_DEBUG SerialMon

#include <TinyGsmClient.h>

void setup() {
  // Set console baud rate
  SerialMon.begin(115200);
  delay(2000);
}

void loop() {

  SerialAT.begin(115200, SERIAL_8N1, 26, 27);

  // Set up the GSM module
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);

  digitalWrite(4, LOW);
  digitalWrite(5, HIGH);

  delay(2000);

  // Access AT commands from Serial Monitor
  SerialMon.println(F("***********************************************************"));
  SerialMon.println(F(" You can now send AT commands"));
  SerialMon.println(F(" Enter \"AT\" (without quotes), and you should see \"OK\""));
  SerialMon.println(F(" If it doesn't work, select \"Both NL & CR\" in Serial Monitor"));
  SerialMon.println(F("***********************************************************"));

  while(true) {
    if (SerialAT.available()) {
      SerialMon.write(SerialAT.read());
    }
    if (SerialMon.available()) {
      SerialAT.write(SerialMon.read());
    }
    delay(0);
  }
}`

And now if I enter any command it give me one empty row.

fabioce, 2020-07-10:
Which AT cmd I can launch to understand if https works?

fabioce, 2020-07-10:
I configured the code Diagnostics with my settings and in HTTP it works fine, when I enable HTTPS the debug AT give me these:
11:12:54.923 -> Connecting to vsh.pp.uaAT+CIPCLOSE=0,1
11:12:54.923 -> 
11:12:54.923 -> +CME ERROR: operation not allowed
11:12:54.923 -> AT+CIPSSL=1
11:12:54.923 -> 
11:12:54.923 -> OK
11:12:54.923 -> AT+CIPSTART=0,"TCP","vsh.pp.ua",443
11:12:54.923 -> 
11:12:54.923 -> OK
11:13:10.201 -> 
11:13:10.201 -> 0, CLOSE OK
11:13:10.201 ->  [fail]

SRGDamia1, 2020-07-10:
You are having EXACTLY the same problem as #378 (same board, same problem) and I do not know the fix.  I do not know if @albertomercurio was ever able to resolve it.

Eiyvor, 2020-07-14:
I'm also having the same issue. I tried using TinyGsmClientSecure and also tried with AT commands, but no luck. I have successfully managed to save the SSL certificate on the module by using FS AT commands.

Mr-HaleYa, 2020-07-17:
I tried for weeks to get HTTPS to work on the sim800 but I never could I was wanting to connect to my website to post data, I was never able to get it to work so instead I create a rest API that uses HTTP and inserts the data into a database which then the HTTP website pulls the data from

HTTP dead work really well tho because I've been using it ever since