vshymanskyy/TinyGSM #419: Implement SIMCom SIM7070/SIM7080/SIM7090 Series

EnumC, 2020-07-13:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[x] Request to support a new module
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM7080G, SIM7070G, SIM7070E, SIM7070G-NG and SIM7090G
Main processor board: SAMD21
TinyGSM version: N/A
Code: N/A

### Notes

I glimpsed around the AT Command Manuals for SIM7000 and SIM7070/SIM7080/SIM7090. Many commands remained the same. However, SIMCOM changed(simplified?) the way TCP/UDP connections are handled. Commands related to HTTP(S) have also changed. Notably, GPRS, 3GPP, "Special SIMCom", FTP, NTP, and File System remains relatively untouched, with the exception of some additions to the command set. 
This could be an interesting one to implement since it is cross-compatible with what seems to be three different series of chips. 

Have anyone messed around with these yet? 

[SIM7080 Series_AT Command Manual_V1.02.pdf](https://github.com/vshymanskyy/TinyGSM/files/4909941/SIM7080.Series_AT.Command.Manual_V1.02.pdf)
[SIM7000 Series_AT Command Manual_V1.04.pdf](https://github.com/vshymanskyy/TinyGSM/files/4909965/SIM7000.Series_AT.Command.Manual_V1.04.pdf)

SRGDamia1, 2020-07-20:
I don't know why SIMCOM seems to like to come up with new sets of AT commands for each module, but it does seem to be their style.

I've seen these modules advertised and they look interesting, but I haven't picked one up.  If you want to submit a PR, I can look over it.  Or you can ship me a breakout of a North America or Global variant.

EnumC, 2020-07-30:
> I don't know why SIMCOM seems to like to come up with new sets of AT commands for each module, but it does seem to be their style.
>
> I've seen these modules advertised and they look interesting, but [...]

Now that you mentioned it, I should have bought two! The shipping for stuff from China is as slow as molasses with everything going on. What I can say so far from playing around with it is that sending a HTTP request of any kind is quite straightforward. However, I can't seem to figure out how to send a packet to a subdirectory of a domain (Ex: https://google.com/foobar) instead of https://google.com. I just get back "ERROR" by sending AT+SHCONN. Other than that, I've got GNSS ready to go and APN config. 

I'll be sure to make a PR once I experiment a bit more and get at least the basic functionalities implemented.

SRGDamia1, 2020-07-30:
I'm not sure exactly what order of commands you're trying, but you generally make the TCP or SSL connection to the domain (host) and then send a request to the directory or sub-directory.  Look at the [WebClient ](https://github.com/vshymanskyy/TinyGSM/blob/cddd356d0cca698ffcd324c0ca4702c1b92a2142/examples/WebClient/WebClient.ino#L199) example around lines 199-213.

Also, as a rule TinyGSM uses only the lower level TCP/SSL commands on all modules, not any application (HTTP(S)/MQTT/COAP/etc) level commands.

marmotton, 2020-08-14:
Hi @EnumC , I started working on the SIM7080 module here: https://github.com/marmotton/TinyGSM/blob/SIM7080/src/TinyGsmClientSIM7080.h

I didn't realize you already started working on it, maybe there will be something interesting for you in this file (work in progress and messy..).

I read that it is not possible to use TCP and GNSS at the same time (they share some RF components) :cry:, so not sure how much I'll continue working on it right now. https://stackoverflow.com/a/61884727/1497984

EnumC, 2020-08-14:
> Hi @EnumC , I started working on the SIM7080 module here: https://github.com/marmotton/TinyGSM/blob/SIM7080/src/TinyGsmClientSIM7080.h
>
> I didn't realize you already started working on it, maybe [...]

Oh, no! I completely overlooked that limitation. That's also going to be a limiting factor for our project too, so we might have to ditch it in favor of the SIM7000. 

If you would like to work on the port for this module, go full steam ahead. I'm experimenting with it, but I don't have an ETA for when the port will be completed. So far, I've got the library to connect to the tower via GPRS, as well as some other basic device info commands. Data is still non-operational, and LTE is also wip.

With that little twist about GNSS and TCP which is not documented anywhere, we're probably going to switch to a Quectel chip, if not the SIM7000.

schmeckm, 2020-10-22:
Hi Benjamin,

have you been able to complete your code to connect the SIM7080? 
Beside your modified Code "TinyGsmClientSIM7080.h" is there something else to add to the code?

Markus

marmotton, 2020-10-22:
Hi @schmeckm ,
I didn't make progress on this. The .h file you mention has some parts working but is not finished. I'm not planning to work on the SIM7080 soon, but hopefully I'll start working again on the SIM7000 SSL version which should be easy to port to SIM7080.

schmeckm, 2020-10-22:
Hi Benjamin,
sorry for the missunderstanding I use a SIM7600G Module from LillyGo, but still struggeling  to get the modem work. 

https://de.aliexpress.com/i/4001142716386.html

This company has posted some code based on the TinyGSM libary but it seems its doesn't work with 7600G

https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/blob/master/examples/SIM7600/SIM7600.ino

I posted therefore already an post in the TinyGSM :  https://github.com/vshymanskyy/TinyGSM/issues/448

I saw in the meanstime that only  SIMCom LTE Modules (SIM7100E, SIM7500E, SIM7500A, SIM7600C, SIM7600E) are supported but not the SIM7600G. Im not a AT Command Expert. 
I would just to consume the service in my coding. 

Markus

SRGDamia1, 2020-10-27:
I *thought* all the 7600 variants followed the same set of AT commands as the 7500 series.  But the 70x0 series definitely uses a different set, or maybe multiple different sets.  SIMCom seems to like to re-invent the TCP and SSL AT commands for every new module.

star297, 2021-03-25:
Before you waste too much time with the 7070 and 7080, 
you might want to take a look at this information below. 

https://stackoverflow.com/questions/61857667/sim7080g-module-cant-send-data-over-tcp-while-using-gnss/61884727#61884727

I got caught out and will stick with the 7000.

You could ditch the GPS functions and use a separate receiver if faster GPS updates are needed.

SRGDamia1, 2021-03-25:
Thanks for the heads up.  I may be doing some tests with a 7080 in the relatively near future and write out the implementation for it, but it's good to know the limitations before I dump too much into it.

Talexro, 2021-12-16:
Where can I download drivers ?  SIM7070 SIM7080 WIN7 WIN10 USB Driver V1.00.rar

star297, 2021-12-16:
Here is some useful information.
https://github-wiki-see.page/m/botletics/SIM7000-LTE-Shield/wiki/USB-Drivers

Win8 driver works on Windows 10.
Once the drivers are installed connect the module to USB AFTER you have a attached an LTE antenna!!
Assuming you are using the 'BK modules, ground the 'K' pin if the module does not power on by itself. 
The green LED should start blinking.
Use TeraTerm or similar to communicate.
You will see 6 available ports.
Try each one until it works.
You can't use LTE and GPS at the same time on the SIM7070.

rhr407, 2022-04-25:
Hello Guys!
I am working on SIM7070G development to replace the SIM7000 in our existing system. Has anyone tried to connect HTTP? In my case, the SHCONN command never succeeds. Any thoughts?

star297, 2022-04-25:
Connects to MQTT broker and works on pubsubclient for us on 2G and CAT-M1.
Check with your SIM supplier that it can support 2G , CAT-M1 or NB-IOT.
Not all countries and cell areas can support this.
If you don't want simultaneous GPS then the 7070 works much the same as the 7000.

rhr407, 2022-04-26:
Thank you so much. I contacted SIMCOM regarding this issue and they told me that it might be a timestamp issue of NTP or the firmware version. For the NTP they sent me the following settings: 
```

[Send->]AT+CGREG?
[Recv<-]
+CGREG: 0,1//First confirm that the module is registered online

OK

[Send->]AT+CCLK="19/04/10,09:27:31+32"//First set an arbitrary time for the module. The +32 time zone should be set correctly. The time zone in China is East 8*4=32 time zone.
[Recv<-]
OK

[Send->]AT+CNTP="ntp1.aliyun.com",32//Set the NTP server address, the time zone parameter is 32
[Recv<-]
OK

[Send->]AT+CNTP//Enable NTP synchronization
[Recv<-]
OK//This OK can be replied within 500 milliseconds, don’t judge this OK

[Recv<-]
+CNTP: 0//This is to be replied after the synchronization is completed. It is necessary to judge this 0, which means the synchronization is completed, and non-zero, which means the synchronization has failed. 
Because it is interacting with the network, if the signal is not good, the reply will be slow. The waiting timeout for this report Recommended not less than 40 seconds

[Send->]AT+CCLK?//After synchronization is complete, query CCLK, the time is correct 
[Recv<-]
+CCLK: "19/11/20,17:19:00+32"

OK

NTP server address
ntp1.aliyun.com
ntp2.aliyun.com
ntp3.aliyun.com
ntp4.aliyun.com
ntp5.aliyun.com
ntp6.aliyun.com
ntp7.aliyun.com
On Mon, Apr 25, 2022 at 5:18 PM Paul Staron <notifications@github.com> wrote:
Connects to MQTT broker and works on pubsubclient for us on 2G and CAT-M1.
Check with your SIM supplier that it can support 2G , CAT-M1 or NB-IOT.
Not all countries and cell areas can support this.
If you don't want simultaneous GPS then the 7070 works much the same as the 7000.—
Reply to this email directly, [view it on GitHub](https://github.com/vshymanskyy/TinyGSM/issues/419#issuecomment-1108778291), or [unsubscribe](https://github.com/notifications/unsubscribe-auth/AGGSH4OLT7ON4MO4PS7DP4LVG3ATPANCNFSM4OYBSBIQ).
You are receiving this because you commented. Message ID: <vshymanskyy/TinyGSM/issues/419/1108778291@github.com>
```

star297, 2022-04-26:
I haven't tried for a while but I used something like this, you will have to adjust the network and band settings to suit your location.
Initialise with modemSTART then call modemCONNECT when you need data connection or after disconnecting from data.

```
int modemSTART() {
  if (!modem.init()) {
    Serial.printf("Can't connect to Modem or sim not registered\n");
    return 0;
  } else {
    Serial.printf("Modem starting\n");
  }
  /*
    2 Automatic
    13 GSM only
    38 LTE only
    51 GSM and LTE only
  */
  int result = modem.setNetworkMode(38);
  Serial.printf("Set network MODE: %s\n", result ? "Ok" : "No");

  modem.sendAT(GF("+CMNB=3"));                    //  set auto NB-IOT / CAT-M
  if (modem.waitResponse(5000L)) {
    Serial.printf("Auto NB-IOT/CAT-M set\n");
  } else {
    Serial.printf("Auto NB-IOT/CAT-M fail!\n");
  }
  modem.sendAT(GF("+CBANDCFG=\"CAT-M\",20"));    //  select NB-IOT or CAT-M and set correct band for your network
  if (modem.waitResponse(5000L)) {
    Serial.printf("Band set\n");
  } else {
    Serial.printf("Band not set!\n");
  }
  if (!modem.waitForNetwork(60000)) {             //  wait upto 60 seconds to connect
    Serial.printf("Can't connect to network or sim not registered\n");
    return 0;
  } else {
    Serial.printf("Modem ready\n");
    return 1;
  }
}

int modemCONNECT() {
  if (modem.isGprsConnected()) {
    Serial.printf("gprs already connected\n");
    return 1;
  }
  if (!modem.waitForNetwork(20000)) {   // wait 20 seconds for network data connection
    Serial.printf("Can't connect to network or sim not registered\n");
    return 0;
  }
  if (!modem.gprsConnect(SIMapn, gprsUser, gprsPass)) {
    Serial.println("gprs connect fail");
    return 0;
  }
  Serial.printf("gprs connected\n");
  return 1;
}

```
I've never needed to manually set the modem time, once you have a successful modemSTART you can check the modem's time;
```
Serial.printf("Modem Time: %s\n", modem.getGSMDateTime(DATE_FULL).c_str());
```

Djacksonspears, 2022-09-04:
Did you figure out how to do a subdirectory im stuck on this as well 
@EnumC 
> Now that you mentioned it, I should have bought two! The shipping for stuff from China is as slow as molasses with everything going on. What I can say so far from playing around with it is that [...]

camp-easy, 2023-12-06:
Good evening, could somebody help me find SIM7080usb drivers?? Thank you in advance :)

kjm1102, 2023-12-28:
Anybody manage to upgrade the firmware in these? I've got a 7080G with 1951B07SIM7080 firmware that works as advertised but my second purchase with 1951B05SIM7080 firmware is useless.

Elitec-MrEli, 2024-01-19:
Boa Tarde, alguém poderia me ajudar a encontrar drivers SIM7080usb?? Agradeço antecipadamente :)

junaidaslam1, 2025-03-20:
> Hello Guys!
> I am working on SIM7070G development to replace the SIM7000 in our existing system. Has anyone tried to connect HTTP? In my case, the SHCONN command never succeeds. Any thoughts?

I have the exact same issue. Have you been able to solve the problem?

junaidaslam1, 2025-03-20:
> I eventually managed to get the sim7020 & (later the 7080) working by using https://www.waveshare.com/wiki/Pico-SIM7020E-NB-IoT as a starting point.

When i look at the SIM7070 AT command manual, i don't find CHTTPxx commands. Do you know a link where i can get a manual with full command set for http? What did you specifically do to fix AT+SHCONN ERROR?

kjm1102, 2025-03-21:
The trouble with simcoms is that each model is a whole new world of pain
when it comes to http AT cmds, chttp is for the 7020.  The 7070 & 7080 cmds
seem more similar. Maybe try
https://www.waveshare.com/wiki/Pico-SIM7080G-Cat-M/NB-IoTv or
https://m2msupport.net/m2msupport/http-testing-with-simcom-sim7070-sim7080-modules/

kjm1102, 2025-03-21:
first link is dead, try
https://www.waveshare.com/wiki/Pico-SIM7080G-Cat-M/NB-IoT

Djacksonspears, 2025-03-21:
I used hologram and it was a lot easier. Check out the repo on my profile it was easy doing it that way.

junaidaslam1, 2025-03-21:
> I used hologram and it was a lot easier. Check out the repo on my profile it was easy doing it that way. 

Actually i need to use AT Commands because I am integrating SIM7070 in abare metal platform.

trevorbattle, 2025-03-26:
I've been happily using TinyGSM with the "and-global" breakout modules for months. They are available on Aliexpress, the manufacturer web page is:

https://www.and-global.com/sim7080g-lte-cat-m-nb-iot-module-sim7080-breakout-board-with-gps-and-4g-antenna.html

I've been able to use SSL as well, both by using SSL from the card as well as using the mobizt/ESP_SSLClient wrapper (which is big but  for ESP32 is manageable).

Note that all my modules were purchased in the last year though, so I may have always had newish firmware that works.

Possible the wrapper might work with older firmware.

SRGDamia1, 2025-03-26:
@trevorbattle Before upgrading, I think my modules were 1951B03SIM7080. I was content on that firmware until I wanted to use custom certificates and connect to AWS IoT core.