vshymanskyy/TinyGSM #420: u-blox LTE-M/NB-IoT Modems (SARA-R4xx, SARA-N4xx, but NOT SARA-N2xx)

tcpipchip, 2020-07-16:
One question

To use this lib to access LTE-M/NB-IoT using SARA-R4/SARA-N4 needs some special initialization before to have access to the network ?

Some different AT COMMAND that those used to LISA ?

For example, LISA didnt need any special initialization! Works fine fist time!

Thank you!!!

Mr-HaleYa, 2020-07-17:
mythical issue #420 🙏 🌿 🤘

tcpipchip, 2020-07-20:
Hi, @SRGDamia1 

Well, it wont work until you do some first initializations before to use the TinyGSM lib, my opnion.

Do TinyGSM does not implement this sequence ?

at+cfun=4

4: disables both transmit and receive RF circuits by deactivating both CS and PS services and sets the MT into airplane mode

at+umnoprof=100

Mobile Network Operator (MNO) configuration

at+cfun=15

To reset the module

at

OK

at+cedrxs=0

use of eDRX disabled

at+cedrxs=3,2

disable the use of eDRX and reset all parameters for eDRX to factoryprogrammed values 

at+cedrxs=3,4

4: E-UTRAN (WB-S1 mode) 

at+cedrxs=3,5

5: E-UTRAN (NB-S1 mode) 

at+cpsms=0

 disable the use of PSM

at+upsv=0

0 (default and factory-programmed value): power saving disabled

at+urat=7

Indicates the radio access technology and may be:

 LTE CAT-M1

at+ubandmask=0,134217732

VIVO ?

at+ubandmask=1,134217732

VIVO ?

at+cfun=15

To reset the module

at+cfun=4

4: disables both transmit and receive RF circuits by deactivating both CS and PS services and sets the MT into airplane mode

at+cgdcont=1,"IPV4V6","smart.m2m.vivo.com.br"  <- se você recebeu um APN diferente, altere a seção destacada para o valor correto

Defines the connection parameters for a PDP context

at+cgdcont?

Check current APN

at+umnoprof?

Check selected MNO Profile

 

at+cedrxs?

The following command example reads the requested eDRX value

 

at+cfun=1

Turn on the radio

 

at+cmee=2

SRGDamia1, 2020-07-20:
In general, this library does not attempt to customize anything for a particular network operator or SIM card provider for any of the modules.  That is completely left up to the user.  If your operator requires you to manually set bands and disable PSM or EDRx, that is annoying, but that is not a universal problem.

I can't say I've ever used the PSM/EDRx commands and see no particular reason they'd be necessary unless it's something your SIM provider specifies.

Depending on your SIM card, you may not need to specify the operator profile.   The "SW default" (0) or "SIM ICCID Select" (1) profiles might be just fine.  I believe the commands for setting the band mask are only necessary if you are using the "Standard Europe" (100) operator profile.  We use AT&T for the Hologram network in the USA and don't set the band.  For while we didn't bother to set the operator at all, but we found it connected _faster_ when manually selected.

I also would use ```AT+CFUN=0``` rather than 4 to turn off the radios between switching bands or operators.  The airplane mode set with ```AT+CFUN=4``` will persist between resets, 0 will not.  (That is, if you set to 0, then reset, the module will return to full functionality (1) after reset.  If you set to 4 and then reset, you have to remember to manually set back to 1 afterwards.)

tcpipchip, 2020-07-20:
Thank so much the info @SRGDamia1 

In wich point of source do you suggest to make a call to a specialized AT commands before the make the final attach ?

SRGDamia1, 2020-07-20:
If you need to set the bands, I'd do it right at the top of you program.  If you're following the flow of most of the examples, you'd put it right after the ```modem.restart();``` or ```modem.init()```.

You can use the ```sendAT``` and ```waitResponse``` library functions to send your custom commands, ie:

```cpp
modem.sendAT(GF("+ubandmask=0,134217732"));
modem.waitResponse();
```

tcpipchip, 2020-07-22:
@SRGDamia1 

Thank you so much! You help me and will help a lot of other members!