vshymanskyy/TinyGSM #421: Sim7000G never connects

Mr-HaleYa, 2020-07-21:
[x ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[x ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x ] Question or request for help

### What are you working with?

Modem: SIM7000G R1529 Revision:1529B03SIM7000G
Main processor board: ESP32
TinyGSM version: 0.10.5
Code: 
```cpp
/*
  FILE: AllFunctions.ino
  AUTHOR: Koby Hale
  PURPOSE: Test functionality
*/

#define TINY_GSM_MODEM_SIM7000
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
#define SerialAT Serial1

// See all AT commands, if wanted  
// #define DUMP_AT_COMMANDS 

/*
   Tests enabled
*/
#define TINY_GSM_TEST_GPRS    true
#define TINY_GSM_TEST_GPS     true
#define TINY_GSM_POWERDOWN    true

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials, if any
const char apn[]  = "hologram";     //SET TO YOUR APN
const char gprsUser[] = "";
const char gprsPass[] = "";

#include <TinyGsmClient.h>
#include <SPI.h>
#include <SD.h>
#include <Ticker.h>

#ifdef DUMP_AT_COMMANDS  
#include <StreamDebugger.h> 
StreamDebugger debugger(SerialAT, Serial); 
TinyGsm modem(debugger);  
#else 
TinyGsm modem(SerialAT);
#endif

#define uS_TO_S_FACTOR 1000000ULL  // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  60          // Time ESP32 will go to sleep (in seconds)

#define UART_BAUD   9600
#define PIN_DTR     25
#define PIN_TX      27
#define PIN_RX      26
#define PWR_PIN     4

#define SD_MISO     2
#define SD_MOSI     15
#define SD_SCLK     14
#define SD_CS       13
#define LED_PIN     12

void setup() {
  // Set console baud rate
  Serial.begin(9600);
  delay(10);

  // Set LED OFF
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  pinMode(PWR_PIN, OUTPUT);
  digitalWrite(PWR_PIN, HIGH);
  delay(300);
  digitalWrite(PWR_PIN, LOW);

  SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  if (!SD.begin(SD_CS)) {
    Serial.println("SDCard MOUNT FAIL");
  } else {
    uint32_t cardSize = SD.cardSize() / (1024 * 1024);
    String str = "SDCard Size: " + String(cardSize) + "MB";
    Serial.println(str);
  }

  Serial.println("\nWait...");

  delay(5000);

  SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  Serial.println("Initializing modem...");
  if (!modem.restart()) {
    Serial.println("Failed to restart modem, attempting to continue without restarting");
  }

  String name = modem.getModemName();
  Serial.println("Modem Name: " + name);

  String modemInfo = modem.getModemInfo();
  Serial.println("Modem Info: " + modemInfo);

#if TINY_GSM_TEST_GPRS
  // Unlock your SIM card with a PIN if needed
  if ( GSM_PIN && modem.getSimStatus() != 3 ) {
    modem.simUnlock(GSM_PIN);
  }
#endif

  /*
    2 Automatic
    13 GSM only
    38 LTE only
    51 GSM and LTE only
  * * * */
  String res;
  do {
    res = modem.setNetworkMode(2);
    delay(500);
  } while (res != "OK");

  /*
    1 CAT-M
    2 NB-Iot
    3 CAT-M and NB-IoT
  * * */
  do {
    res = modem.setPreferredMode(3);
    delay(500);
  } while (res != "OK");

}

void loop() {

  Serial.println("Waiting for network...");
  if (!modem.waitForNetwork()) {
    delay(10000);
    return;
  }

  if (modem.isNetworkConnected()) {
    Serial.println("Network connected");
  }

#if TINY_GSM_TEST_GPRS
  Serial.println("\n---Starting GPRS TEST---\n");
  Serial.println("Connecting to: " + String(apn));
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    delay(10000);
    return;
  }

  Serial.print("GPRS status: ");
  if (modem.isGprsConnected()) {
    Serial.println("connected");
  } else {
    Serial.println("not connected");
  }

  String ccid = modem.getSimCCID();
  Serial.println("CCID: " + ccid);

  String imei = modem.getIMEI();
  Serial.println("IMEI: " + imei);

  String cop = modem.getOperator();
  Serial.println("Operator: " + cop);

  IPAddress local = modem.localIP();
  Serial.println("Local IP: " + String(local));

  int csq = modem.getSignalQuality();
  Serial.println("Signal quality: " + String(csq));

  delay(2000);
  int i = 10;
  while (i) {
    SerialAT.println("AT+CPSI?");     //Get connection type and band
    delay(500);
    if (SerialAT.available()) {
      String r = SerialAT.readString();
      Serial.println(r);
      if ( r.indexOf("OK") >= 0 ) break;;
    }
    delay(500);
    i--;
  }
  Serial.println("\n---End of GPRS TEST---\n");
#endif

#if TINY_GSM_TEST_GPRS
  modem.gprsDisconnect();
  if (!modem.isGprsConnected()) {
    Serial.println("GPRS disconnected");
  } else {
    Serial.println("GPRS disconnect: Failed.");
  }
#endif

#if TINY_GSM_TEST_GPS
  Serial.println("\n---Starting GPS TEST---\n");
  // Set SIM7000G GPIO4 HIGH ,Open GPS power
  // CMD:AT+SGPIO=0,4,1,1
  // Only in version 20200415 is there a function to control GPS power
  modem.sendAT("+SGPIO=0,4,1,1");

  modem.enableGPS();
  float lat,  lon;
  while (1) {
    if (modem.getGPS(&lat, &lon)) {
      Serial.printf("lat:%f lon:%f\n", lat, lon);
      break;
    } else {
      Serial.print("getGPS ");
      Serial.println(millis());
    }
    delay(2000);
  }
  modem.disableGPS();

  // Set SIM7000G GPIO4 HIGH ,Close GPS power
  // CMD:AT+SGPIO=0,4,1,0
  // Only in version 20200415 is there a function to control GPS power
  modem.sendAT("+SGPIO=0,4,1,0");
  Serial.println("\n---End of GPRS TEST---\n");
#endif

#if TINY_GSM_POWERDOWN
  // Try to power-off (modem may decide to restart automatically)
  // To turn off modem completely, please use Reset/Enable pins
  modem.poweroff();
  Serial.println("Poweroff.");
#endif

  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  delay(200);
  esp_deep_sleep_start();

  // Do nothing forevermore
  while (true) {
    modem.maintain();
  }
}
```

### Scenario, steps to reproduce
Buy a TTGO SIM7000G and run the sketch with a Hologram SIM

### Expected result
I'm using a TTGO SIM7000G and I CANOT get it to connect to anything. I use to use 2G with Twilio but we changed Sim carriers to Hologram. It was working on the 2G Hologram network then I tried connecting to Cat-M1 and It took hours to get that to work but it finally connected to AT&T Band12 Cat-M1. It was perfect... then I literally did nothing but push the reboot button on the board and it has never connected to ANY network again... I can do COPS=? and see the networks.

```
(1,"313 100","313 100","313100",7),(1,"AT&T","AT&T","310410",7),(3,"T-Mobile","T-Mobile","310260",0),(1,"Verizon","Verizon","311480",7),(3,"T-Mobile","T-Mobile","310260",9),(1,"Verizon","Verizon","311480",9),(1,"AT&T","AT&T","310410",9),,(0,1,2,3,4),(0,1,2)
```

COPS? is set to 0
setNetworkMode(2); Ive tried 2-13-38-51 and no diffence
setPreferredMode(3); Ive tried 1-2-3 and nothing

I even Ran the reset code that is in the library and nothing.

### Actual result
It never connects to anything...

### Debug and AT command log
```
Initializing modem...
AT+CFUN=0

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
⸮⸮AT
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CPIN: READY

OK
AT+GMM

SIMCOM_SIM7000G

OK
Modem Name: SIMCOM SIM7000G
ATI

SIM7000G R1529

OK
Modem Info: SIM7000G R1529
AT+CPIN?

+CPIN: READY

OK
AT+CNMP=2

OK
AT+CMNB=3

OK
Waiting for network...
AT+CEREG?

+CEREG: 0,0

OK
**Around 100 of these lines ^^^ then changes to this**

AT+CEREG?

+CEREG: 0,3

OK
**Does this forever more ^^^**

```

SRGDamia1, 2020-07-21:
I've gotten the SIM7000A to work on AT&T/Hologram in the USA but I haven't tried the G version.  I think the only difference is the band optimization so it *should* work.  But, obviously, it doesn't.  Ugh.

We've had issues a number of times with Hologram where (despite everything Hologram says) a SIM that was previously fully functional on the 2G network, won't work with LTE.  Sometimes we've been able to contact Hologram support and they've done something on their end to make it work and sometimes we've given up and switched to a totally virgin SIM.  Most of what we've done has been with the SARA R4, but we saw the same symptoms - registration was denied (CREG: 0,3).

I doubt it's an issue with your preferred mode/band.  "Automatic" and "CAT-M and NB-IoT" should be fine. Last I checked Hologram in the USA still wasn't doing any NB IoT, but as long as you didn't select NB IoT only that shouldn't matter.  

You can try setting COPS to manual and trying to force connection to AT&T (```AT+COPS=1,2,310410```) but I'd guess you'll still be denied and wouldn't recommend you manually set COPS unless it's the only thing that works.  

You could also try making sure your mobile operator configuration is explicitly set to AT&T (ie, not Verizon) by setting to manual, to AT&T and then rebooting.  (```AT+CMCFG=0``` then ```AT+CMCFG=2,"ATT"``` and ```AT+CFUN=1,1```)  Hologram, despite what they sometimes say, doesn't usually work on Verizon.  This might be necessary on the "G" version - I never set it on the "A" one I have.  I'm not particularly confident, though.

Only other thing I can think of is generic advice like check your antenna and power supply.  Sometimes the operator will deny registration if you have a weak or inconsistent signal.

Mr-HaleYa, 2020-07-22:
@SRGDamia1 
Thank you for your reply, I also had concerns about how good my antenna is so I have ordered a large gain antenna that will be arriving tomorrow. 

You were saying that you have had problems with hologram. Are you located in the USA as I am? If so all I need to be able to do is connect to something **other** than 2G and 3-g to send data to my server. What carrier do you use? what's their website? how reliable are they and what connections standards? I would really prefer to use cat M1 just because it sounds cool and apparently has some power savings (that I still have not seen actually do anything, the battery lasts 7 days on catm and 7 Days on 2G...). It also says to save bandwidth but when I check my data package size that was used during transmission it is the same over 2G or can't so I really don't care if it's specifically Catm or just LTE. 

I don't know if you follow the news closely with T-Mobile but they put out a really big alert saying they are shutting down all to 2-3G very soon and our old service provider Twilio is shutting down at the end of the month and then 3-g will be shut down at the end of the year, that's why I am scrambling to get anything else to work because all of our clients that are connected right now are having very bad connection issues on the 2G because they are starting to turn off the towers.

SRGDamia1, 2020-07-22:
I'm in the USA (Pennsylvania/Philly area) and we primarily use Hologram as our carrier.  They basically use AT&T for LTE/LTE-M and T-Mobile for 2/3G.  The don't do LTE NB IoT.  Definitely try contacting their support about you SIM issues.  Use their email contact, not the forum.

I'm very aware of the 2G shutdown.  While they've said the whole time that service will continue at full strength up to the actual shutdown date, we've been seeing the 2G coverage area dwindling for months.  I really wish 2G would continue - the SIM800 we used for 2G mostly just worked and LTE has been so much hassle.  We've been deploying with Digi XBee3's (u-blox SARA R410M) which are certified, easy to get, and not too expensive, but they're more expensive and less reliable than the Sodaq 2GBee's (SIM800H) that they are replacing.

There's definitely real power saving with LTE-M, but it can take some work.  Most of the advertised LTE power savings is based on PSM and eDRx.  In our applications, we haven't actually played with those (we cut power) but we have seen some power savings in places where the 2G signal was marginal and the LTE is strong making the connection time lower.

We haven't seen any data/bandwidth savings at all.  If anything, the data use is *higher* on LTE-M.

SRGDamia1, 2020-07-22:
Something else I thought of you could try.

Usually the connection process is to search for the cellular network, register on it, supply the network with your APN for the data (EPS/GPRS) network, and then make a data connection.  You could try making sure the APN is set before even searching for the network.  Just run ```modem.gprsConnect(apn, gprsUser, gprsPass)``` without waiting for the network.  It will fail if you're not registered, but it should get the APN saved to NVM.

Mr-HaleYa, 2020-07-23:
> I really wish 2G would continue - the SIM800 we used for 2G mostly just worked and LTE has been so much hassle.

ha yep I have 5 sim800land they were perfect.... so sad to see it go 😭 

> Just run `modem.gprsConnect(apn, gprsUser, gprsPass)` without waiting for the network. It will fail if you're not registered, but it should get the APN saved to NVM.

so go 
```cpp
  modemPowerOn();
  modem.gprsConnect(apn, gprsUser, gprsPass)
  modem.setPreferredMode(Net_Mode);
  modem.setPreferredMode(LTE_Mode); 

  if (!modem.isNetworkConnected()) {
    if (!modem.waitForNetwork(45000L)) {          // try for 45 sec to find network before failing
      Serial.println("Failed to connect to network Rebooting ESP");
      modem_reset();
      reboot();
    }
  }
  gprsStatus = modem.isGprsConnected();
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    Serial.println("Fail");
  }
```

basically
turn on
set apn
set mode
connect to a network
check if apn is connected and if not try again

___

when i try and set `AT+CMCFG=2,"ATT"`  i get this far
```
AT+CMCFG=?

+CMCFG: (0-4),40

OK
AT+CMCFG?

+CMCFG: 0,0

OK
AT+CMCFG=0

OK
AT+CMCFG=2,"ATT"

ERROR
```

Any suggestions???

Mr-HaleYa, 2020-07-23:
I enabled CMEE=2 so I get error codes/descriptions instead of just ERROR ( should really be added to AT debugging!!! )

when I do `AT+CMCFG=2,"ATT"` I get  `+CME ERROR: operation not allowed`
when I try to set cops to `AT+COPS=1,2,310410` I get `+CME ERROR: no network service`

SRGDamia1, 2020-07-23:
That's exactly what I meant for setting the APN before and after.  

Honestly, I don't know if there is a way to completely reset the thing.  The AT&F command is the only one supplied, and I don't think it works.  I'd guess re-flashing its firmware might do it.  The Botletics wiki has instructions here:  https://github.com/botletics/SIM7000-LTE-Shield/wiki/Updating-Firmware  There's a fair amount of good info on the SIM7000 there.   You might want to try updating or re-flashing anyway; you're not likely to make an already-not-cooperative module worse.

A CSQ of 19 should be more than enough to work with.  That's pretty good signal.

Have you tried manually selecting the bands?  ```AT+CBANDCFG="CAT-M",2,4,12,13```  I'm pretty much grasping at straws with that.

Also, ```CMEE=2``` should be set if you ```#define TINY_GSM_DEBUG SerialMon``` https://github.com/vshymanskyy/TinyGSM/blob/cd002c5adfca4d8440c10c141f79dec9147eec87/src/TinyGsmClientSIM7000.h#L165

Mr-HaleYa, 2020-07-24:
I don't think flashing the firmware will help because its ALL of my devices that have the same problem... and I have like 10 (I've only been testing a lot on 3) and none will connect.

I finally got mine into +CEREG: 0,2 so it's searching but still no connection. It just searches for hours, I am on the top floor with amazing cell service on all carriers (got an AT&T, Verison, and T-Mobile phone to check quality)

I don't explicitly need the sim7000g, all I need i a reliable connection to which i can upload data to and in not dying in the next year (2-3G)... What other modems are there and are they even any good ones worth using?

Mr-HaleYa, 2020-07-26:
Any other suggestions??? Or modem that work better? I really need this to work as we have requests for devices from a lot of clients but I cannot get the hardware to cooperate...

If you have no other ideas, I'm planning to build my own PCB with the modem baked in so if you know of a "reliable" one please let me know so I can do testing

Mr-HaleYa, 2020-07-26:
uhhhhhhhhhhhh I ran like 50 commands and now it works 😶😶😶

says connected to `+CPSI: LTE CAT-M1,Online,310-410,0x9709,111159055,393,EUTRAN-BAND12,5110,3,3,-17,-103,-69,8`

Im going to try on my 3 others that won't connect.....

Mr-HaleYa, 2020-07-26:
oh my gosh now 2 are working.... can I get three????

Mr-HaleYa, 2020-07-26:
> Honestly, I don't know if there is a way to completely reset the thing. The AT&F command is the only one supplied, and I don't think it works.

Ya, I read the docs and it says in V1.02 they `Delete ATZ,AT&F,AT&V` and those where the reset commands...

page 12
[SIM7000 Series_AT Command Manual_V1.05.pdf](https://github.com/vshymanskyy/TinyGSM/files/4978192/SIM7000.Series_AT.Command.Manual_V1.05.pdf)

Mr-HaleYa, 2020-07-26:
Holy freaking cow... I did it. I got all 3 restored into working order and connected to catm1.

Mr-HaleYa, 2020-07-26:
@SRGDamia1 @lewisxhe 
turns out it WAS a tinyGSM problem! well not intentionally or explicitly there fault BUT they should have the capabilities to fix this.

I don't want to get too excited with this but I may have hit a gold mine... I was literally running every command in the PDF one at a time to try and figure wtf was so wrong. as I made my way through I stumbled upon the command that reads the Defined PDP Context. this is set by what APN you _FIRST_ put in the sketch. 

(Background: I used Twilio then switched to Hologram)
The command is `CGDCONT `and when I first ran `AT+CGDCONT?` I got
```
+CGDCONT: 1,"IP","wireless.twilio.com","0.0.0.0",0,0,0,0
+CGDCONT: 13,"IP","wireless.twilio.com","0.0.0.0",0,0,0,0
+CGDCONT: 14,"IP","wireless.twilio.com","0.0.0.0",0,0,0,0
```

I instantly realized something was wrong as I looked down and saw I had a Hologram sim in NOT Twilio, I then though oops I must have set it wrong in the code NOPE it was set to `hologram`. That got me thinking and I remembered that this particular device NEVER connected to CatM. so I decided to change it to hologram APN instead of the Twilio APN with

```
AT+CGDCONT=1,"IP","hologram","0.0.0.0",0,0,0,0
AT+CGDCONT=13,"IP","hologram","0.0.0.0",0,0,0,0
AT+CGDCONT=14,"IP","hologram","0.0.0.0",0,0,0,0 
```

After doing so I checked the configured bands for CatM and NBIOT with `AT+CBANDCFG?` I was replied to with the default bands of 
```
+CBANDCFG: "CAT-M",1,2,3,5,8,12,13,17,18,19,20,26,28
+CBANDCFG: "NB-IOT",1,2,3,5,8,12,13,17,18,19,20,26,28
``` 
I then remembered that CATM can only run on select bands in my country (USA) so I changed the CATM to those with `
AT+CBANDCFG=CAT-M,2,3,4,12,13`

I then set these just to make sure it was searching
```
AT+CREG=2
AT+CGREG=2
AT+CEREG=2
```

after doing this I ran the `AT&W` command twice to save everything then ran the `AT+CPSI?` command and BOOM
 `+CPSI: LTE CAT-M1,Online,310-410,0x9709,111159055,393,EUTRAN-BAND12,5110,3,3,-13,-101,-72,12`
 I WAS CONNECTED!!!

I think the rush of excitement triggered a mini-stroke because I stared at that one like for about 10 minutes...

After recovering I quickly uploaded the AllFunctions sketch to ensure I wasn't hallucinating and sure enough, it connected to the network within 5 seconds of searching. With this, I ran out of the room to search for the other boards that I was struggling to get connected to CatM and found 2. I plugged in the first and started running commands and this time I got 
```
+CGDCONT: 1,"IP","YOUR-APN","0.0.0.0",0,0,0,0
+CGDCONT: 13,"IP","wireless.twilio.com","0.0.0.0",0,0,0,0
+CGDCONT: 14,"IP","hologram","0.0.0.0",0,0,0,0
```

I was perplexed by this. Then I remembered that this device was the only one I ever got to connect to the CatM network, before it use to run Twilio 2g. It was also one I had accidentally uploaded the test sketch to without setting an APN. 
So changed all of them to hologram just like the first then saved and _**magic**_ It worked and I was connected to CATM.

So same procedure with my third and just like that all 3 where fixed...

So, I believe that these problems are happening because the sim7000 is filling the PDP with whatever APN you have set then if you change APNs it doesn't change the PDP. if you go and manually do so it fixes the problem. I really hope that this was the reason that it was broken and with it solving the problem 3 times I think that the evidence shows it was. 

___

@SRGDamia1  Can you add a method that clears then sets the new PDP APN automatically so it doesn't happen to anyone else. I think doing so would solve this issue.

SRGDamia1, 2020-07-29:
I'm really glad you got it working! I'm sorry I haven't gotten to making changes for TinyGSM.  I'll try to look into it by the end of the week.

Mr-HaleYa, 2020-07-29:
Just to make sure this was the problem I set 1,13,14 to Hologram and tested it with the hologram sim and instant connection. I then changed my APN in the code to the Twilio APN and popped in a Twilio sim and no connection even 5 hours later. changed 1,13,14 to the Twilio APN and instant connection.

I think this is the best thing I have found while working with the Sim7000...

Mr-HaleYa, 2020-07-29:
after some more testing
when I change my APN in the code it only changes 1,13 while 14 remains the old APN and my device struggles to connect. as soon as 14 is manually changed the modem connects to the network quickly and stable. I think a new function is needed in the sim7000 file that when APN is set it changes the 1,12,13 to match the APN. I do not know if 1,13,14 are the only ones it uses (doc says 1-24 but all mine are 1,13,14) so it should detect the numbers then change those. I could try writing this function if you would like.

Mr-HaleYa, 2020-07-31:
I believe I have fixed this so I will close now. I will reopen it if I have more issues. Thanks all for your help

closed 2020-07-31 by Mr-HaleYa

ClemensGruber, 2024-01-09:
Seems this is 01/2024 still not fixed in the current version 0.11.7 of the lib! I'm working with a SIM7080 and run in the same issue Koby / @Mr-HaleYa reported: 

- I uploaded the TinyGSM WebClient example, accidently with the default APN "YourAPN" and did not think of any not-reversible effects
- after noticing the wrong APN I changed "YourAPN" to "iot.1nce.net", the right one of my telecom provider and uploaded the modified code
- but nothing happende only `Waiting for network...` was the output, no other, more helpfull debug information

So I searched here in the issues but it was a bit hard to find this information, also because this issue is closed, so it seemed fixed but it is not. Please reopen this issue, so the information would be more visible! 

So I tried `AT+CGDCONT?` and got

```
+CGDCONT: 1,"IP","YourAPN","0.0.0.0",0,0,0
+CGDCONT: 2,"IPV4V6","ims","::",0,0,0
```

As reported above the wrong, "old" APN, even I had set the right one in the sketch via `const char apn[] = "iot.1nce.net";`

I found in https://github.com/vshymanskyy/TinyGSM/issues/592#issuecomment-1596596603 the hint to execute `AT+CGDCONT=1,"IPV4V6","" ` and after this "cleaning" the once more uploaded code did work! 

Sara / @SRGDamia1 or whoever maintains the code, is it possible to have a fix for this buggy behavior? 

I was very glad that I found the posting of Koby / @Mr-HaleYa, he wrote: 
> I think this is the best thing I have found while working with the Sim7000...

So true, I wouldn*t come to any solution without your posting, many thanks!

brettbeeson, 2024-01-12:
Hi! 

I had the same problem with a SIM7670SA running via a [lilygo32 ](https://www.lilygo.cc/products/t-sim-a7670e) board. I post here to assist in others in this situation. (I'm Australia using Telstra.)

Since the lilygo uses a FORK of TinyGSM, it assumedly has the 'old' version of TinyGSM.

Here is an annotated log of the problem and solution, using the [AT_DEBUG](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/examples/ATdebug/ATdebug.ino)  to manually run AT commands.

Short version: set the APN 'manually' with `AT+CGDCONT=1,"IP","telstra.internet","",0,0,,,,`

```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
Start Sketch
Trying baud rate 115200
Modem responded at rate:115200
***********************************************************
 You can now send AT commands
 Enter "AT" (without quotes), and you should see "OK"
 If it doesn't work, select "Both NL & CR" in Serial Monitor
 DISCLAIMER: Entering AT commands without knowing what they do
 can have undesired consiquinces...
***********************************************************

AT
OK
AT+CSQ
# Signal rssi=19
+CSQ: 19,99
OK

PB DONE
AT+CGREG?
# 0=no network?, 11=emergency only
+CGREG: 0,11

OK

+CGEV: MDE ETACH

# fails even after a long time
AT+CGREG?
+CGREG: 0,0

# correct
OK
AT+CFUN?
+CFUN: 1

# change to LTE only (not 2=auto) and reset
AT+CNMP=38
OK

# reset and try
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
AT+CSQ
+CSQ: 21,99

OK
AT+CGREG?
# With LTE (set via AT+CNMP=38), connection is now "forbidden" not 11: "emergency only"
+CGREG: 0,3

OK
AT+CSQ
+CSQ: 21,99

# current operator: 0=automatic, 2=numeric, 50501=telstra,7=EUTRAN (???)
# ok
OK
AT+COPS?
+COPS: 0,2,"50501",7

OK
AT+COPS=?
+COPS: (1,"Telstra Mobile","Telstra","50501",7),(3,"Optus AU","Optus","50502",7),(3,"vodafone AU","voda AU","50503",7),,(0,1,2,3,4),(0,1,2)

# Wait a few minutes
# Network changes from 3 (denied) to 11 (emergency)
AT+CGREG?
+CGREG: 0,11

OK

AT+CSQ
+CSQ: 21,99

# Check bands  <mode>[,<lte_mode>][,<lte_modeExt>]
+CNBP?
+CNBP: 0X0000000000280180,0X00000000080000DF,0X0002
# check bitwise operators on datasheet
# LTE BANDS ARE: 1,2,3,4,5,7,8,28 as per datasheet on lilygo and
# 0X0002 means band66 as per datasheet
# so: as expected

AT+CGDCONT?
+CGDCONT: 1,"IP","Internet","",0,0,,,,

OK
# **this is incorrect (should be telstra.internet!)**

# set to APN correctly
AT+CGDCONT=1,"IP","telstra.internet","",0,0,,,,

# Reset and it connects!
AT+CRESET
OK
*ATREADY: 1
+CPIN: READY
SMS DONE
+CGEV: EPS PDN ACT 1
+CGEV: ME PDN DEACT 8
PB DONE
#  "1" means registered
+CGREG: 0,1

luizalbertotoledo, 2024-04-15:
Hello.
Thanks for this help. I resume the neeeded comandos here to help. I use the SIM7000G in Brasil.

I USED THE "ATdebug" LIBRARY TO MAKE THESE SETTINGS AND TEST THE MODEM CONNECTION
[](url)

CHECK THAT THERE IS A CONNECTION TO SEND AT COMMANDS
AT
IT SEEMS THAT WHEN WE BUY A NEW SIM7000G IT IS MISSING SOME IMPORTANT SETTINGS FOR 4G CONNECTION, WHICH MUST BE INSERTED ON THE FIRST USE. AFTER INSERTED THEY ARE SAVED.

CHECK CONNECTION STATUS
AT+CPSI?

SETA COMMUNICATION ON CAT-M AND NB-IOT CHANNELS IN BRAZIL
CHECK THE CHANNELS YOUR COUNTRY USES AND THE APN OF YOUR MOBILE CARRIER

AT+CGDCONT=1,"IPV4V6","timbrasil.br","0.0.0.0",0,0,0,0
AT+CGDCONT=8,"IPV4V6","timbrasil.br","0.0.0.0",0,0,0,0
AT+CGDCONT=13,"IPV4V6","timbrasil.br","0.0.0.0",0,0,0,0
AT+CGDCONT=14,"IPV4V6","timbrasil.br","0.0.0.0",0,0,0,0

AT THIS MOMENT, THE AT+CPSI COMMAND MUST RETURN SOME MODEM CONNECTION WITH THE OPERATOR
AT+CPSI?
EXAMPLE ANSWER
+CPSI: GSM,Online,724-03,0x0204........

COMMAND TO CHECK THE OPERATOR’S IP ADDRESS
AT+CNACT?
IF YOU RETURN TO THE PHRASE BELOW, YOU MUST BE NECESSARY TO START THE MODEM NETWORK WITH THE OPERATOR
+CNACT: 0,"0.0.0.0"

STARTS THE NETWORK CONNECTION WITH THE OPERATOR
AT+CNACT=1
IF YOU SEE THE ANSWER BELOW, IT WAS RIGHT
+PDP APP: ACTIVE

WHEN ENTERING THE COMMAND AGAIN, THE RESPONSE MUST RETURN AN IP ADDRESS
AT+CNACT?
+CNACT: 1,"100.71.68.190"

NOTE: ENABLE THE BOTH NL AND CR OPTION IN THE ARDUINO IDE.
FOLLOWING THESE STEPS EVERYTHING WAS RIGHT.

THANKS FOR [Mr-HaleYa](https://github.com/Mr-HaleYa) FROM THE HELP.

KubaDavid, 2024-07-03:
> Seems this is 01/2024 still not fixed in the current version 0.11.7 of the lib! I'm working with a SIM7080 and run in the same issue Koby / @Mr-HaleYa reported:
> 
> * I uploaded the TinyGSM WebClient example, accidently with the default APN "YourAPN" and did not think of any not-reversible effects
> * after noticing the wrong APN I changed "YourAPN" to "iot.1nce.net", the right one of my telecom provider and uploaded the modified code
> * but nothing happende only `Waiting for network...` was the output, no other, more helpfull debug information
> 
> So I searched here in the issues but it was a bit hard to find this information, also because this issue is closed, so it seemed fixed but it is not. Please reopen this issue, so the information would be more visible!
> 
> So I tried `AT+CGDCONT?` and got
> 
> ```
> +CGDCONT: 1,"IP","YourAPN","0.0.0.0",0,0,0
> +CGDCONT: 2,"IPV4V6","ims","::",0,0,0
> ```
> 
> As reported above the wrong, "old" APN, even I had set the right one in the sketch via `const char apn[] = "iot.1nce.net";`
> 
> I found in [#592 (comment)](https://github.com/vshymanskyy/TinyGSM/issues/592#issuecomment-1596596603) the hint to execute `AT+CGDCONT=1,"IPV4V6","" ` and after this "cleaning" the once more uploaded code did work!
> 
> Sara / @SRGDamia1 or whoever maintains the code, is it possible to have a fix for this buggy behavior?
> 
> I was very glad that I found the posting of Koby / @Mr-HaleYa, he wrote:
> 
> > I think this is the best thing I have found while working with the Sim7000...
> 
> So true, I wouldn't come to any solution without your posting, many thanks!

Can confirm that this is still on ongoing issue and cleaning the APN manually works. Thanks for the sumup!

Sig2018, 2024-07-29:
Faced same problem here with A7670SA.
set APN with AT command works.
Thanks a lot for sharing.