vshymanskyy/TinyGSM #422: ESP32 > Not Sure if  TinyGSM nor HTTPClient.h

bgondell, 2020-07-21:
Summary: Code works fine on WiFi, but on GPRS it crash. Given examples works fine also.

Modem: SIM800L
Main processor board: ESP32 DevKit v1 - ESP core 1.0.4
TinyGSM version: 0.10.6 (latest on Library Manager)
Code: < related code because full code has  900 lines >

```
...
#define TINY_GSM_MODEM_SIM800
#define TINY_GSM_RX_BUFFER 1024
#define SerialAT Serial2
#define SerialMon Serial
#include <HTTPClient.h>

#define USE_GPRS true
#define USE_WIFI false

#ifdef USE_GPRS
  #include <TinyGsmClient.h>
#else
  #include <WiFi.h>
#endif

#ifdef USE_GPRS
    #ifdef DUMP_AT_COMMANDS
      #include <StreamDebugger.h>
      StreamDebugger debugger(SerialAT, SerialMon);
      TinyGsm modem(debugger);
    #else
      TinyGsm modem(SerialAT);
    #endif
  TinyGsmClient client(modem);
#else
  WiFiClient wifi;
#endif

const char apn[]  = "virtueyes.vivo.com.br";
const char gprsUser[] = "vivo";
const char gprsPass[] = "vivo";
const char* url = "http://XXX.XXX.XXX.XXX/ws/saveData";

void setup() {
...

  if ( USE_GPRS ) {
    SerialAT.begin(38400); // Set GSM module baud rate.
    ConnectGPRS(1); // 0 = modem init. 1 = modem restart. Pode levar algunms minutos.
  };

  if ( USE_WIFI ) {
    Serial.println("Connecting to ");
    Serial.println(ssid);
    WiFi.begin(ssid, password);
    
    while (WiFi.status() != WL_CONNECTED) {
      delay(1000);
      Serial.print(".");
    }
    Serial.println("");
    Serial.println("WiFi connected..!");
    Serial.print("Got IP: ");  Serial.println(WiFi.localIP());
    Serial.print("Gateway IP: "); Serial.println(WiFi.gatewayIP());
  }
 ...
}

// Conecta a rede GSM é activa comunicaçao GPRS
void ConnectGPRS(int mode) { 

[... 142 lines cut ...]

> Decoding stack results
> 0x4008c50c: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
> 0x4008c73d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
> 0x4008890d: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1442
> 0x401217a2: sys_mutex_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 78
> 0x401219a2: sys_arch_protect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 469
> 0x40113e04: do_memp_malloc_pool at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 302
> 0x40113e6d: memp_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 398
> 0x4011100a: netconn_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_msg.c line 742
> 0x4010fe94: netconn_new_with_proto_and_callback at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_lib.c line 133
> 0x40113688: lwip_socket at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c line 1587
> 0x400d759e: WiFiClient::connect(IPAddress, unsigned short, int) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/lwip/lwip/sockets.h line 593
> 0x400d6e7a: WiFiClient::connect(char const*, unsigned short, int) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiClient.cpp line 281
> 0x400d683e: HTTPClient::connect() at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 1017
> 0x400d6921: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 552
> 0x400d69f3: HTTPClient::POST(unsigned char*, unsigned int) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 493
> 0x400d6a13: HTTPClient::POST(String) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 498
> 0x400d491d: TaskCommunication(void*) at C:\Users\evelis\Documents\Arduino\PortaV2.3/PortaV2.3.ino line 234 **>>>>>>   int status = http.POST(Dados);**
> 0x40088c21: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
> 

### Debug and AT command log : 

> Now the circuit is under review by the electronic technician to check if there are faults on that side. When the circuit board is reviewed, I will debug if necessary.

SRGDamia1, 2020-07-22:
The [HTTPClient](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/src/HTTPClient.h) that is part of the ESP32 core is only for WiFi connections using the ESP's WiFiClient  It don't think it can be used for any other connection.  You have to use the [ArduinoHttpClient](https://github.com/arduino-libraries/ArduinoHttpClient) or read and write the headers and other parts of the HTTP yourself.  Sorry!

closed 2020-07-22 by SRGDamia1

alexandreberg, 2020-11-20:
> The [HTTPClient](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/src/HTTPClient.h) that is part of the ESP32 core is only for WiFi connections using the ESP's WiFiClient [...]

Hi Sara, thanks for your repply. I checked again the sketch and found an error. Now it is transmitting ok without WiFi.