vshymanskyy/TinyGSM #423: TinyGSMClientSecure does not name a type

far1998, 2020-07-22:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: 
Main processor board: 
TinyGSM version: 
Code: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### Debug and AT command log

TinyGSMClientSecure does not name a type, are you sure it is still there or it is already removed?
I try to see the code but it seems like it doesn't support secure mode (SSL)

SRGDamia1, 2020-07-22:
SSL isn't supported yet on the SIM7000 series.

closed 2020-07-22 by SRGDamia1

EmanueleFeola, 2021-04-14:
@far1998a ssl support for sim7000 is in phase of development, I tested it and it is working. Here there is the discussion #437