vshymanskyy/TinyGSM #424: compile problems on platformio 'isGprsConnected' not a member

vossie123, 2020-07-23:
Hello i just started using platformio and starting to love it:)

The problem that i have is that tinygsm won't compile
I am getting this issue:
error: 'class TinyGsm' has no member named 'isGprsConnected'

same issue #114
lathoub commented on 17 Dec 2017
however that one is closed so i am not sure if he got it fixed.
Sorry for reporting / reposting this.

Great work on the library it is working great on arduino ide.

Best regards

SRGDamia1, 2020-07-23:
Thanks for opening a new issue. 👍 

What code are you trying to run?  One of the examples?  Which module are you using and what board are you trying to compile for?  Do you have a ```#define TINY_GSM_MODEM_xxxx``` right at the top of your program, above the ```#include <TinyGsmClient.h>```.

SRGDamia1, 2020-07-23:
Did you use the Arduino IDE's "include" button?  If so, it probably included every h file in the library, which you don't want.  You should only include TinyGsmClient.h, none of the others?

vossie123, 2020-07-24:
Hello thank you for replying
i am using the sim800 module witch is defined before including TinyGsmClient.h
[code]
#define TINY_GSM_MODEM_SIM800
#include <Arduino.h>
#include <TinyGsmClient.h>
[/code]
using the AllFunctions.ino example i am getting the error.

Nothing else included.
Thanks for helping out

vossie123, 2020-08-02:
This issue can be closed as it is caused by a noob... (me)
When installing an library on platformio it helps to select the latest version...
i just clicked on install never seen that the first version of tinygsm was installed :)