vshymanskyy/TinyGSM #426: sim800L download file via https

EmanueleFeola, 2020-07-28:
does anyone know how to download a file via https on sim800l?
In the example folder there is a download example via http, but not https

Mr-HaleYa, 2020-07-29:
https downloads are complicated because you need a certificate to authenticate to the server that you aren't a hacker. that's why HTTP(S-ecure) . 
My company uses a RESTapi (basically an HTTP server) to do all file downloads and uploads then have the have that server do the handshake and authing to the HTTPS. so if you are looking for long term file transferring look into that.

the HTTPS certificates tend to expire every 30-90 days and you would have to re-flash a new one. (I know there are other ways but using a certificate is the easiest) that's why HTTP is the example provided since way easy to use

EmanueleFeola, 2020-07-30:
Thank you very much for your answer.
In the scenario where I have to download updates from the server (via gprs) and flash them on esp32, how would you do?
For now, the esp32 makes an http request and downloads the file and then compares the md5 hash of the file with the one the server delivered. If the hash are not the same, I do not flash the update

Mr-HaleYa, 2020-07-30:
wow, I literally just finished helping somebody with this exact problem and as I told them, I'm the only person I know that has got this to work correctly.

Like I said previously I use an HTTP server to hold my data that the esp needs so that includes my bin files for flashing. do you have an Instagram? it is easier to talk on there for me. @hales_tech if not then I can try and get it all explained here. 

I use a function to get and HTTP req and read the data and write it directly to the flash mem of the esp. after this is done it does and MD5 hash check and if it passes it reboots into the new sketch. If it gets cut off or is not a good hash it just clears what it wrote so you don't get a corrupted sketch.

EmanueleFeola, 2020-07-30:
I do not have an instagram account, this is my skype id live:emanuele.feola_1, and this is my telegram username emanuele_feola
Can you contact me there?

Mr-HaleYa, 2020-07-30:
lol, I don't have skype and I have never heard of telegram...

EmanueleFeola, 2020-07-30:
Ah ok no problem
Anyway, I do kind of the same thing you do downloading via http and then checking the integrity of the file with md5 hash.
The thing I'm worried about is that I use http to download, which is not secure. Could that be a security flaw?
(i'm deleting the last message since it is useless now)

Mr-HaleYa, 2020-07-30:
no lol 
since you are using md5 hash there is literally no way that they could alter your code and your ESP not know it unless they completely replaced your bin file basically as soon as you make the request they would have to cut your stream and start inserting their stream and then what's it going to do? brick your ESP? its literally going to do nothing for them. are they going to steal your bank account off your ESP? 🤣 

so is you method working? bc mine used HTTP as well

EmanueleFeola, 2020-07-30:
Yes it is working, I'm using http to download the file, an SD card to store the file and hash md5 to do the security check.
Could someone do packet sniffing on http traffic and gather some data? I don't think it is possibile on httpS.
The reason I'm very careful is that this software is going to be on some electric wheelchair and this imposes lot of "security constraints"

Mr-HaleYa, 2020-07-30:
even IF they "packet sniffed" your whole bin wtf are they going to do with it??? I could write my SSN and Credit card info and compile as a bin and give it to you. it would take SERIOUS work to get ANYTHING out of it but 6d6ha231skjds31akj6dhsa (garbage txt) so yes and no, it could be done but they could also just steal the wheelchair...

EmanueleFeola, 2020-07-31:
ok, thank you for your answers!

Mr-HaleYa, 2020-07-31:
You are welcome. I have solved your problem please Close this issue

closed 2020-07-31 by EmanueleFeola

SRGDamia1, 2020-07-31:
Thank you @Mr-HaleYa for working through this!

Mr-HaleYa, 2020-07-31:
@SRGDamia1 Least I could do for all the help you have given me