vshymanskyy/TinyGSM #428: Auzure Iot MQTT SSL on SIM800 - needs  "+SSLOPT=1,1"

better-adapted, 2020-07-31:
This problem occurred on a 
- TTGO T-Call V1.3 ESP32 will a SIM800L
- Running MQTT code adapted from the example - for secure MQTT on TLS 1.2
- TinyGsmClientSecure client(modem); Secure verison used
- Azure Iot MQTT TLS 1.2 Hub

Testing to a ClouldMQTT on a SSL port - all good working fine - but then i could not establish a connection to an Azure IoT MQTT server uing TLS1.2

in **modemConnect()** I have had to add 
    **if (ssl)
    {
      sendAT(GF("+SSLOPT=1,1"));
      rsp = waitResponse();
      if (rsp != 1) {
        return false;
      }
    }**
just before 
**sendAT(GF("+CIPSSL="), ssl);**

After that it worked great - its been on solid for 3 days now - good work

the star I just gave you brought you up to 1000 - no fireworks?

Should I raise a pull for this - not sure if this is the exact place to put the patch - perhaps it would be better in setup code?

[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: 
Main processor board: 
TinyGSM version: 
Code: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### Debug and AT command log

SRGDamia1, 2020-08-07:
Woah.  I hadn't even noticed there were that many stars!

I added the code to the SIM800.  Activate it by setting the pre-processor flag ```TINY_GSM_SSL_CLIENT_AUTHENTICATION```.  Look ok?

better-adapted, 2020-08-11:
yes that sounds good

closed 2020-08-11 by SRGDamia1

better-adapted, 2020-08-11:
cheers i will be using it next week

gsvitak, 2020-10-26:
@better-adapted thanks for your post..

do you have a full sample to share? Maybe a Gist?

better-adapted, 2020-10-27:
I think it is now in the master - you can check my fork too

kedzior-io, 2021-01-28:
@better-adapted 
Hey, I'm struggling to get this working with Azure IOT Hub. Seems like you have it working. 

Here is what I have: 
https://github.com/kedzior-io/esp32-sim800l-tinygsm-azure-iot-hub-mqtt-ssl/blob/master/mqtt-azure.ino

Here is the error I'm getting:
https://github.com/kedzior-io/esp32-sim800l-tinygsm-azure-iot-hub-mqtt-ssl/issues/1

Here is the device I'm using:
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800

Any idea? 
Thanks!

volant0098, 2021-09-14:
Thanks for share your trying.