vshymanskyy/TinyGSM #430: SIM7600E connects to GPRS successfully but fails to connect to MQTT server

shiftops, 2020-08-03:
## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[X] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIMCOM SIM7600E on ESP32 over UART
Main processor board: ESP32 Dev
TinyGSM version: 0.10.7
Code: See below debug output

### Scenario, steps to reproduce
Connect to an MQTT server using the example provided in the source code

### Expected result
Successful MQTT connection

### Actual result
Connection fails with ### Unhandled: 0,0

```
Wait...
Initializing modem...
[2258] ### TinyGSM Version: 0.10.7
[2258] ### TinyGSM Compiled Module:  TinyGsmClientSIM7600
[2274] ### Modem: SIMCOM SIM7600E
[2274] ### Modem: SIMCOM SIM7600E
Modem Info: Manufacturer: SIMCOM INCORPORATED Model: SIMCOM_SIM7600E Revision: SIM7600M21-A_V1.1 IMEI: XXXXXXXXXXXX +GCAP: +CGSM
Waiting for network... success
Network connected
Connecting to telstra.iph success
GPRS connected
=== MQTT NOT CONNECTED ===
Connecting to broker.hivemq.com[18929] ### Unhandled: 0,0

// Actual AT command...
AT+CIPOPEN=0,"TCP","broker.hivemq.com",1883
OK

+CIPOPEN: 0,0
```

llaneiro, 2020-08-06:
Do you have any update?

I have a similar problem. I'm connected to GPRS but I can't connect to the broker.
These are the commands
```
* Trying to connect to the MQTT Broker: maqiatto.com
AT+CIPCLOSE=0,1

ERROR
AT+CIPSSL=1

OK
AT+CIPSTART=0,"TCP","maqiatto.com",1883

OK

0, CLOSE OK
Failed to reconnect to the broker.
Status: -2
```

shiftops, 2020-08-06:
Sorry, nothing yet. I am considering looking for another library with SIM7600 support.

I was hoping the devs might be able to take a look at this, as they have developed this for the SIM7600.

Your issue seems to be failing on a different AT command so it may not be related to this issue.

SRGDamia1, 2020-08-07:
Sorry, I've been working on other things. This hasn't been open *that* long - it's been less than a week.

@llaneiro - are you also using a SIM7600?  Or are you talking about the same issue you had with the SIM800 and MQTT SSL?  I don't think that's related to this at all.

@shiftops - I'll add a step to catch the second half +CIPOPEN response.  It was being thrown out, which is probably what was causing your error.  Can you pull master from here to test:  https://github.com/EnviroDIY/TinyGSM

shiftops, 2020-08-08:
Thanks for your help. I am testing this out now.

shiftops, 2020-08-08:
Seems to get past this step now, however I am now getting a "Network error" response.

I can confirm that GPRS is connected, as an IP is returned.

Here is the full verbose output:

`
Cycle modem power...
[248] Pin HIGH
[348] Pin LOW
[1358] Pin HIGH
Wait...
Initializing modem...
AT
AT
OK
AT+CRESET
AT+CRESET
OK
[12466] ### TinyGSM Version: 0.10.8
[12466] ### TinyGSM Compiled Module:  TinyGsmClientSIM7600
AT
AT
OK
ATE0
ATE0
OK
AT+CMEE=2

OK
AT+CGMM

SIMCOM_SIM7600E

OK
[12482] ### Modem: SIMCOM SIM7600E
[12482] ### Modem: SIMCOM SIM7600E
AT+CTZR=0

OK
AT+CTZU=1

OK
AT+CPIN?

+CPIN: READY

OK
ATI

Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600E
Revision: SIM7600M21-A_V1.1
IMEI: XXXXXXXXXXXXX
+GCAP: +CGSM

OK
Modem Info: Manufacturer: SIMCOM INCORPORATED Model: SIMCOM_SIM7600E Revision: SIM7600M21-A_V1.1 IMEI: XXXXXXXXXXXXXX +GCAP: +CGSM
AT+CPIN?

+CPIN: READY

OK
AT+CPIN="0282"

OK
Waiting for network...AT+CGREG?

+CGREG: 0,1

OK
 success
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to telstra.iphAT+NETCLOSE

+NETCLOSE: 2

ERROR
AT+CGDCONT=1,"IP","telstra.iph","0.0.0.0",0,0

OK
AT+CIPMODE=0

OK
AT+CIPSENDMODE=0

OK
AT+CIPCCFG=10,0,0,0,1,0,75000

OK
AT+CIPTIMEOUT=75000,15000,15000

OK
AT+NETOPEN

OK

+NETOPEN: 0 success
AT+NETOPEN?

+NETOPEN: 1

OK
AT+IPADDR

+IPADDR: 10.141.125.90

OK
GPRS connected
AT+CIPRXGET=4,0

+IP ERROR: 13

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
=== MQTT NOT CONNECTED ===
Connecting to broker.hivemq.comAT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
AT+CIPRXGET=1

OK
AT+CIPOPEN=0,"TCP","broker.hivemq.com",1883

OK

+CIPOPEN: 0,0
AT+CIPSEND=0,27

GsmClientTest␀
OK

+CIPSEND: 0,27,27
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPEVENT:[19720] ### Network error!
AT+NETOPEN?

+NETOPEN: 0

OK
[20721] ### Unhandled: +NETOPEN: 0

OK
AT+NETCLOSE

+NETCLOSE: 2

ERROR
AT+CIPCLOSE=0
 fail
AT+CIPRXGET=4,0
AT+CIPCLOSE?
`

QMand, 2020-08-14:
> Sorry, I've been working on other things. This hasn't been open _that_ long - it's been less than a week.
>
> @llaneiro - are you also using a SIM7600? Or are you talking about the same issue you had [...]

I also tested this and it fixes that problem.

For my device shuts down the modems power with a mosfet so when testing I 
Power on
Connect
Send
Shutdown
This is done frequently because I'm testing it.

I then went on to see why CIPOPEN does not connect every time because I have got very high signal quality and its LTE so I put a DBG("### opened_result: ", opened_result); and often I get a value 10. So I believe that CIPOPEN times out sometimes. 

I then thought that maybe the DNS resolution does not work and used an IP address but this does not fix the issue of sometimes not working.

But in anycase just wanted to put this here.

Amit-Agrawal0177, 2021-01-22:
if you stop calling stop function in mqtt loop then it will not get disconnect.
please go through my pubsubclient library
https://github.com/Amit-Agrawal0177/PubSubClient

dncex, 2021-04-15:
I do have a similar problem, but with SIM800F. The strange thing is that the modem always succeded in conneting to GPRS network, but then, around 50% of the times, it fails to connect to the mqtt:

AT+CIPSTART=0,"TCP","xxx.xxx.xxx.xxx",1883

OK
[250626] ### Unhandled: OK

The strange point is that:
- after the error if I retry to connect to the mqtt , it fails
- after the error, if discconect from gprs, and then reconnect to gprs and then to the server...it works (sometimes at the first retry, sometimes after some retry)

Is this normal? I mean... it might be related to the network congestion?

SRGDamia1, 2021-04-28:
From my perspective, it's totally normal for all low-level Arduino-type cellular modules to be finicky and inconsistent.  But I'm a little cynical about them.

Can you try lengthening your timeout in making the connection?  (Set a value for timeout_s in `client.connect(host, port, timeout_s)`)  The default is already a lengthy 75s, but maybe you need  even more.

When you look at your AT dump, does it look like all the commands are being parsed correctly in the response.  If it looks like the modem is sending back the "CONNECT OK" but the library isn't reading it correctly for some reason, try adding a short yield for TinyGSM to the top of your program (`#define TINY_GSM_YIELD_MS 2`).

osugiw, 2021-09-10:
I have same problem using LilyGo sim 7600 G-H it always fail to connect to broker, did you already found the solution?
[image]

youssef02, 2022-10-29:
@SRGDamia1, I did add the yield but I have still the same problem,

20:16:56.362 -> OK
20:16:57.376 -> AT+CIPSTART=0,"TCP","7.tcp.eu.ngrok.io",16387
20:16:57.410 -> 
20:16:57.410 -> OK
20:17:02.596 -> 
20:17:02.630 -> 0, CONNECT OK
20:18:12.425 ->  fail
20:18:12.530 -> AT+CREG?
20:18:12.530 ->