vshymanskyy/TinyGSM #431: SIM800 - ArduinoHTTPClient POST. Server got empty content but content-length has size.

bgondell, 2020-08-05:
[ X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
   
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[X] Question or request for help

### What are you working with?

Modem: SIM800L
Main processor board: ESP32 DevKit v1 - ESP core 1.0.4
TinyGSM version: 0.10.7 (latest on Library Manager)
Code: < related code because full code has 900 lines >

```cpp
      Serial.println("Executando HTTP POST... ");
      String contentType = "application/json";

      http.post(resource, contentType, Dados);

      // read the status code and body of the response
      int statusCode = http.responseStatusCode();
      String response = http.responseBody();
    
      Serial.print("Status code: ");
      Serial.println(statusCode);
      Serial.print("Response: ");
      Serial.println(response);

      if (statusCode = 200) {
        Dados = "";
        cant =  0;
        success = true;
      }

      http.stop();
```

### Scenario, steps to reproduce
When board start, it send and initial JSON data correctly. But if there are array of JSON, got 200 status but server get empty packet. _The same code works fine if use WiFi instead_.

### Initial JSON

**Serial Print**
> Enviando 1 dados. Size: 187
> [{"ID":"2500.1","t":1596648657,"uuid":51304,"c":8786,"I1":0,"I2":1,"I3":1,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":0,"T2":0,"T3":0,"C1":0,"C2":0,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0}]
> Executando HTTP POST...
> Status code: 200
> Response:

**Packet Recived by Server**
> {"data":[],"json":"[{\"ID\":\"2500.1\",\"t\":1596648657,\"uuid\":51304,\"c\":8786,\"I1\":0,\"I2\":1,\"I3\":1,\"I4\":1,\"I5\":1,\"I6\":0,\"I7\":0,\"I8\":0,\"T1\":0,\"T2\":0,\"T3\":0,\"C1\":0,\"C2\":0,\"C3\":0,\"S1\":0,\"S2\":0,\"S3\":0,\"S4\":0,\"S5\":0}]","files":[],"header":{"Host":"13.58.16.133","User-Agent":"Arduino\/2.2.0","Connection":"close","Content-Type":"application\/json","Content-Length":"187"}}

### Actual result
After the Initial JSON, following POST are coming empties.

**Serial Print**
Enviando 12 dados. Size: 2317
[{"ID":"2500.1","t":1596648660,"uuid":51305,"c":8786,"I1":0,"I2":1,"I3":1,"I4":1,"I5":0,"I6":0,"I7":0,"I8":0,"T1":0,"T2":0,"T3":0,"C1":0,"C2":0,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648671,"uuid":51306,"c":8786,"I1":0,"I2":1,"I3":1,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":0,"T2":0,"T3":0,"C1":0,"C2":0,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648673,"uuid":51307,"c":8787,"I1":0,"I2":1,"I3":0,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":0,"T2":0,"T3":0,"C1":1.52,"C2":2,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648679,"uuid":51308,"c":8787,"I1":0,"I2":1,"I3":1,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":104,"T2":143,"T3":0,"C1":1.52,"C2":2,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648684,"uuid":51309,"c":8787,"I1":0,"I2":1,"I3":1,"I4":1,"I5":0,"I6":0,"I7":0,"I8":0,"T1":104,"T2":143,"T3":0,"C1":0.52,"C2":0.52,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648695,"uuid":51310,"c":8787,"I1":0,"I2":1,"I3":1,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":104,"T2":143,"T3":0,"C1":0.52,"C2":0.52,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648697,"uuid":51311,"c":8788,"I1":0,"I2":1,"I3":0,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":104,"T2":143,"T3":0,"C1":1.32,"C2":1.8,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648703,"uuid":51312,"c":8788,"I1":0,"I2":1,"I3":1,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":106,"T2":144,"T3":0,"C1":1.32,"C2":1.8,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648708,"uuid":51313,"c":8788,"I1":0,"I2":1,"I3":1,"I4":1,"I5":0,"I6":0,"I7":0,"I8":0,"T1":106,"T2":144,"T3":0,"C1":0.5,"C2":0.47,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648719,"uuid":51314,"c":8788,"I1":0,"I2":1,"I3":1,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":106,"T2":144,"T3":0,"C1":0.5,"C2":0.47,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648721,"uuid":51315,"c":8789,"I1":0,"I2":1,"I3":0,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":106,"T2":144,"T3":0,"C1":1.3,"C2":1.75,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0},{"ID":"2500.1","t":1596648727,"uuid":51316,"c":8789,"I1":0,"I2":1,"I3":1,"I4":1,"I5":1,"I6":0,"I7":0,"I8":0,"T1":107,"T2":145,"T3":0,"C1":1.3,"C2":1.75,"C3":0,"S1":0,"S2":0,"S3":0,"S4":0,"S5":0}]
Executando HTTP POST...
Status code: 200
Response:

**Packet Recived by Server**
{"data":[],"json":"","files":[],"header":{"Host":"13.58.16.133","User-Agent":"Arduino\/2.2.0","Connection":"close","Content-Type":"application\/json","Content-Length":"2317"}}

### Debug and AT command log
[... 261 lines cut ...]
> 
> +CIPRXGET: 4,0,367
> 
> OK
> AT+CIPRXGET=2,0,367
> 
> +CIPRXGET: 2,0,367,0
> HTTP/1.1 200 OK
> Date: Wed, 05 Aug 2020 21:05:46 GMT
> Server: Apache/2.4.29 (Ubuntu)
> Set-Cookie: super_mydev_session=55t0o5d8rb2j76ou2r8cf3l62pflav3h; path=/; HttpOnly
> Expires: Thu, 19 Nov 1981 08:52:00 GMT
> Cache-Control: no-store, no-cache, must-revalidate
> Pragma: no-cache
> Content-Length: 0
> Connection: close
> Content-Type: application/json; charset=utf-8
> 
> 
> OK
> Status code: 200
> Response: 
> AT+CIPCLOSE=0,1
> 
> 0, CLOSE OK

SRGDamia1, 2020-08-07:
Did you try breaking your json array into smaller chunks.  I'm not sure if you can send more than 1500 bytes at once with CIPSEND.

tkralj, 2020-08-10:
> Did you try breaking your json array into smaller chunks?  I'm not sure if you can send more than 1500 bytes at once with CIPSEND.

Hello. I'm having similar problem. I have to send JSON string with approx 2000 characters in it. Can you please elaborate on what did you mean by "break json into smaller array" because i am  having one large JSON that i need to POST to my server. How can i break it into smaller chunks and send chunk by chunk to the server ?

SRGDamia1, 2020-08-10:
Instead of using a single ```http.post(resource, contentType, really_big_json_data);``` you should use the start/end commands to break it up:

```cpp
http.startRequest(resource, "POST", ""application/json", total_length);
http.write(data_chunk_1, length_chunk_1);
http.write(data_chunk_2, length_chunk_2);
// etc ... 
http.write(data_chunk_x, length_chunk_x);

// read the status code and body of the response
int statusCode = http.responseStatusCode();
```

You should break the data into <1500 characters.

bgondell, 2020-08-13:
@SRGDamia1 Thanks!! This approach worked fine, would be nice if it was automatic.

Here goes my code for split in chunks, I'm just an enthusiastic on programing, so advices and corrections are welcome!

```cpp
 if ((Data.length() < 1024) || USE_WIFI) http.post(resource, contentType, Data);
      else {  // split in 1024 chunks
        uint8_t temp[1024];
        int chunks = (Data.length() / 1024);
        http.startRequest(resource, "POST", "application/json", Data.length());
        for(int i = 0; i < chunks; i++){
          Data.substring( i*1024 , ((i+1)*1024)-1 ).getBytes(temp, 1024);
          http.write(temp, 1024);
        }
        int lastpiece = Data.length()-(chunks*1024);
        if (lastpiece > 0){
          uint8_t temp2[lastpiece];
          Data.substring(chunks*1024).getBytes(temp2, lastpiece); //Last piece of data
          http.write(temp2, lastpiece);
        } 
      }
```

Kaktusbaum, 2022-02-19:
Here is the code above, but for MQTT. 

        requestBody.toCharArray(requestBody_char_Buffer, 2000);

        int json_length = requestBody.length();
    
         if (json_length <= 1024) {
          mqtt.publish(topic, requestBody_char_Buffer);
         }
          else {  // split in 1024 chunks
            uint8_t temp[1024];
            int chunks = (json_length / 1024);
            mqtt.beginPublish(topic, json_length, false);
            
            for(int i = 0; i < chunks; i++){
              String str = requestBody.substring( i*1024 , ((i+1)*1024) );
              mqtt.print(str);
            }
            int lastpiece = json_length-(chunks*1024);
            if (lastpiece > 0){
              uint8_t temp2[lastpiece];
              String str = requestBody.substring(chunks*1024);
              mqtt.print(str);
            } 
            mqtt.endPublish();
          }