vshymanskyy/TinyGSM #434: other services like websocket.loop() or timeClient.begin () over GSM

holajuanfe, 2020-08-26:
Hello Everyone, thanks for reading my question.
I am trying to program a TTCall with SIM 800L and the TinyGSM library. It's for a college project. Our server works with socket.io.

From the Wifi.h libraries I have no problem, but with the TinyGSM, when I launch services like websocket.loop () or timeClient.begin (), the system reboots and crashes and I have this error in Terminal.

I have tried using TaskHandle_t to use the two cores of ESP32 but I still can't figure out how to do it.

Can someone give me a hand?

Thank you very much!

[ X ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ X] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ X] Question or request for help

### What are you working with?

Modem: 
Main processor board: 
TinyGSM version: 
Code: 

### Scenario, steps to reproduce

### Expected result

### Actual result

holajuanfe, 2020-09-12:
Any Help? Thanks!

SRGDamia1, 2020-10-01:
The ESP core libraries are only built for using the integrated wifi and sockets built on that.  When you're using TinyGSM, the ESP is communicating with the cellular chip using a serial line.  A TinyGSMClient (or any other Arduino Client) cannot be used in place of the ESP's internal client's in this case.  They're completely different classes.

closed 2020-10-01 by SRGDamia1

holajuanfe, 2020-10-06:
Hi! With TinyGSM I was able to make it work with a web socket. Do you know why? It’s that different with socket.io ?

Thanks !

SRGDamia1, 2020-10-27:
I'm sorry, I know this is pretty old.

TinyGSM works on OSI level 4 (or 5 for a few modules that support SSL).  Websockets are a level 7 protocol that works on top of TCP.  So if you found a library that creates a WebSocket on top of a standard Arduino client, then it would work with TinyGSM.  But functions like the websocket and timeclient that are part of the ESP32 core are *NOT* built on a standard "Client" object; they use a separate class unique to the ESP32 using its internal wifi connection.  Other libraries may use the ESP's own Client type or they may use the standard Arduino Client, it depends on the library.  But you cannot supply a TinyGSM Client in place of the ESP32's separate WebClient.

maggusscheppi, 2020-11-29:
I am on the same search for a way to create a secure websocket via GSM on the TTGO T-Call. Does anyone has a running sample? The ESP-IDF PPPos sample misses some preconfiguration of the SIM800L and in the Arduino-Libraries there seems to be no way to connect a websocket to a instanciated GSM-client.
Would be great, if you got a hint.
kind regards, Markus

ArduChecker, 2021-02-08:
@holajuanfe and @maggusscheppi 
Hi there, I recently came across the same issue...I want to run a wss connection via TinyGSM / TTGO T-Call 800L...the libraries I googled to use are mainly 
#include <TinyGsmClient.h>
#include <ArduinoWebsockets.h>

However, I have not been able to them them interworking, I am not a developer to that depth..., but I was lucky to read that it is possible in principle...would it be possible for you  to share some code snippets on this?
Thank you very much in advance!
Kind regards,
     Michael

maggusscheppi, 2021-02-09:
Hi, 
finally I got it running. GSM, GPS and sending data via a secure websocket connection.
I would like to share the code but I am not familiar with dropping code in github.
So i can tell only this link to my webserver for the sources.
https://schepplications.de/esp32/esp_gsm_gps_secure_websocket.zip

I have commented out 
In case u want to use GPS, pls. check  main/nmea_parser.h:
#define NMEA_PARSER_CONFIG_DEFAULT()
.rx_pin = 2     tells, on which pin to connect the tx of the GPS device. I did not connect the rx of the GPS device.

kind regards, markus

DomolinBolivia, 2022-01-31:
No se puede descargar el codigo.

jasononline, 2022-03-06:
> Hi, finally I got it running. GSM, GPS and sending data via a secure websocket connection. I would like to share the code but I am not familiar with dropping code in github. So i can tell only this link to my webserver for the sources. https://schepplications.de/esp32/esp_gsm_gps_secure_websocket.zip
> 
> I have commented out the and for connecting via secure websocket in main.c. Pls. use your own installation. In case u want to use GPS, pls. check main/nmea_parser.h: #define NMEA_PARSER_CONFIG_DEFAULT() .rx_pin = 2 tells, on which pin to connect the tx of the GPS device. I did not connect the rx of the GPS device.
> 
> kind regards, markus

Hi @maggusscheppi,

I am doing project with same functionality. Unfortunately I can't make WebSocket and SIM800L work together. The link you provided gives this response `The requested URL /esp32/esp_gsm_gps_secure_websocket.zip was not found on this server.` I would be very grateful if you could provide the code

maggusscheppi, 2022-03-07:
Hi, try again pls.
Kind regards, Markus

in main.c you have to change
#define CONFIG_WEBSOCKET_URI "wss://<secureWebSocketServer>:<Port>"
with your data, e.g. "wss://myPrettyWebSockSever.tw:8888"
Also I have soldered the GPS RX pin on GPIO 2 as far as I remember.

Dhanvanth888, 2025-06-25:
> Hi! With TinyGSM I was able to make it work with a web socket. Do you know why? It’s that different with socket.io ?
> 
> Thanks !

Could you please tell me how did you make tinygsm work if possible can you share your code