vshymanskyy/TinyGSM #435: Problem with the HttpClient example.

khepri420, 2020-09-02:
[ x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x ] Question or request for help

### What are you working with?

Modem: SEQUANS Communications GM01Q UE5.2.0.1
Main processor board: Nucleo L476RG
TinyGSM version: 0.10.8
Code: 
`/**************************************************************
 *
 * This sketch connects to a website and downloads a page.
 * It can be used to perform HTTP/RESTful API calls.
 *
 * For this example, you need to install ArduinoHttpClient library:
 *   https://github.com/arduino-libraries/ArduinoHttpClient
 *   or from http://librarymanager/all#ArduinoHttpClient
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 * For more HTTP API examples, see ArduinoHttpClient library
 *
 * NOTE: This example may NOT work with the XBee because the
 * HttpClient library does not empty to serial buffer fast enough
 * and the buffer overflow causes the HttpClient library to stall.
 * Boards with faster processors may work, 8MHz boards will not.
 **************************************************************/

// Select your modem:
//#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
 #define TINY_GSM_MODEM_SEQUANS_MONARCH

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

#ifdef ARDUINO_ARCH_STM32
HardwareSerial Serial1(PA10, PA9);  // RX, TX
#endif

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
[... 417 lines cut ...]
AT+CTZU=1

OK
AT+CPIN?

OK

ERROR
AT+CPIN?
AT+CPIN?

+CPIN: READY

OK
AT&F0

+CPIN: READY

OK
ATZ
AT+IPR=0
***********************************************************
 Return settings to Factory Defaults: FAIL
***********************************************************
`

SRGDamia1, 2020-10-01:
I'm sorry; I'm finally checking through a month of issues.  Did you get this working?  I fried the one Monarch chip I had (DOH!) so I can't do much troubleshooting myself.

UCIS, 2021-06-17:
I noticed a similar problem, and think it might have been fixed by https://github.com/vshymanskyy/TinyGSM/commit/38a712500ed862128c5f88e8761a88f4066771dc#diff-54854492bf719b8c61334c1cf1017e0ad996eda65cf986c179b53485358e2e6eL723

Specifically, according to the documentation, it appears that the modem only expects a CR character as a line terminator rather than a CRLF (or even just LF). The old situation caused the redundant LF character to be prepended to the data sequence sent to the remote end, and the last data character being prepended to the next command, corrupting it and causing the subsequent "operation not supported" error.

Anyway, in my case just changing
`const char* gsmNL = CRLF;`
to
`const char* gsmNL = "\r";`
appears to fix the issue, even using when using used "plain text" mode instead of hex mode.