vshymanskyy/TinyGSM #437: SIM7000 SSL

marmotton, 2020-09-05:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[(x)] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[X] Question or request for help

### What are you working with?

Modem: SIM7000E
Main processor board: ESP32
TinyGSM version: github master

### Question
Hello,
I'm currently trying to implement SSL support for the SIM7000E module. To my understanding, one has to use the "CA" commands (CACID, CAOPEN, CASEND, CARECV etc.). This means that not only the modemConnect method must be modified, but also modemSend, modemRead etc.
I guess it is better to stay with the "CIP" commands for plain TCP as some functions wouldn't be available with "CA" commands, e.g. CIPRXGET=4 to query how much data is available and more importantly the possibility of using 8 connections (mux).

Now I'm wondering what would be the best way to go:
* Leave the current SIM7000 implementation as is, and create a new one, e.g. SIM7000SSL (this would also be a very good starting point for adding SIM7080 support later)
* Add another #define for activating the SSL mode (and thus switching to CA commands), but I fear it'll make the code quite messy
* Some other way ?

Thanks for your help,

Ben

SRGDamia1, 2020-10-01:
Hmm.  If they're that different, you might want to make a new implementation.  I'd suggest inheriting from the current implementation and overwriting what you need.  That's what the SIM808 does to add GPS functionality to the SIM800.

marmotton, 2020-10-21:
Thanks, I'll do this then !

gsvitak, 2020-10-26:
@marmotton Can we work together to add support for SSL with the SIM7000? 

I would be happy to contribute.

Do you have branch started?

marmotton, 2020-10-28:
Hi @gsvitak , yes of course ! I did some cleanup, this is the branch: https://github.com/marmotton/TinyGSM/tree/SIM7000SSL . I also wrote the [SIM7000SSL-notes.md](https://github.com/marmotton/TinyGSM/blob/SIM7000SSL/SIM7000SSL-notes.md) to help us getting started.

I have not started writing code yet, only tried to find out which AT commands to use. Would you like to create the new file containing the "backbone" according to what SRGDamia1 proposed ?

marmotton, 2020-11-15:
I made some progress, for now I created a separate file as it was getting quite messy with inheritance, maybe once it works we can make it inherit from SIM7000 cleanly. I couldn't manage to use the methods that override parts of the templates, but I must say I'm not very experienced in C++.

The current status is that it can connect with SSL but I couldn't find out how to read the received data. @SRGDamia1 or @gsvitak  I'd greatly appreciate some help on this.
When data arrives, the unrequested code `+CADATAIND: 0` (0 = mux) is received. Apparently there is no way to ask the module how much and if data is available (apart from listening to CADATAIND). When trying to read up to 100 bytes (100 is an example value) with `AT+CARECV=0,100`, the reply will be e.g. `+CARECV: 6,hello`, 6 being the length of the message.

I'm testing with the HttpsClient example (with `#define TINY_GSM_MODEM_SIM7000_CA`). The source code is here: https://github.com/marmotton/TinyGSM/blob/SIM7000SSL/src/TinyGsmClientSIM7000CA.h

gsvitak, 2020-11-16:
@marmotton Thanks for the great start! Congrats on getting it the modem to connect! 

Sorry for the delay. I have been working on the OTA library first. This week I have allocated to completing this work. Be happy to help.

I will be in touch mid week. I will create a branch off your branch.

Thanks,
Greg

gsvitak, 2020-11-16:
@marmotton one other use case we need to support is the ability to send data via MQTT over TLS. 

I can work on this as well because this use case is what I really need since I need my device to work with AWS IOT Core and port 883 as well as HTTPS Get.

gsvitak, 2020-11-16:
@marmotton sorry one last comment, can you please add the example you are testing with to the branch so we can work off the same thing. Maybe name it SIM7000_SSL and place it under the examples\more directory.. similar to the SIM800_SetSslCert.

marmotton, 2020-11-16:
Hi @gsvitak, I'm quite sure that once the HTTPS example works, MQTT over TLS will work as well. We'll just have to switch from the HttpClient library to PubSubClient. In fact I'll also need MQTT for my project. I once used MQTT over TLS and it took me a while to find out that I simply had to change TinyGsmClient to TinyGsmClientSecure, it is that simple. See here: https://github.com/vshymanskyy/TinyGSM/pull/194/files#diff-0a9018de0839a0b83124170713f574c5a6a10fcb1f95c099e3b643f17277f3de

I put the example that I'm using (it is the https example adapted for the ESP32 board I'm using) and the notes I took about the AT commands here: https://github.com/marmotton/TinyGSM/tree/SIM7000SSL/SIM7000SSL-dev

gsvitak, 2020-11-17:
@SRGDamia1 quick question.. I feel bad asking but do not see much documentation on the required interfaces we need to implement.

We are trying to understand when modemRead is called via HTTP client. Can you please provide a little explanation?

In my research I believe it is called based on the proper defined TINY_GSM_BUFFER_READ_AND_CHECK_SIZE/TINY_GSM_BUFFER_READ_NO_CHECK/TINY_GSM_NO_MODEM_BUFFER. Can you please let us know if we are in the right direction.

https://github.com/marmotton/TinyGSM/blob/6142e360b601b9441adbf1c89e5982deb38f0c93/src/TinyGsmTCP.tpp#L213

Thank you for your great work!
G

marmotton, 2020-12-31:
@gsvitak I figured out how to receive data, I guess we were too scared to look into `int8_t waitResponse(...)`. :facepalm: 

I'm now able to get the file with the tinygsm logo. Next step will be to make sure it doesn't try to read when there is nothing left to read.

gsvitak, 2020-12-31:
@marmotton congrats!! did you push the latest?

marmotton, 2021-01-02:
Yes my repo is up to date

javipelopi, 2021-01-25:
Hi @gsvitak  and @marmotton ! I have been working on parallel in my own fork for the SIM7000. I changed practically all the library to work with the new application that supports SSL. 

@marmotton you can ask if there is data available using AT+CARECV?, however, if there is no active connections, it will stale, so for example I changed the order to check if there is a sock connected before that.

Nevertheless, after a week working on this, I still couldn't connect to Google Core IOT with SSL, and I will be working on something else.

If you want to check my fork and see if you can continue where I left, feel free!

marmotton, 2021-01-25:
Hi @javipelopi, thanks for the info about +CARECV?, I didn't know this. Too bad it is not in the manual.

Using my fork I could retrieve a file using HTTPS and also connect (and send/receive messages) to my own MQTT broker over SSL. So there is some hope ! Maybe you can try my fork with Google Core IOT ?

I had to slightly modify TinyGsmTCP.tpp to make it work. Maybe I won't have to with +CARECV?, let's see..

I'll check out your fork when I start working on this again. I also didn't really do some testing yet, what happens when the connection is closed etc.

javipelopi, 2021-01-25:
Helo @marmotton ! There are no commits on your fork, maybe you forgot to push it from your local repository, please check!

marmotton, 2021-01-25:
Hello @javipelopi , I guess you selected the master branch, check out the "SIM7000SSL" branch: https://github.com/marmotton/TinyGSM/tree/SIM7000SSL

javipelopi, 2021-02-03:
@marmotton I got everything to work on my fork. I wanted to tell you that modemRead() might fail if there is 0 data to read.

I solved it in my fork peeking on the stream like this:
`    if (waitResponse(GF("+CARECV:")) != 1) {
      sockets[mux]->sock_available = 0;
      return 0;
    }

    stream.read();
    if (stream.peek() == '0') {
      waitResponse();
      sockets[mux]->sock_available = 0;
      return 0;
    }

    int16_t len_confirmed = streamGetIntBefore(',');`

If you don't do that, you might get stuck forever in that streamGetIntBefore(','). Just so you know!

Jeroen88, 2021-02-05:
Hi guys, 

I do not use the TinyGSM client, but AT modem commands directly. I am able to connect to google.com using google's root certificate and both TLS encryption and authentication. But when I try to connect to my own website, protected by a Let's Encrypt certificate, I get error 26 on AT+CAOPEN. 

In @marmotton 's implementation, I also see [here](https://github.com/marmotton/TinyGSM/blob/06149b0d1b76a9be63b5f4c014f0f866c2c95a08/src/TinyGsmClientSIM7000CA.h#L484) that the AT+CASSLCFG=<mux>, "CACERT","dstrootx3.crt" is commented out, effectively leading to TLS encryption but **without authentication**.

Has any of you got a working TLS connection using encryption **and authentication** using a SIM7080, SIM7000 or a similar module and the AT+CAOPEN command? Please share your modem commands!

marmotton, 2021-02-05:
@javipelopi thanks and congrats, I'll try this out then ! I haven't been working on this lately..

@Jeroen88 this is commented out as one has to first copy the root certificate to the modem. I tested with the command that is commented out on my own website (also let's encrypt) and it worked. But I first had to copy the certificate as explained here: https://github.com/marmotton/TinyGSM/blob/SIM7000SSL/SIM7000SSL-dev/SIM7000SSL-notes.md#uploading-a-certificate

Jeroen88, 2021-02-06:
@marmotton Thx or the reply. Yes I know I have to copy the certificate. My situation is as follows: I can connect (CAOPEN) to [google.com](google.com) using Google's certificate. I can connect to [valid-isrgrootx1.letsencrypt.org](valid-isrgrootx1.letsencrypt.org) if I also set the SNI. I can connect to my own server at 443 using TLS but no certificate. But I can not connect to my server using the Let's encrypt root certificate, even when setting the SNI. I also can not connect to [letsencrypt.org](letsencrypt.org), even with setting the SNI to [lencr.org](lencr.org) or [letsencrypt.org](letsencrypt.org).

A difference I see is that for the sites I can not connect to the certificate has multiple Subject Alt Names. Might that be the problem?

Could you try to connect to [valid-isrgrootx1.letsencrypt.org](valid-isrgrootx1.letsencrypt.org) (should be succesfull if you set the SNI to this host too) and to [letsencrypt.org](letsencrypt.org)? The latter fails at my set up, I am curious if you can connect!

For all the cases where I can not connect it fails on CAOPEN with error 26: Certificate's common name does not match and time expired.

By the way, my backend is NodeJS with express with no virtual hosts on port 443.

Your help is very much appreciated!

Jeroen88, 2021-02-06:
@marmotton I was using the wrong certicate. I used the active cert ISRG Root X1 but I needed DST Root CA X3. With this root cert it works like a charm, without using SNI. Thnx for your help again!

Jeroen88, 2021-02-06:
@marmotton I come across another thing. It seems that the SIM7080 (and the SIM7080 is quite similar to the SIM7000 at least in command set) does not handle a remotely closed connection properly. If I set the request header 'Connection: close' and I 'GET' a quite long HTTPS document, I get an unsolicited +CASTATE: 0,0 _before_ the full page is received. On the SIM7080 this indicates either an internal error or a connection closed by the remote end. Issuing an AT+RECV after this will yield one (most of the time) incomplete frame or buffer. After that AT+RECV will return that 0 (zero) bytes are available. Did you run into this too? If not, could you check if you can read a full large web page while setting a request header 'Connection: close' on a server that will actually close the connection after sending the payload? Or am I doing something wrong?

marmotton, 2021-02-12:
@Jeroen88 I also observed than once the connection is closed by the server, it is not possible to retrieve the data anymore. I guess one should use connection: keep-alive to avoid this, unfortunately.. Hopefully we'll find something cleaner, but I didn't look into this in more details.

kantum, 2021-02-24:
Hi,
I'm also trying to connect on Google Core IOT with a LilyGO-T-SIM7000G, I would like to achieve mqtts with this sim7000 quite soon, if not I will have to use sim800 which is not as efficient as it does not use lte-m :/ I'm trying to make example from https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master/examples/Esp32-lwmqtt  work with TinyGsmSecureClient from javipelopi 's fork.
@javipelopi did you give up connecting to Google Core IOT using this module? What are you using instead? You said that we can continue where you left, is there a example file in your repo that we can start with or should I just define TINY_GSM_MODEM_SIM7000 allready existing example files?

javipelopi, 2021-02-25:
@kantum Hi!

My fork works, I have tried with Google Core IOT and it does connect. In my case I was not connecting because the expiration date sent in the JWT was 1 month in the future.

However I am using PubSubClient, and based my code in the example used in the TinyGSM library [here](https://github.com/vshymanskyy/TinyGSM/blob/master/examples/MqttClient/MqttClient.ino)

The only change is I added a SecureClient like this

`TinyGsmClientSecure secureClient(modem);`
`PubSubClient mqttClientGsm(secureClient);`

Appart from that and that you need to provide clientID and JWT, I did not do much

kantum, 2021-03-01:
Ok nice! I think I have a problem similar, when I check my token at jwt.io it says it is invalid. Maybe something related to time, I'll give it a try today...
Do you plan to do a PR and close this issue?

gsvitak, 2021-03-02:
@javipelopi Can you provide a link to your fork with a working sample for PubSubClient?

Thanks
G

MarianoFr, 2021-03-12:
Hi everyone, how are you evolving with this implementation? I arrived to a point where I get connection to my broker with an SSL TCP connection, but as an MQTT CONNACK message the server responds 0x20 0x02 0x0D 0x0A, a message not addressed in the MQTT specification, supposedly I should take this as a connection refused, BUT I'm still able to correctly publish BUT unable to receive messages to the topics my client is subscribed to. Any ideas?
To get to this point, I replaced all CIP instructions of the SIM7000G with CA commands, that is CIPRXGET with CARECV (and some others) needing to also change the logic of the program given that responses from both set of commands are different and they also have a slightly different functionality. Also it's very important to check weather your device's time is to date, I added a few lines in the gprsConnect method to correctly update my modem's time and date with the NTP commands.

FStefanni, 2021-03-12:
Hi,

by using the code by @marmotton and @javipelopi (plus some tweak, debug, and extras)
I have created a pr.
I hope this will be helpful to everyone, and anyone is welcome to contribute.

Regards

MarianoFr, 2021-03-12:
Hi @FStefanni that's great news

javipelopi, 2021-03-12:
Hi!

@FStefanni thank you for your efforts!

Please check what I said here as if there is nothing to read this command

` const int16_t len_confirmed = streamGetIntBefore(',');`

will halt the program forever (there will be no ',' to be found)

Appart from that and the quick look that I gave, everything seems okay!

FStefanni, 2021-03-12:
Hi,

thank you for the reply.
I have seen your code, but I was not able to make it work.  Basically, the only true issue I had was in the `modemRead()` during MQTT keepalive polling after some time:
* I set the MQTT keepalive to 30s
* The first keepalive was fine
* The second started huge lately (after 50s instead of 30s), so it was too late to keep the connection open. This was due to the modem code blocked somewhere in the read. 
* So I wrote the read from scratch, but there was still an issue (a double call to the read...) and it was due to `sock_available` not reset to zero

Actually, all the reads have a timeout, and in case of timeout or error it should return some error code (`<= 0` probably, but I am not 100% sure).
This is the case, for example, when the modem replies with: `+CARECV: 0`, where it omits the coma, since there is no data to read. 

But, if you can reproduce the error and post the AT commands sequence, we can try to improve the code by working on it together. Maybe differences could occur, if we are using different SIM7000x versions (mine is SIM7000G R1529).

Regards.

EmanueleFeola, 2021-04-01:
@FStefanni Hi! Is it possible to use your pull request to connect to a mqtt "secure" broker on port 8883 using a ssl certificate?
Currently I'm working with mqtt "not secure" broker on port 1883, but now I have to configure the esp32 to connect to the mqtt broker on 8883 (which is the port that requires the certificate)

Other info: I'm using the pubsubclient library, the mqtt broker is a ubuntu vm on digitalocean platform, I used letsencrypt to generate the certificates

Thank you,
Emanuele

kyriazis, 2021-04-04:
@FStefanni I tried your merge request, and I think there is an issue using SSL.  I was wondering if I'm missing something:

The modemConnect() function in TinyGSMClientSIM7000.h takes in an ssl flag with a default setting on false.  However, modemConnect() is protected and it is called from connect() which is public, but explicitly sets ssl to false.  So, I don't see any way ssl would be turned on from the caller.

It would be nice to have a method that sets ssl before the modem connects.  Most mqtt libraries hide the modem connection under the hood, so one should let TinyGsm know that you want to connect via SSL before telling mqtt to connect.

I've added some temp code in my branch that enables ssl if certificates[mux] is set, but it's quite clumsy.  I'm sure there must be a more elegant solution.

BTW, I'm testing this on a TTGO T-SIM7000G board connecting to Google Cloud IOT.  Works great after some tuning.

Thanks again for adding the functionality to TinyGsm!!

javipelopi, 2021-04-04:
> @FStefanni I tried your merge request, and I think there is an issue using SSL. I was wondering if I'm missing something:
>
> The modemConnect() function in TinyGSMClientSIM7000.h takes in an ssl flag [...]

You just need to use TinyGSMClientSecureSIM7000

I am not sure if it's the exact name but it needs the SECURE surname to use SSL

kyriazis, 2021-04-04:
Ah yes, missed it. :-)

However, there is another fix that I think is needed.  For Google Cloud LTS, I couldn't get it to work without enabling SNI for that host.  Diff below:

```
--- a/src/TinyGsmClientSIM7000.h
+++ b/src/TinyGsmClientSIM7000.h
@@ -593,6 +593,9 @@ class TinyGsmSim7000 : public TinyGsmModem<TinyGsmSim7000>,
     sendAT(GF("+CASSLCFG="), mux, ',', GF("protocol,0"));
     waitResponse();
 
+    sendAT(GF("+CSSLCFG=\"sni\","), mux, ',', GF("\""), host, GF("\""));
+    waitResponse();
+
     sendAT(GF("+CAOPEN="), mux, ',', GF("\""), host, GF("\","), port);
 
     if (waitResponse(timeout_ms, GF(GSM_NL "+CAOPEN:")) != 1) { return 0; }

```

I don't think this will do any harm to enable across the board.

FStefanni, 2021-04-06:
Hi,

@EmanueleFeola yes, it is possible. In my tests, I used the mqtt pubsubclient too. I only have not loaded any certificate, so any certificate will be accepted as valid. On the other hand, you can easily upload a specific certificate to enable the checks. Please look at the discussion in the pr.

@kyriazis thank you for the improvement. At the moment I have no time to work on the pr, but I'll test your fix asap.

Regards

EmanueleFeola, 2021-04-07:
@FStefanni It works perfectly, now esp32 can connect to the mosquitto broker on port 8883 and send/receive messages.
Thank you all for adding this functionality!!

FStefanni, 2021-04-07:
Hi,

@kyriazis I have done a quick test, and actually your fix seems fine. So I have added it. Thank you for the help.

Regards

EmanueleFeola, 2021-04-08:
@FStefanni This morning I have noticed that sometimes the esp32 fails to connect to the mqtt broker.
Usually when it connects to the server, a line is generated on the mosquitto log file, for example: ` New client connected from 151.46.41.126 as 10992`
When it fails to connect no line is generated on mosquitto log file.
Then, to understand what is going on I used tcpdump on port 8883 to see incoming traffic and I noticed a difference in the amount of traffic generated when esp32 connects to server and when, instead, it doesn't connect (very little traffic generated).
Usually it only connects 50% of the time.
Do you have any idea how I can debug this situation?

Thank you in advance, emanuele

FStefanni, 2021-04-09:
Hi,

my suggestion is to enable the logging of AT commands, and then check/compare them.

Also, for SIM7000 sometimes there are some startup issues, which I do not understand, but I implemented a workaround in my app code. Please see this [my reply](https://github.com/vshymanskyy/TinyGSM/pull/507#issuecomment-815505300).

Regards

EmanueleFeola, 2021-04-09:
On the server side everything is working properly, via web app I connect always without problems.
So I guess the issue is on esp32, the AT command that seems to cause the issue is CAOPEN.
If anyone has any suggestions they are really appreciated because I have no idea what to do.

Here is the log when it successfully connects:

```
13837] ### AT: +CGNSPWR=1
[mqttConnect] Connecting to *******.net
[13863] ### AT: +CASTATE?
[13923] ### AT: +CACLOSE= 0
[13978] ### AT: +CACID= 0
[14023] ### AT: +CSSLCFG="sslversion",0,3
[14064] ### AT: +CSSLCFG="ctxindex",0
[14183] ### AT: +CASSLCFG= 0 , ssl, 1
[14225] ### AT: +CASSLCFG= 0 , protocol,0
[14279] ### AT: +CAOPEN= 0 , " *******.net ", 8883
[18772] ### AT: +CASEND= 0 , 35
[19060] ### AT: +CARECV?
[19472] ### AT: +CARECV?
[19533] ### AT: +CARECV= 0 , 4
[mqttConnect] connection success
[19583] ### AT: +CASEND= 0 , 34
[19870] ### AT: +CARECV?
[19932] ### AT: +CASEND= 0 , 23
```

when it fails:
```
[mqttConnect] Connecting to ******.net
[12859] ### AT: +CASTATE?
[12918] ### AT: +CACLOSE= 0
[12974] ### AT: +CACID= 0
[13019] ### AT: +CSSLCFG="sslversion",0,3
[13060] ### AT: +CSSLCFG="ctxindex",0
[13179] ### AT: +CASSLCFG= 0 , ssl, 1
[13221] ### AT: +CASSLCFG= 0 , protocol,0
[13275] ### AT: +CAOPEN= 0 , " ******.net ", 8883
[89275] ### [modemConnect] waitResponse CAOPEN failed (timeout_ms: 
) 75000
[mqttConnect] connection fail
[89289] ### AT: +CASTATE?
[92402] ### AT: +CASTATE?
[95515] ### AT: +CASTATE?
```

javipelopi, 2021-04-14:
@FStefanni I actually hadn't check your PR until now, I like that you kept my modemGetConnected(const char* host, uint16_t port, uint8_t mux) as I need it (I put the arduino to sleep and leave the modem working so this is needed when it wakes up)

I also like the code seems cleaner now, I wanted to use the same syntaxis as in the original but I have to say it was hard to follow.

Good job! I will come back with what I find in the tests!

closed 2021-04-28 by SRGDamia1

TrinVeerasiri, 2021-11-03:
Hi @gsvitak,

I see at the top of the post that you're trying to connect SIM7000 with AWS IoT Core. Do you get it work now?

We use SIM7000E and use AT command to connect to the AWS but it can't. I guess the problem is about the Root CA checking process. We've already stored the Root CA, Device and Private key in SIM7000E. Still stuck at this step for a while.

It doesn't have to be an AT command, it can be TinyGSM. Please suggest me.

kantum, 2021-11-12:
@TrinVeerasiri I was stuck as well with all those certificate with GCP (I think it is similar to AWS), what I did is making it work with [test server of mosquitto](https://test.mosquitto.org/ ) first and then change the CA certificate with the GCP one.

Seems like SIM7000 needs client certificate even if it is not needed by the server

TrinVeerasiri, 2021-11-15:
@kantum Thank you for your reply.

We already tried connecting to Mosquitto. Using this tools from @tmcadam (https://github.com/tmcadam/sim7000-tools). I can connect to Mosquitto without cert and CA. Then I tried connect with Mosquitto again using only CA. It failed when using "AT+SMCONN" (It's time out after 60 secs). The line "AT+SMSSL=0, ca.crt" is error too.

I generate cert and key follow the step in this guideline http://www.steves-internet-guide.com/mosquitto-tls/. Then put it in "customer" folder in file system. I already upgrade firmware to "1351B08SIM7000E".

The result is in the log below.

```
++++++++++++++++++++ MQTT - CA Cert Only +++++++++++++++++++++

-----------  AT+CNACT=1,ciot  -----------
Error (0.04secs):
        +CME ERROR: operation not allowed

-----------  AT+CNACT?  -----------
Success (0.08secs):
        +CNACT: 1,"10.226.213.196"
        OK

-----------  AT+SMCONF="CLIENTID", "TOMTEST01"  -----------
Success (0.04secs):
        OK

-----------  AT+SMCONF="KEEPTIME",60  -----------
Success (0.04secs):
        OK

-----------  AT+SMCONF="CLEANSS",1  -----------
Success (0.04secs):
        OK

-----------  AT+SMCONF="URL","test.mosquitto.org","8883"  -----------
Success (0.04secs):
        OK

-----------  AT+CSSLCFG="ctxindex", 0  -----------
Success (0.09secs):
        +CSSLCFG: 0,1,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0,1,""
        OK

-----------  AT+CSSLCFG="sslversion",0,3  -----------
Success (0.04secs):
        OK

-----------  AT+CSSLCFG="convert",2,"ca.crt"  -----------
Success (0.04secs):
        OK

-----------  AT+SMSSL=0, ca.crt  -----------
Error (0.04secs):
        +CME ERROR: operation not allowed

-----------  AT+SMSSL?  -----------
Success (0.08secs):
        +SMSSL: 0,"",""
        OK

-----------  AT+SMSTATE?  -----------
Success (0.08secs):
        +SMSTATE: 0
        OK

-----------  AT+SMCONN  -----------
Timeout (60.00secs):
        +CFUN: 1
        +CPIN: READY
        SMS Ready

```

nicklasb, 2022-03-03:
Any progress on this, anyone have a fork that I can try and see how far I get?

kantum, 2022-03-20:
You can check [this repository](https://github.com/kantum/sim7000-tracker/tree/dev), it was working back then, I just made it public so you can try to get something out of it. The keys are outdated, so I kept it in the repository, you may want to do differently. See the dev branch for last commits.

nicklasb, 2022-03-20:
@kantum Thanks! Will check it out and get back!

JayPalm, 2022-05-03:
> I simply had to change TinyGsmClient to TinyGsmClientSecure,

@[marmotton](https://github.com/marmotton) I've been banging my head against the wall for several hours with this. Thank you, it really was that simple!

Fishtronics, 2022-06-23:
Hi everyone 
I bought the LilyGO-T-SIM7000G and i am trying to send data to google spreadsheet with it see link:https://www.electroniclinic.com/google-spreadsheet-or-google-sheets-with-esp8266-nodemcu-for-data-logging/
but the example shows only how to do it via wifi not via a sim card.
I Have tested the lilygo see link: https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino
 works fine but I don't know how to get it to work with the google spreadsheet?
please help me as have been trying for over two weeks
thanks!!

Fishtronics, 2022-08-31:
please is there any one out there to help me!!?
 how do i put a valid google certifcat on the sim7000

gotnull, 2023-02-23:
> I also wrote the [SIM7000SSL-notes.md](https://github.com/marmotton/TinyGSM/blob/SIM7000SSL/SIM7000SSL-notes.md) to help us getting started.

Link not working.

gotnull, 2023-02-23:
Does this SSL fix work for 7670SA modules?

https://github.com/Xinyuan-LilyGO/T-A7670X

I have raised a ticket:
https://github.com/Xinyuan-LilyGO/T-A7670X/issues/35

gotnull, 2023-02-23:
> Hi @gsvitak , yes of course ! I did some cleanup, this is the branch: https://github.com/marmotton/TinyGSM/tree/SIM7000SSL . I also wrote the [SIM7000SSL-notes.md](https://github.com/marmotton/TinyGSM/blob/SIM7000SSL/SIM7000SSL-notes.md) to help us getting started.

Does this SSL fix work for 7670SA modules?

https://github.com/Xinyuan-LilyGO/T-A7670X

I have raised a ticket:
https://github.com/Xinyuan-LilyGO/T-A7670X/issues/35

calvarez03, 2023-09-16:
> Hi everyone I bought the LilyGO-T-SIM7000G and i am trying to send data to google spreadsheet with it see [...]

Hi Fishtronics, 

Were you able to make this happen? I having the same issues as you.

dalbert2, 2024-06-21:
For anyone finding this thread, on a SIM7000G running 1529B08SIM7000G firmware:

There are numerous issues, including incorrect documentation in the user manual.  For example, the +CARECV: <cid>,<recvlen>  documentation is wrong, it just returns +CARECV: <recvlen>

Unfortunately, AT+CARECV? command just returns OK; no count of characters pending.  

Worse, the <recvlen> provided in the AT+CARECV response is not always correct. For example, when receiving multiple segments (e.g. downloading a file), you may ask for 1024 bytes 
send: AT+CARECV=0,1024\r
and receive an indication that 1024 bytes are ready:
\r\n+CARECV: 1024,
...but the modem won't actually send 1024 bytes after the comma (so include a timeout!)
It works better if you ask for smaller chunks (e.g. 512 bytes) (mismatch in internal buffer sizes?).  I'm still working on figuring out a reliable solution and will post here if I find it.  If anyone else has solved the SIM7000 SSL AT+CAxyz magic, please share it!

PR #507 (2021-04-28): SIM7000: TLS support plus other minor improvements
Hi,

I have based this pr on the great code by @marmotton and @javipelopi.
I have just:
* Packaged their code, aligning it also to the newer master version
* Performed some minor cleanup
* Performed a good debug
* Supported factory reset of the modem

Now it works perfectly with MQTT both with and without TLS.

Fixes #437
Fixes #474 
Fixes #495

Regards