vshymanskyy/TinyGSM #442: Example code for HTTPS Post?

ZebNZ, 2020-09-11:
Hi all,

I can't find any example code for making an HTTPS post,  am I missing something or is there none?

I understand that only certain chips are capable of SSL/HTTPS communication.

Thanks, Zeb

grichfitz, 2020-09-20:
ZebNZ,
Did you get this resolved?
I´m also looking for an example code for HTTPS POST.
G

ZebNZ, 2020-09-20:
Hi @grichfitz,

This code here should make a HTTPS Get request.
https://github.com/vshymanskyy/TinyGSM/blob/master/examples/HttpsClient/HttpsClient.ino

To make a HTTPS Post I am going to take only the GSM code and then just add some basic code below it to perform the post. When I have done that I'll make a pull request to this repo.

Kind regards, Zeb

grichfitz, 2020-09-23:
Hay Zeb,

Yeah I tried this, but it fails. Tried rewriting some of this example, but it never connects..

Proving to be a difficult one!

Thanks anyway

SRGDamia1, 2020-10-01:
I'm sorry, I don't have time right now to write up this example.  I'd suggest you start with the WebClient example, switch to the secure client, and add the post tag in place of the get.

SanFable, 2020-10-18:
@ZebNZ @SRGDamia1 
Hello, any progress with basic example?
I'm trying to make POST request from GET using with http://httpbin.org/anything as a server but all what I'm getting is:
Connecting to server
... failed

I'm totally noob with web services and I don't even know if httpbin.org accepts TCP connection, basic example would be really helpful.

grichfitz, 2020-10-18:
I found a workaround for now, writing to the server via http, read the website via https. Not a real solution. 
I will have to solve this and will post when I do. 1-2 months. Soz no sooner.

genotix, 2021-01-13:
Looking forward to see the result of your efforts, @grichfitz !

mlrochaa23, 2021-05-06:
Hi grichfitz,

Were you able to solve this problem?

genotix, 2021-05-06:
You should instead use SSLClient , a separate library that can also do HTTPS.