vshymanskyy/TinyGSM #444: AT+GSN must be AT+CGSN on sim 7100/7500/7600

QMand, 2020-09-16:
In sim7100_at_command_manual_v1.00 I can only find AT+CGSN Request product serial number identification and not AT+GSN like in the library. 

 String getIMEIImpl() {
    thisModem().sendAT(GF("+GSN"));// must be thisModem().sendAT(GF("+CGSN"));
    thisModem().streamSkipUntil('\n');  // skip first newline
    String res = thisModem().stream.readStringUntil('\n');
    thisModem().waitResponse();
    res.trim();
    return res;
  }

SRGDamia1, 2020-10-01:
can you write up the pr?

garudaonekh, 2022-10-10:
both work

14:54:37.096 -> AT+CGSN

14:54:37.096 -> +CGSN: "869777028421538"
14:54:37.096 -> 
14:54:37.096 -> OK
14:54:45.654 -> AT+GSN

14:54:45.654 -> +GSN: "869777028421538"