vshymanskyy/TinyGSM #447: File download with SARA 412.

naturalmechanics, 2020-10-04:
[ x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
[ ] Bug or problem compiling the library
[ x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?
```

Modem: u-blox Model: SARA-R412M-02B Revision: M0.10.00 [Mar 28 2019 17:13:41] SVN: 05 
Main processor board: sodaq sara R412 AFF
TinyGSM version: 0.10.8
Code:
void handleUpdate() {
	
	bool isValidContentType = false;
	
	const char* PATH = "/update.bin";
	const unsigned long CHECK_INTERVAL = 5000;
	
	
	
	
	if (!modemGSM.gprsConnect(APN, gprsUser, gprsPass)) {															debug2(" fail apn");
	} else {
		if (!client.connect(server, updatePort))	{																debug2("Cannot connect to " + String(server));
			return;
		}
	}
	
	client.print(String("GET ") + PATH + " HTTP/1.1\r\n");
	client.print(String("Host: ") + server + "\r\n");
	client.print("Connection: close\r\n\r\n");
	
	unsigned long timeout = millis();
	while (client.available() == 0)	{
		if (millis() - timeout > RESPONSE_TIMEOUT_MS)	{															debug2("Client Timeout !");
			client.stop();
			return;
		}
	}
	
	if(!SD.begin(SD_CS)) {																							debug2("SD FAILURE ::: ");
		return;
	}
	
	File file = SD.open(BIN_FILENAME, O_CREAT | O_WRITE);
	if (!file) {
		client.stop();																							debug2("Could not create bin file. Can't continue with update.");
		return;
	}
	
	
	const uint32_t clientReadTimeout = 5000;
	uint32_t clientReadStartTime = millis();
	String headerBuffer;
	bool finishedHeader = false;
	uint32_t contentLength = 0;
	uint32_t knownFileSize = 1024; 
	uint32_t timeElapsed = millis();
	
	while (!finishedHeader) {
		int nlPos;
		
		if (client.available()) {
			clientReadStartTime = millis();
			while (client.available()) {
[... 264 lines cut ...]

while(1) {
	MODEM_STREAM.write("AT+CSQ\r\n");																				debug2("AT signal quality checked");
	sodaq_wdt_safe_delay(650);
	String r;
	if (MODEM_STREAM.available())		{																		
		r = MODEM_STREAM.readString();																				debug2(r);
		if ( r.indexOf("OK") >= 0 ) break;;
	}
}

int i = 0;
while(1) {
	i = i + 1;
	MODEM_STREAM.write("AT+CGACT=1,1\r\n");																			debug2("AT GPRS attached");
	sodaq_wdt_safe_delay(650);
	String r;
	if (MODEM_STREAM.available())		{																		
		r = MODEM_STREAM.readString();																				debug2(r);
		if ( r.indexOf("OK") >= 0 ) break;;
	}
	
	if (i > 10) break;
}
```

SRGDamia1, 2020-10-27:
Dump the AT commands with something like this:
```cpp
#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif
```

naturalmechanics, 2020-11-03:
Hi, where exactly should this go? once in void setup() will be enough?

SRGDamia1, 2020-11-03:
On the global scope before setup.  See any of the examples.

IoTThinks, 2021-06-14:
I also hit a lot of FF when download a bin file using this library.
Any solution yet?

Thanks a lot @naturalmechanics