vshymanskyy/TinyGSM #450: ATmega4809/Arduino Every compile error

Gurkengewuerz, 2020-10-09:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
[x] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM800L
Main processor board: Arduino Every
TinyGSM version: 0.10.8

### Scenario, steps to reproduce
I am trying to compile a simple TinyGSM Code for the ATmega4809. In PlatformIO I am using the following enviroment.
```ini
[env:every]
platform = atmelmegaavr
board = nano_every
framework = arduino
monitor_speed = 115200
build_flags =  -D TINY_GSM_MODEM_SIM800
lib_deps =
    ArduinoTime
    ErrorBlink
    SdFat
    TinyGSM
    GSMWrapper
src_filter = +<*> -<writer.cpp> -<counter.cpp>
```
I could already figure out which define cause the error. In TinyGSMCommon.h in line 44 at `#if defined(__AVR__)` the improvements for AVR controller causing some errors. For testing I commented the lines and it worked.
```C++
/*#if defined(__AVR__)
#define TINY_GSM_PROGMEM PROGMEM
typedef const __FlashStringHelper* GsmConstStr;
#define GFP(x) (reinterpret_cast<GsmConstStr>(x))
#define GF(x) F(x)
#else*/
#define TINY_GSM_PROGMEM
typedef const char* GsmConstStr;
#define GFP(x) x
#define GF(x) x
//#endif
```

### Debug and AT command log
```
Processing every (platform: atmelmegaavr; board: nano_every; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/nano_every.html
PLATFORM: Atmel megaAVR (1.1.1) > Arduino Nano Every
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash  
PACKAGES:
 - framework-arduino-megaavr 1.8.6
 - toolchain-atmelavr 1.70300.191015 (7.3.0)        
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TinyGSM> 0.10.8+sha.ad437a5
|-- <GSMWrapper>
|   |-- <ArduinoTime>
|   |-- <TinyGSM> 0.10.8+sha.ad437a5
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\every\src\control.cpp.o
In file included from lib\TinyGSM\src/TinyGsmClient.h:13:0,
                 from lib\GSMWrapper/GSMWrapper.h:20,
                 from src\control.cpp:24:
lib\TinyGSM\src/TinyGsmClientSIM800.h: In member function 'bool TinyGsmSim800::powerOffImpl()':
lib\TinyGSM\src/TinyGsmClientSIM800.h:267:56: error: cannot bind non-const lvalue reference of type 'arduino::String&' to an rvalue of type 'arduino::String'
     return waitResponse(10000L, GF("NORMAL POWER DOWN")) == 1;
                                                        ^
In file included from C:\users\nikla\.platformio\packages\framework-arduino-megaavr\cores\arduino/api/Print.h:24:0,
                 from C:\users\nikla\.platformio\packages\framework-arduino-megaavr\cores\arduino/api/Stream.h:25,
                 from C:\users\nikla\.platformio\packages\framework-arduino-megaavr\cores\arduino/api/Client.h:22,
                 from C:\users\nikla\.platformio\packages\framework-arduino-megaavr\cores\arduino/api/ArduinoAPI.h:29,
                 from C:\users\nikla\.platformio\packages\framework-arduino-megaavr\cores\arduino/Arduino.h:23,
                 from src\control.cpp:19:
C:\users\nikla\.platformio\packages\framework-arduino-megaavr\cores\arduino/api/String.h:67:2: note:   after user-defined conversion: arduino::String::String(const char*)
  String(const char *cstr = "");
  ^~~~~~
In file included from lib\TinyGSM\src/TinyGsmClient.h:13:0,
                 from lib\GSMWrapper/GSMWrapper.h:20,
                 from src\control.cpp:24:
lib\TinyGSM\src/TinyGsmClientSIM800.h:619:10: note:   initializing argument 2 of 'int8_t TinyGsmSim800::waitResponse(uint32_t, arduino::String&, GsmConstStr, GsmConstStr, GsmConstStr, GsmConstStr, GsmConstStr)'
   int8_t waitResponse(uint32_t timeout_ms, String& data,
          ^~~~~~~~~~~~
lib\TinyGSM\src/TinyGsmClientSIM800.h: In member function 'byte TinyGsmSim800::NTPServerSync(arduino::String, byte)':
lib\TinyGSM\src/TinyGsmClientSIM800.h:487:49: error: cannot bind non-const lvalue reference of type 'arduino::String&' to an rvalue of type 'arduino::String'
     if (waitResponse(10000L, GF(GSM_NL "+CNTP:"))) {
```

SRGDamia1, 2020-10-27:
Is the error you posted with or without the change you made?

Gurkengewuerz, 2020-10-27:
@SRGDamia1 the posted error is WITHOUT the changes. The error can be "fixed" with the changes above.

SRGDamia1, 2020-10-27:
Ok.  Good to know.  I'm not sure now the best way to fix it for future users, but I'll leave this issue open for them.

luismanuelrms, 2020-11-06:
Thanks for the solution, I was struggling with this error too and now it works in my Arduino Nano Every. Anyways, if I flash the same program into a normal Arduino Nano, the dynamic memory gets almost full, so it doesn't work in there.

As far as I understand, the lines you commented in the solution are memory optimizations. The normal Nano needs those optimizations due to its low dynamic memory available but in the Arduino Nano Every there is much more memory available, so it's not a problem. I guess I can change the TinyGSMCommon.h depending on the board I want to flash, but there should be another solution...

fishingbit, 2020-11-06:
I've create PR #459 to solve this

Gurkengewuerz, 2020-11-24:
As seen above merged into master

closed 2020-11-24 by Gurkengewuerz