vshymanskyy/TinyGSM #451: init() returns false without SIM card

septatrix, 2020-10-16:
* [x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

* [ ] Request to support a new module
    
* [ ] Bug or problem compiling the library
* [x] Bug or issue with library functionality (ie, sending data over TCP/IP)
* [ ] Question or request for help

### What are you working with?

Modem: SIM808 v2
Main processor board: Arduino Mega 2560
TinyGSM version: 0.10.8
Code: AllFunctions Example Code with only GPS tests enabled (and `restart` replaced with `init` though the same can be observed with both)

### Scenario, steps to reproduce
I am trying to use the GPS functionality of the SIM808 without a SIM card.

### Expected result
Like in 0.7.9 I expected the library to still return `true` on `init()` without a SIM card.

### Actual result
The method returns `false`. 

### Debug and AT command log

septatrix, 2020-10-16:
This was introduced in #327. Alternatively I would like an init method which ignores SIM status.

SRGDamia1, 2020-10-27:
You can just carry on using the library and ignore the fact that the init returns false.  If you were using the init's response as a verification that the modem is responding, use ```testAT()``` instead.

septatrix, 2020-10-29:
`init` also seems to perform some other setup but I have settled with calling and checking `testAT` first and afterwards run `init` without checking the return code. Given that this library still has `0.x.x` I am fine with this little backwards incompatibility and will close this. Thanks for pointing me to the `testAT` method.

closed 2020-11-02 by septatrix