vshymanskyy/TinyGSM #452: error: 'bool TinyGsmSMS<modemType>::sendSMS_UTF8_end() [with modemType = TinyGsmSim7000]' is protected

andymelichar, 2020-10-16:
[X ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[X ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM7000G R1529
Main processor board: LILYGO® TTGO T-SIM7000G ESP32-WR0VER-B,SIH7000G
TinyGSM version: 0.10.8
Code: AllFunctions.cpp example code

### Scenario, steps to reproduce

1. Load example file
2. Add APN value
3. Uncomment `#define TINY_GSM_TEST_SMS true`
4. Uncomment and add phone# `#define SMS_TARGET  "+1234567890"

### Expected result
Expecting code to compile and upload to device

### Actual result
Code fails to compile with the following error:
`In file included from C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClientSIM7000.h:22:0,

                 from C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h:30,

                 from C:\Users\andy\AppData\Local\Temp\arduino_modified_sketch_269509\AllFunctions.ino:105:

C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmSMS.tpp: In function 'void loop()':

C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmSMS.tpp:185:8: error: 'bool TinyGsmSMS<modemType>::sendSMS_UTF8_begin(const char*) [with modemType = TinyGsmSim7000]' is protected

   bool sendSMS_UTF8_begin(const char* const number) {

        ^

AllFunctions:336:44: error: within this context

   res = modem.sendSMS_UTF8_begin(SMS_TARGET);

                                            ^

In file included from C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClientSIM7000.h:22:0,

                 from C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h:30,

                 from C:\Users\andy\AppData\Local\Temp\arduino_modified_sketch_269509\AllFunctions.ino:105:

C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmSMS.tpp:201:13: error: 'TinyGsmSMS<modemType>::UTF8Print TinyGsmSMS<modemType>::sendSMS_UTF8_stream() [with modemType = TinyGsmSim7000]' is protected

   UTF8Print sendSMS_UTF8_stream() {

             ^

AllFunctions:338:45: error: within this context

     auto stream = modem.sendSMS_UTF8_stream();

                                             ^

In file included from C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClientSIM7000.h:22:0,

                 from C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h:30,

                 from C:\Users\andy\AppData\Local\Temp\arduino_modified_sketch_269509\AllFunctions.ino:105:

C:\Users\andy\Documents\Arduino\libraries\TinyGSM\src/TinyGsmSMS.tpp:196:8: error: 'bool TinyGsmSMS<modemType>::sendSMS_UTF8_end() [with modemType = TinyGsmSim7000]' is protected

   bool sendSMS_UTF8_end() {

        ^

AllFunctions:341:34: error: within this context

     res = modem.sendSMS_UTF8_end();

                                  ^

exit status 1
within this context`

### Debug and AT command log
Can't compile so no debug

SRGDamia1, 2020-10-27:
Those UTF begin/end functions aren't meant to be public.  You should be using ```sendSMS_UTF16(const char* const number, const void* text, size_t len)``` not the separate bits.

closed 2020-10-27 by SRGDamia1

InQuize, 2022-07-04:
I am having the same issue.
This is form unmodified (except for enabled SMS define) **example** scketch "AllFunctions" that comes with the library v0.11.5.

talktosanjay3, 2023-08-17:
I am having the same issue.

kemploe, 2025-05-12:
I am also having the same issue. Any help would be really appreciated.