vshymanskyy/TinyGSM #453: BG96 modem.waitForNetwork() always end up in 0,3 (Registration denied)

schmidie, 2020-10-19:
Hi I have problems to get the TinyGSM running on my board.
With this AT commands I can establish successfully a connection:
```
// Band Configuration
// 1NCE: 80 for Band 8
bg96_at("AT+QCFG=\"band\",0,0,80,1", 500);

// Configure RAT(s) to be Searched
// 0 Automatic, 1 Take effect immediately
bg96_at("AT+QCFG=\"nwscanmode\",0,1", 500);

// Configure RAT Searching Sequence
// (LTE Cat NB1 -> LTE Cat M1 -> GSM)
bg96_at("AT+QCFG=\"nwscanseq\",030201,1",500); 

// Configure Network Category to be Searched under 
// 0 LTE Cat M1,1 LTE Cat NB1,2 LTE Cat M1 and Cat NB1
bg96_at("AT+QCFG=\"iotopmode\",1,1",500);

// use PS domain service for data-service
// see http://prog3.com/sbdm/blog/dailinqing1984/article/details/45744565
bg96_at("AT+QCFG=\"servicedomain\",1,1", 500);

// Configure Parameters of a TCP/IP Context 
// <context_type>,<APN>,<username>,<password>,<authentication>
bg96_at("AT+QICSGP=1,1,\"iot.1nce.net\",\"\",\"\",1", 500); 

// Operator Selection
bg96_at("AT+COPS=0,0,\"26201\",9",500);

bg96_at("AT+COPS?",500);

// Query Network Information
bg96_at("AT+QNWINFO",300);

// Query and Report Signal Strength
bg96_at("AT+QCSQ",500);

// Signal Quality Report
bg96_at("AT+CSQ",500);
```

...but I would prefere to use TinyGSM ;-)

[ X ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ X ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: Quectel BG96 Revision: BG96MAR02A07M1G
Main processor board: nrf52840 (https://docs.rakwireless.com/Product-Categories/WisTrio/RAK5010-M/Overview/)
TinyGSM version: 0.10.8
Code: WebClient example

### Scenario, steps to reproduce
Hi, I try to get the TinyGSM running on my board but I always end up in 
**AT+CREG => 0,3 => Registration denied**.
I just configured the tinyGSM like explained in the instructions.
I set the APN (iot.1nce.net) and also tried to call _modem.gprsConnect(apn, gprsUser, gprsPass);_ before network-search.
I also tried completly erase the chip and flashed new bootloader.

### Expected result
AT+COPS=? => List all current network operators

### Actual result
AT+QIACT=1
ERROR

+CREG: 0,2
OK

+CREG: 0,3
OK

than it stays in +CREG: 0,3

### Debug and AT command log
```
Wait...
bg96 power up!

Initializing modem...
[5619] ### TinyGSM Version: 0.10.8
[5619] ### TinyGSM Compiled Module:  TinyGsmClientBG96
AT
AT
AT

RDY
[6497] ### Unhandled: RDY
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+CGMI

Quectel

OK
AT+GMM

BG96

OK
[7234] ### Modem: Quectel BG96
[7234] ### Modem: Quectel BG96
AT+CTZR=0

OK
AT+CTZU=1

OK
AT+CPIN?

+CPIN: READY

OK
ATI

Quectel
BG96
Revision: BG96MAR02A07M1G

OK
Modem Info: Quectel BG96 Revision: BG96MAR02A07M1G
AT+CPIN?

+CPIN: READY

OK
AT+QIDEACT=1

OK
AT+QICSGP=1,1,"iot.1nce.net","",""

OK
AT+QIACT=1

ERROR
Waiting for network...AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,3

OK
AT+CREG?

+CREG: 0,3

OK
AT+CREG?

+CREG: 0,3

OK
AT+CREG?

+CREG: 0,3

OK
AT+CREG?

+CREG: 0,3

OK
```

Anyone has any idea ?

SRGDamia1, 2020-10-27:
If your operator requires the band and technology to be set, you'll have to use some AT commands yourself.  If you want to stick to TinyGSM, you can try something like: (Note:  The ```GF()``` is just an abstraction for flash strings.)

```cpp
// Band Configuration
// 1NCE: 80 for Band 8
modem.sendAT(GF("+QCFG=\"band\",0,0,80,1"));
modem.waitResponse(500);

// Configure RAT(s) to be Searched
// 0 Automatic, 1 Take effect immediately
modem.sendAT(GF("+QCFG=\"nwscanmode\",0,1"));
modem.waitResponse(500);

// Configure RAT Searching Sequence
// (LTE Cat NB1 -> LTE Cat M1 -> GSM)
modem.sendAT(GF("+QCFG=\"nwscanseq\",030201,1"));
modem.waitResponse(500);

// Configure Network Category to be Searched under 
// 0 LTE Cat M1,1 LTE Cat NB1,2 LTE Cat M1 and Cat NB1
modem.sendAT(GF("+QCFG=\"iotopmode\",1,1"));
modem.waitResponse(500);

// use PS domain service for data-service
// see http://prog3.com/sbdm/blog/dailinqing1984/article/details/45744565
modem.sendAT(GF("+QCFG=\"servicedomain\",1,1"));
modem.waitResponse(500);

// Configure Parameters of a TCP/IP Context 
// <context_type>,<APN>,<username>,<password>,<authentication>
modem.gprsConnect("iot.1nce.net");
```

schmidie, 2020-10-28:
:+1: cool, thanks!
works good with
```
  modem.gprsConnect(apn, gprsUser, gprsPass);

  // Band Configuration
  // 1NCE: 80 for Band 8
  modem.sendAT(GF("+QCFG=\"band\",0,0,80,1"));
  modem.waitResponse(300);

  // Configure RAT(s) to be Searched
  // 0 Automatic, 1 Take effect immediately
  modem.sendAT(GF("+QCFG=\"nwscanmode\",0,1"));
  modem.waitResponse(300);

  // Configure RAT Searching Sequence
  // (LTE Cat NB1 -> LTE Cat M1 -> GSM)
  modem.sendAT(GF("+QCFG=\"nwscanseq\",030201,1"));
  modem.waitResponse(300);

  // Configure Network Category to be Searched under 
  // 0 LTE Cat M1,1 LTE Cat NB1,2 LTE Cat M1 and Cat NB1
  modem.sendAT(GF("+QCFG=\"iotopmode\",1,1"));
  modem.waitResponse(300);

  // Operator Selection
  modem.sendAT(GF("+COPS=1,2,\"26201\",9"));
  modem.waitResponse(300);
```

closed 2020-10-28 by schmidie