vshymanskyy/TinyGSM #454: esp32 sim7000G does not connect LTE

EmanueleFeola, 2020-10-20:
Problem: esp32 does not connect to LTE (but it does connect to GPRS!)
In the code, esp32 fails to get network (modem.waitForNetwork() line)
Modem Name: SIMCOM SIM7000G
Modem Info: SIM7000G R1529
Code:

```
#include "utils.h"

File root;

TinyGsm modem(SerialAT);
TinyGsmClient client(modem);
PubSubClient mqtt(client);
MqttClient mqttc(&mqtt, MQTT_BROKER, MQTT_USER, MQTT_PWD);

int counter, lastIndex, numberOfPieces = 24;
String pieces[24], input;

time_t bootTimestamp;
uint8_t mac;

void setup(){
  Serial.begin(115200);
  Serial.println("System is on");

  initSD();
  initSim();

  mqttc.init();
  mqttc.getMqttClient()->subscribe(MQTT_TOPIC_UPDATE);
  mqttc.getMqttClient()->publish(MQTT_TOPIC_SYS, "system startup");

  esp_efuse_mac_get_default(&mac);
  bootTimestamp = getTimestamp();

  randomSeed(42);

  startMqttISR();
  //startLogISR();
}

void loop(){
  delay(1000);
}

/* ISR */
void IRAM_ATTR mqttISR(){
  while (!mqtt.connected()) {
    Serial.println("Disconnected");
    mqttc.connectMqtt();
  }
  
  mqtt.loop();

  char msg[100] = {0};
  int timestamp = bootTimestamp +  esp_timer_get_time() / 10000;
  int accX = random(-10,10);
  int accY = random(-10,10); 
  sprintf(msg, "%d;%d;%d", accX, accY, timestamp); 
  Serial.println(msg);

  bool success = mqtt.publish(MQTT_TOPIC_GPS, msg);
  
  if(!success)
    Serial.println("Failed publishing");
}

/* SIM7000 */
time_t getTimestamp(){
  const char *timestamp = modem.getGSMDateTime(DATE_FULL).c_str();

  struct tm tm;
  time_t now;

  //e.g: 20/05/28,15:59:13+08
  if (strptime(timestamp, "%y/%m/%d,%H:%M:%S", &tm) != NULL )
    now = mktime(&tm);
  else
    now = -1;

  return now;
}

void initSim(){
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  pinMode(PWR_PIN, OUTPUT);
  digitalWrite(PWR_PIN, HIGH);
  delay(300);
  digitalWrite(PWR_PIN, LOW);

  Serial.println("\nWait...");

  delay(5000);

  SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);

  Serial.println("Initializing modem...");
  if (!modem.init()) {
    Serial.println("Failed to restart modem, attempting to continue without restarting");
  }

  String name = modem.getModemName();
  delay(500);
  Serial.println("Modem Name: " + name);

  String modemInfo = modem.getModemInfo();
  delay(500);
  Serial.println("Modem Info: " + modemInfo);

  modem.sendAT("+SGPIO=0,4,1,0");

  if ( GSM_PIN && modem.getSimStatus() != 3 ) {
    modem.simUnlock(GSM_PIN);
  }

  // 2 Automatic , 13 GSM only , 38 LTE only , 51 GSM and LTE only
  String res;
  do {
    res = modem.setNetworkMode(38);
    delay(500);
  } while (res != "OK");

  do {
    res = modem.setPreferredMode(1);
    delay(500);
  } while (res != "OK");

  Serial.println("\n\n\nWaiting for network...");
  if (!modem.waitForNetwork()) {
    delay(10000);
    return;
  }

  if (modem.isNetworkConnected()) {
    Serial.println("Network connected");
  }

  Serial.println("\n---Starting GPRS TEST---\n");
  Serial.println("Connecting to: " + String(APN));
  if (!modem.gprsConnect(APN, GPRS_USER, GPRS_PWD)) {
    delay(10000);
    return;
  }

  Serial.print("GPRS status: ");
  if (modem.isGprsConnected()) {
    Serial.println("connected");
  } else {
    Serial.println("not connected");
  }

  String ccid = modem.getSimCCID();
  Serial.println("CCID: " + ccid);

  String imei = modem.getIMEI();
  Serial.println("IMEI: " + imei);

  String cop = modem.getOperator();
  Serial.println("Operator: " + cop);

  IPAddress local = modem.localIP();
  Serial.println("Local IP: " + String(local));

  int csq = modem.getSignalQuality();
  Serial.println("Signal quality: " + String(csq));

  SerialAT.println("AT+CPSI?");     //Get connection type and band
  delay(500);
  if (SerialAT.available()) {
    String r = SerialAT.readString();
    Serial.println(r);
  }

  Serial.println("\n---End of GPRS TEST---\n");
}
```

SRGDamia1, 2020-10-27:
So setting network mode to 13/GPRS only works, just not 38/LTE only?  Are you sure it's not an issues with your SIM provider or the signal and available bands?  I don't think the SIM7000 requires you to manually set or mess with the bands so I can't think of any other reasons it wouldn't be working.

EmanueleFeola, 2020-10-29:
> So setting network mode to 13/GPRS only works, just not 38/LTE only?
Exactly
I'll try to insert my personal SIM in the SIM7000 so I can see if the problem is the provider SIM or the SIM7000.
Thank you for your input, I'll update this issue as soon as I can!

EmanueleFeola, 2020-11-04:
I inserted the sim in my phone and it works correctly using LTE connection.
So it has to be the SIM7000 that is not working properly.
Can someone help me please?

SRGDamia1, 2020-11-09:
Does the SIM from your phone work with your SIM7000?

EmanueleFeola, 2020-11-10:
No it does not, but I'm not sure my personal sim has a lte connection.
I'm sure the sim on my phone has 4g connection, but I don't know if that implies that I have lte too (maybe it does).

SRGDamia1, 2020-11-10:
I'm sorry, I really have no idea.  I do not think it is a library issue since you can connect on GPRS.  The commands for connection and TCP are identical for both bands.

It seems like an issue with your APN or user/password.

EmanueleFeola, 2020-11-10:
I wish it was a problem related to the APN credentials, that would be easy to fix :)
I have to make some more test.
Do you have a sim7000 your self and is it working with lte?
Anyway, thank you for your time, I'll update it when I have new info

gsvitak, 2020-11-28:
I had the same issues with the same modem.

In you setup function, increase you wait times to delay(10000)

EmanueleFeola, 2020-11-30:
Hi, I tried but it didn't work
I get this output waiting for network. Do you have a test code I can try?

OK
AT+CGREG?
+CGREG: 0,2
OK
AT+CEREG?

gsvitak, 2020-12-01:
@EmanueleFeola 

/**************************************************************
 *
 * This sketch connects to a website and downloads a page.
 * It can be used to perform HTTP/RESTful API calls.
 *
 * For this example, you need to install ArduinoHttpClient library:
 *   https://github.com/arduino-libraries/ArduinoHttpClient
 *   or from http://librarymanager/all#ArduinoHttpClient
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 * For more HTTP API examples, see ArduinoHttpClient library
 *
 * NOTE: This example may NOT work with the XBee because the
 * HttpClient library does not empty to serial buffer fast enough
 * and the buffer overflow causes the HttpClient library to stall.
 * Boards with faster processors may work, 8MHz boards will not.
 **************************************************************/

// Select your modem:
//define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
  #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3); // RX, TX

// Increase RX buffer to capture the entire response
// Chips without internal buffering (A6/A7, ESP8266, M590)
// need enough space in the buffer for the entire response
// else data will be lost (and the http library will fail).
#if !defined(TINY_GSM_RX_BUFFER)
#define TINY_GSM_RX_BUFFER 650
#endif

// See all AT commands, if wanted
// #define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon
// #define LOGGING  // <- Logging is for the HTTP library

// Range to attempt to autobaud
[... 178 lines cut ...]
  }
  if (http.isResponseChunked()) {
    SerialMon.println(F("The response is chunked"));
  }

  String body = http.responseBody();
  SerialMon.println(F("Response:"));
  SerialMon.println(body);

  SerialMon.print(F("Body length is: "));
  SerialMon.println(body.length());

  // Shutdown

    http.stop();
    SerialMon.println(F("Server disconnected"));

    modem.gprsDisconnect();
    SerialMon.println(F("GPRS disconnected"));

  // Do nothing forevermore
  while (true) {
    delay(1000);
  }
}

EmanueleFeola, 2020-12-02:
Hi, thank you for your time. With that code I can't get past "waiting for network"
This is the output I get on the serial monitor

Step 2: [OK] was able to open modem
ATI

SIM7000G R1529

OK
Step 3: Modem details: 
SIM7000G R1529
Waiting for network...
AT+CEREG?

+CFUN: 1

+CEREG: 0,6

OK
AT+CGREG?

+CGREG: 0,6

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

Then after a while of CEREG 0,2 the output changes to 0,0:

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
 fail
Performing HTTP GET request... AT+CIPCLOSE=0

+CME ERROR: operation not allowed
AT+CIPSTART=0,"TCP","vsh.pp.ua",80

+CME ERROR: operation not allowed

gsvitak, 2020-12-02:
@EmanueleFeola can you verify the device is registered on the network. 

I might suggest that you get a free SIM from Hologram. Their platform is designed for device connections on GSM/LTE. You would be able to see via their UI if the device is actually connecting... which I do not think it is. 

I think the real issue is that you probably do not have the SerialAT setup correctly. Depending on your device, this might be a hardware serial via UART instead of software serial. I know for my device, this stumped me a couple of hours.

EmanueleFeola, 2020-12-03:
very well, thank you for your valuable advice.
Today I'm going to order a Hologram SIM, which I did not know has such good deals.
**can you verify the device is registered on the network** I'm testing the SIM on a smartphone and it is working correctly and I see that there is 4G signal working (I can go on google and everything).
I don't know if this is the proper way to verify the device is registered on the network.
**I think the real issue is that you probably do not have the SerialAT setup correctly** I will investigate and make the proper tests as soon as possible

gsvitak, 2020-12-03:
@EmanueleFeola last suggestion that I just thought of. The SIM7000 needs an external power source. I am not sure how you are running the module. If it is through an external shield, you need a LiPO battery if the shield does not provide the extra power. I had to learn this lesson the hard way and I was stuck at one point in the same spot you are now.

EmanueleFeola, 2020-12-03:
I'm using the esp32 with the SIM 7000 module integrated.
The esp32 has a 18650 socket and another lipo input.
Currently I'm using a 18650 battery, but not the lipo input.
I attached a couple of photos, so I explain better the possible power input
[image]
[image]

SRGDamia1, 2020-12-03:
I'm not sure what `CEREG=0,6` means; the manual only documents up to 5.  But 0,2 is searching for network and when it goes to 0,0 that means it's given up.

I'm can't tell without a schematic how the Vin of the SIM7000 is connected to the batteries or USB, but the SIM7000 (along with nearly every other module supported by this library) **_absolutely, positively, without doubt, must have more than the 500mA power in that most Arduino style boards provide_**.  Powering it with anything less than that will give exactly the result you're seeing, it just never connects.  Sometimes you'll see an obvious brown-out when the board suddenly stops responding or resets as it tries to connect, but usually it just doesn't connect.  If you have a LiPo, just plug it in and see what happens.

USB-C connections (there's one on yours) *can* give more juice than the 500mA of an older USB type, but that depends on what's on the other side of the cable.

EmanueleFeola, 2020-12-04:
Hi Sara, thank you.
I should check that aspect too, the schematics are these anyway
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/schematic
Have a nice weekend!

marmotton, 2020-12-08:
@EmanueleFeola are you sure that your provider is compatible with LTE Cat-M1 or NB-IoT ? This might be the reason why the SIM card works in your phone ("normal" LTE) and not on the SIM7000.
I never tested what happens on a network that is not compatible, but I had to search for a while to find a provider which provides LTE Cat-M1, it does not seem very widespread yet.

EmanueleFeola, 2020-12-08:
@marmotton I don't know if my provider is compatible with LTE Cat-M1 or NB-IoT, but I ordered a hologram free sim and it should arrive in a few days.
I just checked and hologram is compatible with CAT-M1
But at this link https://www.hologram.io/pricing/coverage#coverage-table I see that only the USA have CAT-M1 coverage

marmotton, 2020-12-08:
@EmanueleFeola That was also a bit my experience as I'm living in Switzerland. I asked [Thingsmobile](https://www.thingsmobile.com) and they confirmed that it is not available yet. I finally found the [Digitec IOT SIM card](https://www.digitec.ch/de/s1/product/digitec-connect-iot-datenpaket-12-gb-365-tage-mobile-abo-sim-karte-11220218) (not exactly this one, there was an offer for 300MB/365days) which works fine. It is provided by DigitalRepublic, see here what they say about Cat-M1: [FAQ](https://support.digitalrepublic.ch/en/support/solutions/articles/33000225329-do-digital-republic-sim-cards-support-lte-cat-m1-and-nb-iot-)

EmanueleFeola, 2020-12-08:
So the issue was to find a sim carrier that provides either CAT-M1 or NB-IoT coverage.
My current sim provider (WindTre) does not seem to provide such service.
Checking gsma "Mobile IoT Deployment Map" on their website I see that in Italy there is just NB-IoT coverage provided by TIM and Vodafone.
By buying a "Global IoT SIM Card" that supports "2G/GPRS, 3G HSPDA, 4G LTE, LTE CAT-M1" protocols I should still have 4G LTE coverage even without CAT-M1, right?

Elmi77, 2020-12-08:
Same issue here with same modem - and I can say definitely it is not caused by the SIM or the providers capabilities. What I see is exactly the same but when I play around a bit with the network mode value and/or the preferred mode, at some point it starts working. Once it is working, it does so also when I set back my settings to the previous ones, so it can't be an issue of the SIM provider.

One thing I noticed: as long as it fails, I get a signal quality of 99, when it works, it is somewhere around 27.

So as this seems to be exactly the same behaviour, I think there is a problem with that specific hardware.

EmanueleFeola, 2020-12-08:
@Elmi77  **I think there is a problem with that specific hardware** do you mean the esp32 with sim7000 module integrated or just sim7000?

Elmi77, 2020-12-09:
@EmanueleFeola I'm using the LilyGo SM7000G, so yes, it is a ESP32 hardware with the SIM700G. I have two different of them which both are different hardware revisions, both show the same behaviour.

EmanueleFeola, 2020-12-09:
@Elmi77 so it seems to be a common problem with LilyGo SIM7000G.
I have a couple of those and they only connect with gprs. Do yours connect with gprs? 
Anyway, I'm waiting for the Hologram SIM to arrive...hoping to make some improvements
@Elmi77 are you still interested in fixing this issue? maybe we could open a ticket on LilyGo SIM7000G page and ask for support

Elmi77, 2020-12-10:
@EmanueleFeola my board _sometimes_ connects with GPRS - and sometimes it fails completely and does not even registers at the network. As this is random, I'm quite sure it has nothing to do wit hthe SIM-card and it's capabilities. I have opened a bugreport at LilyGo's account, feel free to second it: https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/73

EmanueleFeola, 2020-12-10:
@Elmi77 thank you, once I get the hologram sim and make some tests I will update here and your bugreport if the issues are still there

EmanueleFeola, 2020-12-10:
Hologram SIM has arrived, here the update:
before doing tests I performed modem reset 
- test 1
	- networkMode(38)
	- setPreferredMode(1)
	- result: fail (looking for network never stops)
- test 2
	- networkMode(38)
	- setPreferredMode(3)
	- result: success
- notes: at the beginning I had trouble with the antenna, so I tried a different one and it worked. Then I changed it back to the original one and it began to work (I'm doubtful about the quality of the antenna)

- problem 1: on hologram dashboard I can see only 2g activities, when I connect with networkMode(38) it is like the SIM is not online and not connecting, but it actually downloads "/TinyGSM/logo.txt" correctly!

- problem 2: suddenly it does not connect anymore, not even to gprs and if I try to run reset sketch it fails...I have no idea what happened

Elmi77, 2020-12-11:
So problem 2 sounds like it is the same (random) behaviour like before?

Elmi77, 2020-12-11:
So we still do not know what the reason is...I'm also waiting for the Hologram-SIM, let's see what my results are with it...

EmanueleFeola, 2020-12-12:
I noticed this in my serial monitor log:
Even though I do setNetworkMode(38) to connect to LTE network I get "Contype","GPRS".
Does this mean that it connected to GPRS network instead of LTE?

11:06:30.044 -> AT+SAPBR=3,1,"Contype","GPRS"
11:06:30.091 -> 
11:06:30.091 -> OK
11:06:30.091 -> AT+SAPBR=3,1,"APN","hologram"
11:06:30.138 -> 
11:06:30.138 -> OK
11:06:30.138 -> AT+CGDCONT=1,"IP","hologram"

Elmi77, 2020-12-14:
OK, found something new: my Hologram SIM has arrived and it worked  exactly once the very first time it was inserted. Then it came up with a valid GSM time (call to getGSMDateTime() right after modem was initialised and GPS was started gave 16:30:00).

The next times I tried exactly the same, the GSM-time were 00:00:08 and 00:03:03. In this state getSignalQuality() returned 99 and nothing worked.

The SIM itself is still OK, when I try it in an other device, it is working properly.

EmanueleFeola, 2020-12-14:
@Elmi77 ok, so it is better with the hologram SIM but the connection is very unpredictable with different results within a few minutes of tests (correct me if I understood wrong).
Have you connected successfully to LTE or GPRS? is your country covered by cat-m1 or nb-iot?

Elmi77, 2020-12-14:
I got it working properly with the Hologram SIM but only once - since then it shows the same problem like all my other SIMs. So for the one time it was working it has proved it is not a problem of the network or the provider.

I more have the feeling it has something to do with the SIM7000 module - it "feels" like there are some settings stored of the SIM used that prevent it from working a second time. Unfortunately modem.factoryDefault() does not change anything...

EmanueleFeola, 2020-12-15:
ok, as suggested from lewisxhe it may be useful to debug with AT commands to see if we can gather some info.
but first I have another question I hope someone can answer:
this is what holograms told me about their coverage: "**In Italy, we have 4G LTE coverage but not Cat-M1 or NB-IoT**"
One thing I don't understand: does sim7000g supports only cat-m1/nb-iot and not "normal" lte?
because if that's the case, then I should run my tests with gprs, since I have no chance of connecting to cat-m1 or nb-iot

SRGDamia1, 2020-12-15:
For LTE, the SIM7000 is M1/NB IoT only.

Elmi77, 2020-12-18:
OK, in my opinion it is a hardware problem (means design error). I can see this behaviour with different SIMs, different antennae, different boards and even with the software Lilygo provides. So in my opinion something with this hardware is simply crap.

FStefanni, 2021-03-25:
Hi,

I am also trying to connect a SIM7000G in LTE, but without success: it always access EGPRS...
Have someone succeed in this? Any suggestion?

Moreover, is `AT+SAPBR=3,1,"Contype","GPRS"` truly fine for LTE? Which are `Contype` allowed values? I have been unable to find some documentation.

Regards.

EmanueleFeola, 2021-03-25:
@FStefanni Hi! In my personal experience it was a problem related to my geographic location. I live in Italy and we do not have nbiot or catm1 coverage.
sim7000G works only with nbiot and catm1, _not_ with usual lte that you find in smartphones for example, hence ​I'm forced to use gprs instead of lte.
​So I guess the first step would be to check if in your country there is nbiot or catm1 coverage

FStefanni, 2021-03-25:
Hi @EmanueleFeola,

thanak you for the quick reply. I live in Italy too :) So I believe this is my problem... Nice to understand the solution, i.e. eventually change the modem with one supporting "usual" (standard?) LTE.

Still, it could be nice to know if, when LTE is available, if `AT+SAPBR=3,1,"Contype","GPRS"` is still a desired command to use.

Regards.

EmanueleFeola, 2021-03-25:
"Nice to understand the solution, i.e. eventually change the modem with one supporting "usual" (standard?) LTE"
Do you happen to know if there is a modem that supports usual/standard LTE? i'm interested too

About the command "AT+SAPBR=3,1,"Contype","GPRS"" I honestly have no idea.

Cheers from Verona,
Emanuele

FStefanni, 2021-03-25:
Hi,

@EmanueleFeola: sorry but I do not know, but I am going to check, and I'll let you know.
Cheers from Verona too! LOL!

Regards.

FStefanni, 2021-04-06:
Hi,

@EmanueleFeola: just to let you know, I am testing the UBLOX SARA U201. It seems fine, but I am missing an AT command to get the actually tecnology in use (i.e. a way to check if it is actually using 3G instead of GPRS). The issue is reported here: #513

Regards

EmanueleFeola, 2021-04-07:
@FStefanni good to know, thank you for your feedback!

have a good day,
Emanuele

ejri, 2021-04-22:
Hey @Elmi77 @EmanueleFeola @FStefanni I'm using the same board and it's a hit and miss it seems. same card connects fine on one board, and doesn't connect at all on other boards. Just got 5 boards which are refusing to connect to LTE M1, 3 connect to GPRS, event when I set the preferences: 

networkMode(38)
setPreferredMode(1)

I'm not sure if the issue is the modem itself, but it's definitely not the sim card or the network. 

Any thoughts on this, are you switching to different boards that are esp32-based? any thoughts on this?

EmanueleFeola, 2021-04-22:
@ejri  I wish I could give you a complete answer but troubleshooting these boards is always a nightmare to me.
Anyway, these are the things that I'd check:
- activate AT debug log to see what the modem is sending and receiving (sometimes it happens that the module is not initialized correctly and not even the serial communication works)
- compare logs when the connection is working and logs when the connection is not working
- does your country have cat-m1 coverage? does your carrier provide cat-m1?
- is there a data traffic limit on those sim cards?
You may already checked these steps but that is what I would do.

ejri, 2021-04-22:
Thank you for your response. Indeed, I've already checked all of these. I have suspicions that it's the modem itself (or how it's connected to the board), because I'm also using Quectel's BG96 on the raspberry pi, same sim cards/network, and it seems to work a lot better.

I get that it's a different platform/hardware, but I think the issues are with the modem..

Elmi77, 2021-04-23:
In my personal opinion the hardware is crap. I have no idea if it is the modem or the whole board (I would suspect the last one). But the results, which have been totally random and in no way reproducible, let me think there is a general problem which can't be fixed by the software. I tried with two of these boards, three different SIM-cards/network providers, two different antennae and three different locations (to get connected with other base-stations) but in no case I could get it working properly.

When asking at the GIT-repository of the vendor the answers are in no way helpful. Not sure if they do not have an idea what is going on or if they are aware of the problem and don't want to confess there is a problem with their device.

FStefanni, 2021-04-23:
Hi,

@ejri unfortunately seems you have already tried anything possible... so the last chance is to try another board/modem, because maybe the one you trying could be not working properly.
Sometimes also resetting it to the factory defaults and then retry helps.

Regards.

ejri, 2021-04-23:
Yeah, unfortunately, it seems to be hardware related (or modem.. still unclear, after factory resets...). The idea of having such a board is really great and convenient enough for quick prototyping, but the irreproducibility/inconsistency is a huge hassle...  anyways, thank you all for your responses. I'll try to find other boards..

SRGDamia1, 2021-04-28:
I'd suspect a power issue with that sort of flakiness; maybe not enough amperage or capacitance.  With a all-in-one board, that's probably not easily fixable.

PanSuk21, 2021-06-10:
Thanks for info above,
I have similar issue of sim7070G and sim7000G break board of another manufacturer to share. I used global IoT sim. They could registered to network once and take a few minutes to hours. The 7070G registered to 4G which shown on sim dash board of sim supplier. It can’t registered again until now.
For sim7000G the first used could not registered to network after hours but the second time it did within a few minutes. AT command show GPRS connection.
I also have 7070G dongle version connect to pc via USB. It can registered to network without problem. That ‘s why I bought 7070G break out board to use with microcontroller.
All the boards used can report GPS location doesn’t matter they were registered to network or not. It need to AT command (AT+CGNSPWR=1, note not CGPSPWR) to turn on GPS power first.
One thing I have noted was the way to disconnect power supply from board. I have just simply disconnected from Vin. When reconnect power supply the board auto turn on. This would indicate of something wrong in firmware that cannot recover to initial conditions after power supply disconnection.
Note that the boards used the Vin pin connect power via regulater chip to board at difference point if usb used. However, the board auto turn on either use usb or Vin.
I also put IoT sim in mobile phone. The phone registered to network quickly if APN was correctly set. Please don’t do this unless you are aware of data rate to be paid. So the sim is not an issue.
Finally I don’ think the network or sim card issued. The problem should on the board.

berlingozzo, 2022-06-09:
@EmanueleFeola  any updates on your side?
I have the same exact device, Lilygo TTGO SIM7000g, and an hologram sim.
I've tried everything, including changing country :D
Things I have tried (all combinations):
1) country: Italy or Ireland
2) sim: hologram or Vodafone Italy or WindTre Italy
3) all settings of the AT commands
4) two different antennas (both the "flat" one that comes with the device, and an old rigid finger-looking one
5) two different Lilygo TTGO boards
6) two different sources of power

In the end, I've got a partial refund for the TTGO device.

Hope somebody cracks this issue at some point.

thanks

MarkSDS, 2022-08-25:
Have had four hours of dissapointed views at the screen. 
`Signal: 99 isNetworkConnected: NO CONNECT`
Finally, after altering lines 16 and 19 [in this code](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_NetworkTest/Arduino_NetworkTest.ino) the password ("0000" for my prepaid SIMcard) and for the APN ("advancedinternet") I had immediately a hit! 
`-> Signal: 20 isNetworkConnected: CONNECT
  -> Device is connected .`

This might be a local APN (Netherlands), and I found it by googling the formerly known APN-name of "portalmmm.nl". That's what my regular phone is registrated on. 

Hope this is helpfull for someone. I'm happy for now ;)

cicarelli, 2022-12-11:
Hi everybody.

I think I've addressed this problem, at least in my earlier tests. My LilyGo SIM-7000 v1.1 is acting like all of your descriptions. When I turn it on, it works perfectly, get LTE network, and detects CNMP=2 (auto) and CMNB=3 (both CAT-M and NB-IoT). It gets network, high signal level (about 26) and connects to HTTP via LTE (which I confirmed using +CPSI). If I turn the board off and turn it on again, it works perfectly in all my tests. But if I simple reset the board using the RST button, it takes longer to get network signal, and shows me lower signal strength (about 12), and terminates the test via GSM.

Then, I changed the start command: Instead of using "modem.init()", I forced the restart with "modem.restart()". It takes more time, but it always gets LTE signal in a very high level (26 - 30), at least untill now. One detail: the modem no longer detects CNMP=2 and CMNB=3; the response is CNMP=38 (just LTE) and CMNB=1 (CAT-M). So the HTTP access runs perfectly, and it appears a bit more quickly.

I'm from Brazil, using VIVO network, and a pre paid SIM card. Other SIM card I have (a M2M SIM Card from ARQIA) doesn't work: apparently it doesn't support LTE.

I hope this helps anyone who is having the same problem.

fgnievinski, 2023-06-18:
here it's suggested to use the command CGDCONT to fix the PDP:
https://github.com/vshymanskyy/TinyGSM/issues/421#issuecomment-664029242

iconnor, 2023-09-01:
For me, adding these two lines solved my connection issues:
```
    modem.sendAT("+CNMP=2"); // Needed to use GSM and LTE
    modem.waitResponse();
    Serial.println("Modem apn gprsConnect");
    modem.gprsConnect(apn);
    modem.sendAT("+NETOPEN"); // I could not start anything until I then opened the network
    modem.waitResponse();
 ```
 
 I am using Telstra over Woolworths so to debug I wanted to see if I could see the networks `AT+COPS?` but I could not until I changed the modes.
 ```AT debug
 AT+NETOPEN
ERROR
AT+CGREG?
+CGREG: 0,0

OK
AT+COPS=?
+CME ERROR: phone failure

OK
AT+CNMP?
+CNMP: 13

OK
AT+CNMP=2
OK
+CGEV: ME DETACH
+CGEV: EPS PDN ACT 1
AT+CNBP?
+CNBP: 0X0000000000000180,0X00000000000800D5,0X0000

OK
AT+COPS=?
+COPS: (2,"Woolworths","Woolworths","...),(3,"vodafone AU","voda AU","...),(3,"Optus AU","Optus","...

AT+CGREG?
+CGREG: 0,1

AT+NETOPEN
OK

```

PanSuk21, 2023-12-07:
Thank you very much,

Please accept my apologies for very delaying in reply.

 

Pan Suk

MARCOadb, 2025-05-29:
> Hi everybody.
>
> I think I've addressed this problem, at least in my earlier tests. My LilyGo SIM-7000 v1.1 is acting like all of your descriptions. When I turn it on, it works perfectly, get LTE [...]

Hi!

I'm also trying to use a VIVO chip in Brazil. I'm using a SIMCOM devkit **(8pya00-SIMCOM-EVB_V1.02 and P4A4605AB SIM7000-TE_V1.01**). I'm facing the same problem. I can connect to GPRS but not NB-IOT. I'm sending serial AT comands to the SIM7000G using Termite terminal. I force de CNMP=38 and CMNB=2 but when I try to force a connection like AT+COPS=1,2,"72406",9 (whitch I found in AT+COPS=?), it always connects in GSM (using the AT+CPSI command). 

What APN did you used? I did some research, we DO have NB-IOT coverage in Brazil, right? I'm from Rio Grande do Sul