vshymanskyy/TinyGSM #455: Incoming event (RING) and DTMF decoding questions

justsilviu, 2020-10-25:
Hello,

i have 2 questions related to the voice call functions.
i am a noob, so if the aswer is obvious please forgive me in advance.

i have read all the readme and went through most of the files in the library, examples, and issues.
i have a sim800 modem, which i have managed to connect, send sms, make a phone call. all fine.

so the questions are:

1. how to i use the getRing even, as stated in the readMe ? - i found no example or method for this
2. how do i decode dtmf that the modem recieves from the other caller? - i found no examples for this also.

are these feature still supported? is the readMe correct? 

"Voice Calls
..........
Functions:
Dial, hangup
Receiving calls
Incoming event (RING)
DTMF sending
DTMF decoding

thanks.

SRGDamia1, 2020-10-27:
Oops.  I'm sorry.  The ReadMe is wrong; incoming calling and text events aren't supported, just outgoing.  I think at one point there had been skeletal support for it, but I'm not sure it ever worked and it's not there any more.

justsilviu, 2020-10-27:
ok,
thanks  👍 
i'll manage with the supported features :)

closed 2020-10-27 by justsilviu