vshymanskyy/TinyGSM #458: GSMServer possibility

pierreverbakel, 2020-11-05:
I have been using the TinyGSM library successfully with the SIM800C on a ESP32 using MQTT and HTTP-Client GETs (From Device to Internet). Now I want to use the Device itself as a HTTP-Server. So my device can be accessed remotely for configuration using the Web-Interface I created, which is working fine on WiFi.

In the example for the GSM Library that comes standard with Arduino written by Tom Igoe there is a object that can be created to setup a server: 

`GSMServer server(80); // port 80 (http default)`

And then in loop()-function:
'GSMClient client = server.available();'
 if (client) {
    while (client.connected()) {
      if (client.available()) {
        Serial.println("Receiving request!");
        bool sendResponse = false;
        while (char c = client.read()) {
          if (c == '\n') {
            sendResponse = true;
          }

        }
      }
    }
  }`

My question, how can I accomplish this within TinyGSM, since I do not find an example for that. 

Thanks for clarification, also when you know that this is not possible. Which would surprise me since it seems like an obvious wannahave to me. Or would I have to use another GSM-library for that in parallel to Tiny-GSM?

Thanks in advance for clarification and help!

mabrubombas, 2022-10-02:
Conseguiu informações? Pode compartilhar?

pierreverbakel, 2022-10-03:
Hello, No I did not get any feedback on this question yet. So at this point I cannot tell you about a solution !

gotnull, 2023-01-13:
Any answer on this?

bill210kouk, 2024-01-14:
Anyone here about this?

gotnull, 2024-01-14:
I haven't heard anything about it since the original post.

bill210kouk, 2024-01-14:
Any other work arounnd? How can coexists timygsm and have server functionallities?

gotnull, 2024-01-14:
> Any other work arounnd? How can coexists timygsm and have server functionallities?

Have you tried something like this? https://github.com/vshymanskyy/TinyGSM/issues/514