vshymanskyy/TinyGSM #460: PWR key pin usage issue - sim868

Elsaligi, 2020-11-10:
Hi,

I have been working with sim868 facing issues to poweron / poweroff sim868.
It has been mentioned in datasheet that  can power on SIM868 by pulling down the PWRKEY pin for at least 1 second and release.
power off -  can power off SIM868 by pulling down the PWRKEY pin for at least 1 second and release.

But as am switching on and off module by another controller and there are cases where i am requesting gsm poweron but pin is identified it as command to power off , i couldn`t able to find exact condition of module.

Is there any function available in tinygsm library for proper on/ off or check the status of module (function like ismodulealive).

Regards,

SRGDamia1, 2020-11-10:
No, TinyGSM doesn't support setting or reading pin levels for on/off/status.

You can use the `testAT()` function to check if the module is responding.

adrianca88, 2021-02-08:
Yes, I check my SIM808 status using testAT() (with an smaller timeout) to check if it's sleeping or not.