vshymanskyy/TinyGSM #462: ### Unhandled: +CIPRXGET: 1,0  in MQTT client

imamminer, 2020-11-13:
[ x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ x] Question or request for help

### Currently working with:
[11660] ### TinyGSM Version: 0.10.9
[11660] ### TinyGSM Compiled Module:  TinyGsmClientSIM800
[11844] ### Modem: SIMCOM SIM868
[11844] ### Modem: SIMCOM SIM868
Modem Info: SIM868 R14.18
Main processor board: ESP8266 
TinyGSM version: TinyGSM Version: 0.10.9 

Code Used:  MQTT Client Example Code

Trying to Connect to broker.hivemq.com and update ledstatus. connection is successful but the incoming data is not handled correctly and ledstatus does not update.

This code worked fine for many weeks until 3 days ago. Tried deleting library and installing various versions through IDE but to no avail.

 Problem Experienced-how to process following incoming data? :  ### Unhandled: +CIPRXGET: 1,0 

Serial monitor Output:

`
Modem Info: SIM868 R14.18
Waiting for network... success
Network connected
Connecting to unrestricted success
GPRS connected
=== MQTT NOT CONNECTED ===
Connecting to broker.hivemq.com success
[36288] ### Unhandled: +CIPR
[36303] ### Unhandled: XGET: 1,
[36319] ### Unhandled: 0
[65846] ### Unhandled: +CIPRX
[65862] ### Unhandled: GET: 1,
[65877] ### Unhandled: 0
[80929] ### Unhandled: +CIPRX
[80945] ### Unhandled: GET: 1,0
[95906] ### Unhandled: +CIPRX
[95922] ### Unhandled: GET: 1,0
[111126] ### Unhandled: +CIPRX
[111142] ### Unhandled: GET: 1,
[171063] ### Unhandled: +CIPR
[171078] ### Unhandled: XGET: 1,
[171094] ### Unhandled: 0
`

imamminer, 2020-11-16:
Hi

Could someone possibly assist with this?
Having a difficult time trying to resolve.

SRGDamia1, 2020-11-18:
Try using the ```TINY_GSM_YIELD```.

imamminer, 2020-11-23:
Hi 

> Try using the `TINY_GSM_YIELD`.

Hi

Thanks for this suggestion. It appears to have resolved the issue!

regards

SRGDamia1, 2020-11-23:
Great!

closed 2020-11-23 by SRGDamia1

imamminer, 2020-12-06:
Hi

is it possible to re open this thread as the solution seems intermittent?