vshymanskyy/TinyGSM #463: Does TinyGSM support Client Certificate Authentication for secure MQTT connections?  If so, how?

scubachristopher, 2020-11-15:
Hi all,

[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[X] Question or request for help

### What are you working with?

Modem: u-blox Model: SARA-R410M-02B Revision: L0.0.00.00.05.08
Main processor board: ESPDuino-32 (esp32dev compatible)
TinyGSM version: latest (0.10.9)
Code:  Current code is:

```
// --------------------------------------------------------------------
// Cellular functions
//
#include "includes.h"

#define SERIAL_CELL           Serial2

#define RX                             12
#define TX                             13
#define POWER_ON_PIN      16
#define RESET_PIN               27

TinyGsm modem(SERIAL_CELL);

PUBLIC bool lriSetupCellular();
PUBLIC TinyGsmClientSecure client(modem);
PUBLIC PubSubClient mqttCellularClient(CLOUD_DOMAIN, MQTT_PORT, client);

PUBLIC bool lriSetupCellular()
{

  SERIAL_CELL.begin(115200, SERIAL_8N1, RX, TX, false);

  pinMode(POWER_ON_PIN, OUTPUT);
  digitalWrite(POWER_ON_PIN, LOW);
  delay(100);
  digitalWrite(POWER_ON_PIN, HIGH);

  if (!modem.restart()) {
    lriSendLog(EMERG, "Cannot restart cellular modem\n");
    lriSerialPrintf(CODESEC_CELLULAR, "Cannot restart cellular modem\n");
    return false;
  }
  if (!modem.waitForNetwork()) {
    lriSendLog(EMERG, "Modem up, but unable to connect to a network\n");
    lriSerialPrintf(CODESEC_CELLULAR, "Modem up, but unable to connect to a network\n");
    return false;
  }

  if (!modem.gprsConnect("hologram", "", "")) {
    lriSendLog(EMERG, "Modem up, but unable to establish a data connection\n");
    lriSerialPrintf(CODESEC_CELLULAR, "Modem up, but unable to establish a data connection\n");
    return false;
  }
  lriSerialPrintf(CODESEC_DEFAULT, "Cellular initialized: %s\n", modem.getModemInfo().c_str());
  return true;
}
```

### Scenario, steps to reproduce

This function successfully gets a gprs data connection!  Yay!  I have my ca.crt, client.crt and client.key in SPIFFS, and I successfully connect with client certificate authentication to the secure endpoint via Wifi.  The relevant example snippet I use to connect via Wifi is:

```
...
  lri_ca_crt     = lriLoadSPIFFSIntoMem("/ca.crt");
  lri_client_crt = lriLoadSPIFFSIntoMem("/client.crt");
  lri_client_key = lriLoadSPIFFSIntoMem("/client.key");

  secureWifiClient.setCACert((const char *) lri_ca_crt);
  secureWifiClient.setCertificate((const char *) lri_client_crt);
  secureWifiClient.setPrivateKey((const char *) lri_client_key);

  mqttClient.setServer(CLOUD_DOMAIN, 8883);
  mqttClient.setBufferSize(MQTT_MAX_MESSAGE);
  mqttClient.setCallback(_mqttCallback);
...
```

**I'd like to understand how to set up the ca / client cert / key for a secure MQTT connection via TinyGSM.**

### Expected result

All code behaves as expected, my challenge is setting the certs and key for secure client auth with cert.

Thanks in advance 👍

SRGDamia1, 2020-11-16:
Those functions don't exist in TinyGSM.  There is an example showing how to set the [certificate for a SIM800](https://github.com/vshymanskyy/TinyGSM/tree/master/examples/more/SIM800_SslSetCert), but you will have to use the SSL AT command manual for the SARA R4 to check what the equivalent commands are.  Once you switch the module from the built in certificate to your own certificate, I think the same code should work and use the new certificate.

Sorry!

scubachristopher, 2020-11-16:
@SRGDamia1 -- No apologies!  Appreciate knowing what it can and can't do, and the link to the example is much appreciated.

Thanks man :)

closed 2020-11-16 by scubachristopher

roysG, 2021-10-31:
Hi,
Can you tell us how to add self sign certificate with mqtt when using with sin7070g module?
Thanks!