vshymanskyy/TinyGSM #464: How do you handle incoming calls?

bokolob, 2020-11-19:
Hello.

I found it neither in code nor examples nor docs...

[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: 
Main processor board: 
TinyGSM version: 
Code: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### Debug and AT command log

SRGDamia1, 2020-11-19:
We don't.

closed 2020-11-19 by SRGDamia1

bokolob, 2020-11-19:
But is there a way to do it with your library?

SRGDamia1, 2020-11-20:
There's a ```callAnswer()``` and a ```callHangup()``` function implemented for some of the modules, but there is not any support at this time for listening for an incoming call so you would have to have some other way of knowing there's an call coming (ie, a pin interrupt) and handle everything inside the call outside of this library.