vshymanskyy/TinyGSM #465: HTTPS problem with tinyGSM Esp32-sim800l

ncastellonb, 2020-11-25:
Hi! im new using gsm conection and I have a problem connection this to mi server with httpS. 

`/**************************************************************
 *
 * This sketch connects to a website and downloads a page.
 * It can be used to perform HTTP/RESTful API calls.
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 **************************************************************/
// Select your modem:
#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH
// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1
// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3); // RX, TX
// Increase RX buffer to capture the entire response
// Chips without internal buffering (A6/A7, ESP8266, M590)
// need enough space in the buffer for the entire response
// else data will be lost (and the http library will fail).
#define TINY_GSM_RX_BUFFER   1024 

// See all AT commands, if wanted
// #define DUMP_AT_COMMANDS
// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon
// Range to attempt to autobaud
#define GSM_AUTOBAUD_MIN 9600
#define GSM_AUTOBAUD_MAX 115200
// Add a reception delay - may be needed for a fast processor at a slow baud rate
// #define TINY_GSM_YIELD() { delay(2); }
// Uncomment this if you want to use SSL
// #define USE_SSL
// Define how you're planning to connect to the internet
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
// set GSM PIN, if any
#define GSM_PIN ""
// Your GPRS credentials, if any

#define IP5306_ADDR          0x75
#define IP5306_REG_SYS_CTL0  0x00

const char apn[]  = "imovil.entelpcs.cl";
const char gprsUser[] = "entelpcs";
const char gprsPass[] = "entelpcs";
// Your WiFi connection credentials, if applicable
const char wifiSSID[]  = "YourSSID";
const char wifiPass[] = "YourWiFiPass";
// Server details
const char server[] = "api.citylink.cl";
const char resource[] = "/metrics";
#include <Wire.h>
#include <TinyGsmClient.h>
// Just in case someone defined the wrong thing..
#if TINY_GSM_USE_GPRS && not defined TINY_GSM_MODEM_HAS_GPRS
#undef TINY_GSM_USE_GPRS
#undef TINY_GSM_USE_WIFI
#define TINY_GSM_USE_GPRS false
#define TINY_GSM_USE_WIFI true
#endif
#if TINY_GSM_USE_WIFI && not defined TINY_GSM_MODEM_HAS_WIFI
#undef TINY_GSM_USE_GPRS
#undef TINY_GSM_USE_WIFI
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
#endif

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif
#ifdef USE_SSL
  TinyGsmClientSecure client(modem);
const int  port = 443;
#else
  TinyGsmClientSecure client(modem);
const int  port = 80;
#endif

#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22

TwoWire I2CPower = TwoWire(0);

void setup() {
// Set console baud rate
  SerialMon.begin(115200);
  I2CPower.begin(I2C_SDA, I2C_SCL, 400000);
delay(10);
// !!!!!!!!!!!
// Set your reset, enable, power pins here
// !!!!!!!!!!!
  SerialMon.println("Wait...");
        // Set modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);
  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);
// Set GSM module baud rate
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
// SerialAT.begin(9600);
delay(6000);
// Restart takes quite some time
// To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  modem.restart();
// modem.init();
  String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem Info: ");
  SerialMon.println(modemInfo);
#if TINY_GSM_USE_GPRS
// Unlock your SIM card with a PIN if needed
if ( GSM_PIN && modem.getSimStatus() != 3 ) {
    modem.simUnlock(GSM_PIN);
  }
#endif
}
void loop() {
#if TINY_GSM_USE_WIFI
// Wifi connection parameters must be set before waiting for the network
  SerialMon.print(F("Setting SSID/password..."));
if (!modem.networkConnect(wifiSSID, wifiPass)) {
    SerialMon.println(" fail");
delay(10000);
return;
  }
  SerialMon.println(" success");
#endif
#if TINY_GSM_USE_GPRS && defined TINY_GSM_MODEM_XBEE
// The XBee must run the gprsConnect function BEFORE waiting for network!
  modem.gprsConnect(apn, gprsUser, gprsPass);
#endif
  SerialMon.print("Waiting for network...");
if (!modem.waitForNetwork()) {
    SerialMon.println(" fail");
delay(10000);
return;
  }
  SerialMon.println(" success");
if (modem.isNetworkConnected()) {
    SerialMon.println("Network connected");
  }
#if TINY_GSM_USE_GPRS
// GPRS connection parameters are usually set after network registration
    SerialMon.print(F("Connecting to "));
    SerialMon.print(apn);
if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
      SerialMon.println(" fail");
delay(10000);
return;
    }
    SerialMon.println(" success");
if (modem.isGprsConnected()) {
    SerialMon.println("GPRS connected");
  }
#endif
  SerialMon.print("Connecting to ");
  SerialMon.println(server);
if (!client.connect(server, port)) {
    SerialMon.println(" fail");
delay(10000);
return;
  }
  SerialMon.println(" success");
// Make a HTTP GET request:
  SerialMon.println("Performing HTTP GET request...");
  client.print(String("GET ") + resource + " HTTP/1.1\r\n");
  client.print(String("Host: ") + server + "\r\n");
  client.print("Connection: close\r\n\r\n");
  client.println();
uint32_t timeout = millis();
while (client.connected() && millis() - timeout < 10000L) {
// Print available data
while (client.available()) {
char c = client.read();
      SerialMon.print(c);
      timeout = millis();
    }
  }
  SerialMon.println();
// Shutdown
  client.stop();
  SerialMon.println(F("Server disconnected"));
#if TINY_GSM_USE_WIFI
    modem.networkDisconnect();
    SerialMon.println(F("WiFi disconnected"));
#endif
#if TINY_GSM_USE_GPRS
    modem.gprsDisconnect();
    SerialMon.println(F("GPRS disconnected"));
#endif
// Do nothing forevermore
while (true) {
delay(1000);
  }
}`

Im us gin the example of TinyGsm with changes to run the esp32 module. Please if any know what im doing wrong I will be very grateful.

Regards

SRGDamia1, 2020-11-30:
Unless you have it defined in a build flag, you need to un-comment (remove the two slashes before) `#define USE_SSL` in your code.

Not all versions or firmwares of the SIM800 support SSL/HTTPS.  It's very hard to say if yours will.  Different people have had successes and failures with modules that appear to be identical.

Un-comment `#define DUMP_AT_COMMANDS` and watch the output as the program runs for the command `AT+CIPSSL=1`.  If you see `ERROR` after that command, you're module definitely doesn't support SSL and there's no reason to keep trying.  If you see `OK` there but the module doesn't connect, you can try adding a `#define TINY_GSM_SSL_CLIENT_AUTHENTICATION` to the top of your code in case your site requires it.  If that still doesn't work, you  might be able to get it to work by uploading your own certificate, if you have one. (Look in the examples folder for an example of this.)  If it still doesn't work after that, you're probably just out of luck.

Toumal, 2021-02-05:
My device responds OK to AT+CIPSSL=1 but there's no joy connecting to any https server.
Enabling SSL client authentication does not help (why would it?)
Attempts to disable SSL cert checks also fail. Does anyone have a working example of a https connection using TinyGSM?

SRGDamia1, 2021-02-05:
The HTTPS example in this library *does* work.  The problem is the SIM800.  As I said before, and as numerous issues on this subject attest, people have had very mixed results with getting SSL to work at all on the SIM800, regardless of whether the module itself reports that it will support it.  If you're module is really going to support it, then the already existing example and suggestions I gave are all you need.  If those don't work, your module is probably never going to support SSL natively.

Other people have had success using a TinyGSM insecure client with an alternate SSL library providing the security.  See this issue:  https://github.com/vshymanskyy/TinyGSM/issues/406

closed 2021-02-05 by SRGDamia1