vshymanskyy/TinyGSM #466: Web client can't connect to the sample webside (AT+CIPCLOSE=0,1 +CME ERROR: operation not allowed).

dawidwekwejt, 2020-11-27:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
[ ] Bug or problem compiling the library
[X] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM800L, R14.18
Main processor board: Arduino Uno
TinyGSM version: 0.10.9

```
if (!modem.isNetworkConnected())
{
	SerialMon.print("Waiting for network...");
	if (!modem.waitForNetwork())
	{
		SerialMon.println(" fail");
		delay(10000);
		return;
	}
	SerialMon.println(" success");
}

if (!modem.isGprsConnected())
{
	SerialMon.print(F("Connecting to "));
	SerialMon.print(apn);
	if (!modem.gprsConnect(apn, gprsUser, gprsPass))
	{
		SerialMon.println(" fail");
		delay(10000);
		return;
	}
	SerialMon.println(" success");
}

if (!client.connect(server, port))
{
	SerialMon.print("Connecting to ");
	SerialMon.println(server);

	SerialMon.println(" fail");
	delay(10000);
	return;
}
SerialMon.println(" success");
```

### Scenario, steps to reproduce
I try to make simple connection to sample website. I'm able to connect to the GSM network and open GPRS connection. I don't no why after I had opened GPRS connection the application sent to the module AT+CIPCLOSE=0,1.

### Expected result
I want my module to be connected with webside.

### Actual result
### Debug and AT command log
```
Wait...
Initializing modem...
[6009] ### TinyGSM Version: 0.10.9
[6009] ### TinyGSM Compiled Module:  TinyGsmClientSIM800
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+GMM

SIMCOM_SIM800L

OK
[6175] ### Modem: SIMCOM SIM800L
[6175] ### Modem: SIMCOM SIM800L
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CPIN: READY

OK
ATI

SIM800 R14.18

OK
Modem Info: SIM800 R14.18
AT+CPIN?

+CPIN: READY

OK
AT+CREG?

+CREG: 0,1

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

100.83.70.85

OK
AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=0

OK
AT+CIPSTART=0,"TCP","vsh.pp.ua",80

OK

0, CONNECT FAIL
Connecting to vsh.pp.ua
 fail
AT+CREG?

+CREG: 0,1

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

100.83.70.85

OK
AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=0

OK
AT+CIPSTART=0,"TCP","vsh.pp.ua",80

OK

0, CONNECT FAIL
Connecting to vsh.pp.ua
 fail
AT+CREG?

+CREG: 0,1

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

100.83.70.85

OK
AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=0

OK
AT+CIPSTART=0,"TCP","vsh.pp.ua",80

OK

0, CONNECT FAIL
Connecting to vsh.pp.ua
 fail
AT+CREG?

+CREG: 0,1

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

100.83.70.85

OK
AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=0

OK
AT+CIPSTART=0,"TCP","vsh.pp.ua",80

OK

0, CONNECT FAIL
Connecting to vsh.pp.ua
 fail
```

gsvitak, 2020-11-28:
@dawidwekwejt I have found this occurs when you do not add a `delay(10000)` in the setup function to allow the modem to boot up properly. The sample does not handle the specific setup of you device and can cause issues because the modem is not completely ready/initialized.

```
  SerialMon.println("Wait...");
  SerialAT.begin(9600, SERIAL_8N1, MODEM_RX, MODEM_TX);
  modem.setBaud(9600);
  modem.begin();
  delay(10000);
```

ClemensGruber, 2021-09-06:
I have this error message also with an TTGO T-Call but it occures 

```
OK
AT+CIFSR;E0

100.125.18.1

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 OK
Connecting to example.com
AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=0

OK
AT+CIPSTART=0,"TCP","example.com",80

OK

0, CONNECT OK
 OK

```

I do not call a HTTPS website so I wonder why a SSL statement follows the error, or has this nothing to do with the error? 

I get the `AT+CIPCLOSE=0,1 / +CME ERROR: operation not allowed` also although a TCP connection is established afterwards.

SRGDamia1, 2021-11-24:
That error on the close is just because there wasn't something open.  It can just be ignored.

closed 2021-11-24 by SRGDamia1

attahiruj, 2022-02-10:
> I have this error message also with an TTGO T-Call but it occures
>
> ```
> OK
> AT+CIFSR;E0
>
> 100.125.18.1
>
> OK
> AT+CDNSCFG="8.8.8.8","8.8.4.4"
>
> OK
>  OK
> Connecting to example.com
> AT+CIPCLOSE=0,1
>
> +CME ERROR: [...]

were you ever able to solve this issue? I'm experiencing the same challenge.

sarkaramitabh300, 2022-02-13:
Initializing modem...
AT

OK
AT&W

OK
AT+CFUN=0

+SAPBR 1: DEACT

+PDP: DEACT

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
[15156] ### TinyGSM Version: 0.11.5
[15156] ### TinyGSM Compiled Module:  TinyGsmClientSIM800
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+GMM

SIMCOM_SIM808

OK
[15176] ### Modem: SIMCOM SIM808
[15176] ### Modem: SIMCOM SIM808
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CME ERROR: SIM busy
AT+CPIN?

+CPIN: READY

OK
ATI

SIM808 R14.18

OK
Modem Info: SIM808 R14.18
AT+CPIN?

+CPIN: READY

OK
Waiting for network...AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CRE⸮b⸮

OK
[17991] ### Unhandled: +CREb

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,1

OK
 success
AT+CREG?

+CREG: 0,1

OK
Network connected
Connecting to airtelgprs.comAT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","airtelgprs.com"

OK
AT+CGDCONT=1,"IP","airtelgprs.com"

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"100.83.83.249"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="airtelgprs.com","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

100.83.83.249

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 success
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

100.83.83.249

OK
GPRS connected
Performing HTTP GET request... AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=0

+CME ERROR: unknown
AT+CIPSTART=0,"TCP","http://97.74.85.82/cwcv/api/temperature",80

OK

0, CONNECT FAIL
Error code-1
failed to connect
Waiting for network...AT+CREG?

+CREG: 0,1

OK
 success
AT+CREG?

+CREG: 0,1

OK
Network connected
Connecting to airtelgprs.comAT+CIPSHUT

SHUT OK
AT+CGATT=0

+SAPBR 1: DEACT

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","airtelgprs.com"

OK
AT+CGDCONT=1,"IP","airtelgprs.com"

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"100.70.177.233"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="airtelgprs.com","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

100.70.177.233

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 success
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

100.70.177.233

OK
GPRS connected
Performing HTTP GET request... AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=0

+CME ERROR: unknown
AT+CIPSTART=0,"TCP","http://97.74.85.82/cwcv/api/temperature",80

OK

0, CONNECT FAIL
Error code-1
failed to connect

braysonjohn148, 2022-07-19:
Any solution?

MrCoopa, 2022-09-23:
Same Problem hier with Pushover:

15:53:32.835 -> AT+CSTT="pinternet.interkom.de","",""
15:53:32.835 -> 
15:53:32.835 -> OK
15:53:32.835 -> AT+CIICR
15:53:32.835 -> 
15:53:32.835 -> OK
15:53:32.835 -> AT+CIFSR;E0
15:53:32.835 -> 
15:53:32.835 -> 10.118.101.45
15:53:32.835 -> 
15:53:32.835 -> OK
15:53:32.835 -> AT+CDNSCFG="8.8.8.8","8.8.4.4"
15:53:32.877 -> 
15:53:32.877 -> OK
15:53:32.877 ->  OK
15:53:32.877 -> Connecting to api.pushover.net
15:53:32.877 -> AT+CIPCLOSE=0,1
15:53:32.877 -> 
15:53:32.877 -> ERROR
15:53:32.877 -> AT+CIPSSL=1
15:53:32.877 -> 
15:53:32.877 -> OK
15:53:32.877 -> AT+CIPSTART=0,"TCP","api.pushover.net",443
15:53:32.877 -> 
15:53:32.877 -> OK
15:53:45.739 -> 
15:53:45.739 -> 0, CONNECT FAIL
15:53:45.775 ->  fail

HynekBaran, 2023-01-29:
The same problem on  TTGO T-Call , any solution, please?

ilgarbenli, 2023-12-06:
any solution?