vshymanskyy/TinyGSM #469: MQTT with Sim 7000C NB-IoT Network

najabhaider96, 2020-12-08:
I'm trying to use MQTT on Sim 7000C with NB-IoT Network
I'm using this example code for trying to use MQTT with NB-IoT
Network configured but not connected with broker. 

// Select your modem:
//#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
#define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE

#include <TinyGsmClient.h>
#include <PubSubClient.h>

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

#include <SoftwareSerial.h>
SoftwareSerial SerialAT(5, 4); // RX, TX

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "apn";
const char user[] = "";
const char pass[] = "";

// MQTT details
const char* broker = "m21.cloudmqtt.com";

const char* topicLed = "GsmClientTest/led";
const char* topicInit = "GsmClientTest/init";
const char* topicLedStatus = "GsmClientTest/ledStatus";

TinyGsm modem(SerialAT);
TinyGsmClient gsmClient(modem);
PubSubClient mqtt(gsmClient);

#define LED_PIN 13
int ledStatus = LOW;

long lastReconnectAttempt = 0;

void setup() {
  pinMode(LED_PIN, OUTPUT);

  // Set console baud rate
  SerialMon.begin(9600);
  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(9600);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("\n\n\nInitializing modem...");
  modem.restart();

  String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem: ");
  SerialMon.println(modemInfo);

  // Unlock your SIM card with a PIN
  //modem.simUnlock("1234");

  Serial.print("Sim Operator : ");
  Serial.println(modem.getOperator());
  Serial.print("Signal Strength : ");
  Serial.println(modem.getSignalQuality());

  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork()) {
    SerialMon.println(" fail");
    while (true);
  }
  SerialMon.println(" OK");

  SerialMon.print("Connecting to ");
  SerialMon.print(apn);
  if (!modem.gprsConnect(apn, user, pass)) {
    SerialMon.println(" fail");
    while (true);
  }
  SerialMon.println(" OK");

  Serial.print("Local IP : ");
  Serial.println(modem.localIP());

  // MQTT Broker setup
  mqtt.setServer(broker, 19279);
  mqtt.setCallback(mqttCallback);
}

boolean mqttConnect() {
  SerialMon.print("Connecting to ");
  SerialMon.print(broker);

  // Connect to MQTT Broker
  //  boolean status = mqtt.connect("GsmClientTest");

  // Or, if you want to authenticate MQTT:
  boolean status = mqtt.connect("GsmClientName", "user", "password");

  if (status == false) {
    SerialMon.println(" fail");
    return false;
  }
  SerialMon.println(" success");
  mqtt.publish(topicInit, "GsmClientTest started");
  mqtt.subscribe(topicLed);
  return mqtt.connected();
}

void loop() {

  if (!mqtt.connected()) {
    SerialMon.println("=== MQTT NOT CONNECTED ===");
    // Reconnect every 10 seconds
    unsigned long t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) {
        lastReconnectAttempt = 0;
      }
    }
    delay(100);
    return;
  }

  mqtt.loop();
}

void mqttCallback(char* topic, byte* payload, unsigned int len) {
  SerialMon.print("Message arrived [");
  SerialMon.print(topic);
  SerialMon.print("]: ");
  SerialMon.write(payload, len);
  SerialMon.println();

  // Only proceed if incoming message's topic matches
  if (String(topic) == topicLed) {
    ledStatus = !ledStatus;
    digitalWrite(LED_PIN, ledStatus);
    mqtt.publish(topicLedStatus, ledStatus ? "1" : "0");
  }
}

**OUTPUT**

Waiting for network... OK
Connecting to apnname OK
Local IP : **.45.**.144
=== MQTT NOT CONNECTED ===
Connecting to m21.cloudmqtt.com fail
=== MQTT NOT CONNECTED ===
Connecting to m21.cloudmqtt.com fail
=== MQTT NOT CONNECTED ===

SRGDamia1, 2020-12-08:
I assume you have the real port/username/password in your code.

What's the status code that pubsubclient is giving you?  (Add a `Serial.print(status);` right after `boolean status = mqtt.connect("GsmClientName", "user", "password");`)  Look in the docs for the pubsubclient library to check the meaning of the status code. (https://pubsubclient.knolleary.net/api#state)  That should help you track down the issue.

Can you connect to other sites using the same module?  Do other examples (like the webclient example) work?

Otherwise, try posting your AT log.  There's not too much I can see without that.

SRGDamia1, 2020-12-08:
Depending on your SIM provider, you may have to set your preferred network band and mode: 

```cpp
setNetworkMode(uint8_t mode);
// 2 - Automatic
// 13 - GSMonly
// 38 - LTEonly
// 51 - GSMandLTEonly

setPreferredMode(uint8_t mode);
// 1 - CAT-M
// 2 - NB-Iot
// 3 - CAT-M and NB-IoT
```

Mark-Wills, 2021-03-09:
I notice you are trying to connect to m21.cloudmqtt.com. Are you sure you have access to the internet from your provider? Many NBIoT providers operate their networks in an APN/VPN which does not have access to the big, bad internet, unless a gateway is specifically installed. Try opening a http port 80 connection to google.com or something like that. If it works... then you have access to the internet. If not, you're operating in a closed APN (probably company provided SIM card where all the SIM cards are in the same network so they can talk to each other, but there is no external gateway. This is how the SIM cards that we use work by design - our MQTT box is within our APN IP address range).

Regards

Mark