vshymanskyy/TinyGSM #470: sendSMS() fails repeatedly

Elmi77, 2020-12-08:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[X] Bug or issue with library functionality (ie, sending data over TCP/IP)
[X] Question or request for help

### What are you working with?

Modem: SIM700G 
Main processor board: LilyGO SIM7000G 
TinyGSM version: 0.10.9
Code: modem.sendSMS(String(notifySMS[0]),"blatest ")

### Scenario, steps to reproduce
I try to send a SMS. GSM is connected, the internet functionality is fine, so I'm successfully registered at the network.

### Expected result
The SMS should be sent

### Actual result
sendSMS() returns with "false".

### Debug and AT command log

AT+CMGF=1

OK
AT+CSCS="GSM"

OK
AT+CMGS="+4915000000000"

>blatest 
+CMS ERROR:AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

I'm of course using a valid phone number, +4915000000000 is only a placeholder.

Any idea why my SMS is not sent?

SRGDamia1, 2020-12-08:
Nope.  :( I've never even tested the SMS functionality; I just followed what the AT command manual said.  It *looks like* it should have worked.

It didn't even give you the error code.  That's annoying.  Did you try `#define TINY_GSM_DEBUG Serial` to see if you can at least get it to give you the verbose error code?

Sorry!

Elmi77, 2020-12-09:
The output above is with debug output and TINY_GSM_DEBUG set to serial interface enabled - that's all I get there.

SRGDamia1, 2020-12-09:
Are you sure your SIM and plan supports texting?  The IoT plans we usually get from Hologram don't include it.

I'm sorry, I don't really have any other guesses about what the problem might be.