vshymanskyy/TinyGSM #471: TTGO ESP32 T-Call, SIM800L,

ykolerov, 2020-12-09:
[x] I have read the Troubleshooting section of the ReadMe

[ ] Request to support a new module
[ ] Bug or problem compiling the library
[x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x] Question or request for help

Modem: SIM800L R14.18
Main processor board: ESP32 TTGO T-Call v1.3
TinyGSM version: 0.10.9
Code:

`#define TINY_GSM_MODEM_SIM800

#define SerialMon Serial

#define SerialAT Serial1

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

const char apn[] = "internet.mts.ru";
const char gprsUser[] = "";
const char gprsPass[] = "";

// real IP goes here
const char* broker = "***.***.***.***";

#include <TinyGsmClient.h>
#include <PubSubClient.h>

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

TinyGsmClient client(modem);
PubSubClient mqtt(client);

uint32_t lastReconnectAttempt = 0;

boolean mqttConnect() {
  SerialMon.print("Connecting to ");
  SerialMon.print(broker);

  boolean status = mqtt.connect("GsmClientName");

  if (status == false) {
    SerialMon.println(" fail");
    return false;
  }
  SerialMon.println(" success");
  
  return mqtt.connected();
}

void setup() {
  SerialMon.begin(115200);
  delay(10);

  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(23, OUTPUT);

[... 457 lines cut ...]
AT+CIPSTATUS=0
+CIPSTATUS: 0,0,"TCP","***.***.***.***","1883","CONNECTED"
OK
AT+CIPRXGET=4,0
+CIPRXGET: 4,0,0
OK
AT+CIPSTATUS=0
+CIPSTATUS: 0,0,"TCP","***.***.***.***","1883","CONNECTED"
OK
AT+CIPRXGET=4,0
+CIPRXGET: 4,0,0
OK
AT+CIPSTATUS=0
+CIPSTATUS: 0,0,"TCP","***.***.***.***","1883","CONNECTED"
OK
AT+CIPCLOSE=0,1
0, CLOSE OK
 fail
AT+CIPRXGET=4,0
+CIPRXGET: 4,0,0
OK
AT+CIPSTATUS=0
+CIPSTATUS: 0,0,"TCP","***.***.***.***","1883","CLOSED"
OK
=== MQTT NOT CONNECTED ===

SRGDamia1, 2020-12-09:
Well, in your `mqttConnect()` function in your code you're trying to connect to your MQTT broker using the client named "GsmClientName", so that's why you're seeing that line.  
`boolean status = mqtt.connect("GsmClientName");`
The weird symbols you get are because the MQTT protocol doesn't use all normally printed characters, so they're expected.

It looks to me from your code like you did successfully connect to the broker's IP and sent the MQTT connection request but the server never responded.  Could you see on the server side that you weren't connected?  Does your broker require a different client name/user/password?  

All this library does is provide the TCP connection for the MQTT communication.  The MQTT commands themselves are all on PubSubClient.

ykolerov, 2020-12-17:
Thank you. The root cause was in the cell operator. I switched to another operator, everything worked.

closed 2020-12-17 by ykolerov