vshymanskyy/TinyGSM #474: When will be support SSL/TLS (HTTPS) for SIM7000?

EchoSystemMark, 2020-12-25:
In the TinyGSM library, SSL/TLS support for SIM 7000 is listed as TODO. 
( //typedef TinyGsmSim7000::GsmClientSecureSim7000 TinyGsmClientSecure; TODO! ). 
Everyone has been waiting for this for a long time, when is it planned to be implemented?

SRGDamia1, 2020-12-26:
I'm sorry, it's not a feature that any projects I'm on need, so I don't have any plans to do it. There have been several other issues where people have mentioned working on it. If someone submits a solid pull request, I will accept it.

FStefanni, 2021-03-11:
Hi,

seems duplicate of #437 (I know since I am also interested in this feature :-) ).
@EchoSystemMark I suggest to try some of the forks there proposed (this is what I am going to do).
I believe this issue could be closed as duplicate.

Regards.

PR #507 (2021-04-28): SIM7000: TLS support plus other minor improvements
Hi,

I have based this pr on the great code by @marmotton and @javipelopi.
I have just:
* Packaged their code, aligning it also to the newer master version
* Performed some minor cleanup
* Performed a good debug
* Supported factory reset of the modem

Now it works perfectly with MQTT both with and without TLS.

Fixes #437
Fixes #474 
Fixes #495

Regards
closed 2021-04-28 by SRGDamia1