vshymanskyy/TinyGSM #480: i cant comunication with UART in arduino mega using SIM7000E

murtadhodaf, 2021-01-11:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: WAVESHARE SIM 7000E
Main processor board: ARDUINO MEGA 2560
TinyGSM version:(0.10.09)
Code: //#include <Arduino_JSON.h>
```
#include <SoftwareSerial.h> //koneksi ke modem

#define TINY_GSM_MODEM_SIM7000
#define TINY_GSM_DEBUG SerialMon
#define BAUD 9600
#define SerialMon Serial
const byte rxpin = 8;
const byte txpin = 10;

SoftwareSerial SerialAT(rxpin, txpin); // RX, TX ke modem

const char apn[]  = "nb1internet"; //untuk gprs gunakan "internet"
const char user[] = "";
const char pass[] = "";
byte randomValue = random(0, 37);
const char *mqtt_server = "mqtt.antares.id"; //server antares
const char *mqtt_user = "";
const char *mqtt_pass = "";
const int mqtt_port = 1883;
const char *mqtt_client_name = "KEKW" + randomValue; // random

//dibawah ini sesuaikan dengan akun atares kamu
#define TOPIKPUB "/oneM2M/req/5a185cedc6ad749b:edb0b82be780ba6d/antares-cse/json"
#define accessKey "5a185cedc6ad749b:edb0b82be780ba6d"
#define projDev "pHnEC/ECSENSOR" 
//#define projdev2

// library modem dan mqtt 
#include <TinyGsmClient.h>
#include <PubSubClient.h>

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else

TinyGsm modem(SerialAT);
#endif
TinyGsmClient client(modem);
PubSubClient mqtt(client);

long lastReconnectAttempt = 0;
int val1;
int val2;

void setup()
{
  // Set debug console dan modem
  SerialMon.begin(115200);
  delay(10);
  SerialAT.begin(115200);
  delay(3000);

  String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem: ");
  SerialMon.println(modemInfo);

  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork(240000L)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  if (modem.isNetworkConnected()) {
    SerialMon.println("Network connected");
  }

  SerialMon.print(F("Connecting to "));
  SerialMon.print(apn);
    if (!modem.gprsConnect(apn, user, pass)) {
    SerialMon.println(" fail");
      delay(10000);
      return;
  }
  
    SerialMon.println(" OK");

   // MQTT Broker setup
  mqtt.setBufferSize(512);
  mqtt.setServer(mqtt_server, mqtt_port);
}

boolean mqttConnect() {
  SerialMon.print("Connecting to ");
  SerialMon.print(mqtt_server);

  // koneksi ke MQTT Broker
  boolean status = mqtt.connect(mqtt_client_name,mqtt_user, mqtt_pass);

  if (status == false) {
    SerialMon.println(" fail");
    return false;
  }
  SerialMon.println(" OK");

  return mqtt.connected();
}

void loop()
{

  //cek jika koneksi mqtt gagal maka ulang lagi
   if (!mqtt.connected()) {
    SerialMon.println("=== MQTT NOT CONNECTED ===");
    // Reconnect every 10 seconds
    unsigned long t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) {
        lastReconnectAttempt = 0;
      }
    }
    delay(100);
    return;
  }

  int val1 = random(0-450);  //random1
  int val2 = random(0-350);  //random2
                
  String pubString ;  
 //JSON yang panjang untuk kirim data ADC ke antares
    pubString += F("{");
    pubString += F("\"m2m:rqp\": {");
    pubString += F("\"fr\": \"");
    pubString += String(accessKey) ;
    pubString += F("\",");
    pubString += F("\"to\": \"/antares-cse/antares-id/");
    pubString += String(projDev);
    pubString += F("\",");
    pubString += F("\"op\": 1,");
    pubString += F("\"rqi\": 123456,");
    pubString += F("\"pc\": {");
    pubString += F("\"m2m:cin\": {");
    pubString += F("\"cnf\": \"message\",");
    pubString += F("\"con\": \"{\\\"level 1\\\":");
    pubString += String(val1);
    pubString += F("\",");
//    pubString += F("\\\"level 2\\\":");
//    pubString += String(val2);
//    pubString += F("}\""); 
    pubString += F("}");
    pubString += F("},");
    pubString += F("\"ty\": 4");
    pubString += F("}");
    pubString += F("}");
  
  // kirim atau publish ke antares
  SerialMon.println(pubString);
  char message_buff[pubString.length() + 1];
  pubString.toCharArray(message_buff, pubString.length() + 1);
  mqtt.publish(TOPIKPUB, message_buff);

delay(5000);
}
```

### Scenario, steps to reproduce
7 days ago the modem still can connect on my UART but now its not   working IDK btw  the code its still same like i tried 7 days ago 

### Actual result
this my result, its cant to read the modem im waiting so long 
> 02:40:39.143 -> Modem: 
> 02:40:39.143 -> Waiting for network...
> 

thx ffor helping  me

SRGDamia1, 2021-01-11:
If the same code worked before, it shouldn't be a code issue.

Are you sure you have consistent power, service, and money on your plan/account?

murtadhodaf, 2021-01-11:
Im using 5V in arduino mega, and service i checked in PUTTY its working