vshymanskyy/TinyGSM #482: Unicode SMS issue

kritonpc, 2021-01-15:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x] Question or request for help

### What are you working with?

Modem: SIM800L -->
Main processor board: 
TinyGSM version: 
Code: 

### Scenario, steps to reproduce
I'm trying to send an SMS with Greek characters, and the message is passed as a variable, as it is not constant, using variable smsMessage. I tried using the UTF16 method, but I can't figure out how to convert a string to the desired value. When I pass the message directly to the function using modem.sendSMS_UTF16(SMS_TARGET, u"τεστ", 4), it works. So the problem is the way I will convert a string with greek characters to a format supported by this method

### Expected result

### Actual result

marutichintan, 2021-03-23:
i have the same issue, i cannnot cast string to char16_t.

chaiyapojp, 2023-12-15:
I get chinese characters  after my message who can help? pleaseeeeeeeee