vshymanskyy/TinyGSM #485: MKR1400 can't reconnect after client disconnects

tuloski, 2021-01-26:
Board: MKR1400 (I followed the special setup https://github.com/vshymanskyy/TinyGSM/wiki/Board-configuration)
Modem: SARA-U201

Trying to send http request (GET, POST) with ArduinoHttpClient passing TinyGsmClientSecure as client.
The first requests go through, but when the client disconnects it's not able to reconnect again. I don't know if it's modem fault or client fault
Below the AT dump AFTER the disconnection with some extra to understand what it was doing:

*** making GET request ***
AT+USORD=3,0
+USORD: 3,0
OK
AT+USOCTL=3,10
+USOCTL: 3,10,7
OK
AT+USOCR=6
+USOCR: 4
OK
AT+USOSEC=4,1
OK
AT+USOCO=4,"httpbin.org",443
ERROR
GET error: -1
*** Made GET request ***
+UUSOCL: 4
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
*** Check connections ***
Client gsm not connected
AT+CGATT?
+CGATT: 1
OK
AT+UPSND=0,0
+UPSND: 0,0,"10.175.113.24"
OK
AT+CGREG?
+CGREG: 0,1
OK
*** CHECKED connections ***
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10
ERROR
AT+USORD=4,0
ERROR
AT+USOCTL=4,10

SRGDamia1, 2021-01-26:
Huh.  It looks like the new socket is being closed as soon as it's opened.  (The `+UUSOCL: 4` is the modem telling you the socket has been closed.)  Once the socket is closed, you can't do anything with that socket; you have to intentionally reopen a new one.  You're getting all those `ERROR` messages because you're trying to ask the state of a closed socket.  Try adjusting your code to check the socket state and re-open a new socket if needed before sending or checking for data.

If the socket is always being closed immediately after being opened, you can try enabling "keepAlive" in your local copy of the library to see if that helps.  There's commented-out code for it on [line 631 of TinyGsmClientUBLOX.h](https://github.com/vshymanskyy/TinyGSM/blob/31bb5cbeedc39de3d510bd2527edc4570f392d58/src/TinyGsmClientUBLOX.h#L631).

tuloski, 2021-01-27:
But in the periodical check modem.isGprsConnected() and modem.isNetworkConnected() both return true.
Only client.connected() returns false, so it seems related to the client, not to socket or modem.

SRGDamia1, 2021-01-28:
@tuloski - Sorry, confusing terminology.  In this library client = socket.  If `client.connected()` is false and you try to send data (ie, `client.print()`) the modem will explode with errors because there's no where for the data to go.

- `modem.isNetworkConnected()` returns true if the modem has successfully registered with a cellular tower
- `modem.isGprsConnected()` returns true if the modem has a internet/data connection available.  This can only happen *after* successfully registering.  This does **not** mean there are connections in use, just that they're possible.
- `client.connected()` returns true if there is a TCP connection (socket, client) open to a specific IP address/server.  This can happen only *after* connecting to GPRS.

To send data to a specific website/server/anything **all three** of the above functions must return true.  If any of them return false, you have to step back and fix the lowest level of what's broken and then re-open every higher level.
- if `client.connected()` returns false, you need to open a new socket (aka client/TCP connection) with `client.connect(server, port)`
- if `modem.isGprsConnected()` returns false, you need to reconnect to GPRS with `modem.gprsConnect(apn, gprsUser, gprsPass)`
- if `modem.isNetworkConnected()` returns false, you need to go all the way back to `modem.waitForNetwork()` - and you may need to power cycle.

tuloski, 2021-02-01:
@SRGDamia1 I see, but since I pass the client to httpClient, I have no power to reconnect (I can manually do it but it's wrong). IMO it should be done by the client itself or the httpClient. 
Actually I think it is trying to reconnect but it's failing somewhere, but I'm no good at interpreting AT messages.

SRGDamia1, 2021-02-02:
In the AT commands you posted, there's only one attempt to connect at the very beginning.

It can be very convention, but I'm not a fan of the Arduino httpClient library.  I would recommend you write out the HTTP yourself.