vshymanskyy/TinyGSM #490: Modem.init() failure on Liligo TTGO GPRS SIM800L

maxime400, 2021-02-03:
## What type of issues is this?
[ ] Question or request for help

### What are you working with?

Modem:  Liligo TTGO GPRS SIM800L
Main processor board: ESP32
TinyGSM version: (0.10.9) 
Code: 

#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial2

// or Software Serial on Uno, Nano
// #include <SoftwareSerial.h>
// SoftwareSerial SerialAT(2, 3);  // RX, TX

// See all AT commands, if wanted
// #define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// Range to attempt to autobaud
#define GSM_AUTOBAUD_MIN 9600
#define GSM_AUTOBAUD_MAX 57600

/*
 * Tests enabled
 */
#define TINY_GSM_TEST_GPRS true
#define TINY_GSM_TEST_WIFI false
#define TINY_GSM_TEST_TCP true
#define TINY_GSM_TEST_SSL true
// #define TINY_GSM_TEST_CALL true
// #define TINY_GSM_TEST_SMS true
// #define TINY_GSM_TEST_USSD true
#define TINY_GSM_TEST_BATTERY true
#define TINY_GSM_TEST_TEMPERATURE true
#define TINY_GSM_TEST_GSM_LOCATION true
#define TINY_GSM_TEST_TIME true
#define TINY_GSM_TEST_GPS false
// powerdown modem after tests
#define TINY_GSM_POWERDOWN true

// set GSM PIN, if any
#define GSM_PIN "7213"

// Set phone numbers, if you want to test SMS and Calls
// #define SMS_TARGET  "+380xxxxxxxxx"
// #define CALL_TARGET "+380xxxxxxxxx"

// Your GPRS credentials, if any
const char apn[] = "internet.proximus.be";
// const char apn[] = "ibasis.iot";
const char gprsUser[] = "";
const char gprsPass[] = "";

// Your WiFi connection credentials, if applicable
const char wifiSSID[] = "YourSSID";
const char wifiPass[] = "YourWiFiPass";

// Server details to test TCP/SSL
const char server[] = "vsh.pp.ua";
const char resource[] = "/TinyGSM/logo.txt";

#include <TinyGsmClient.h>

#if TINY_GSM_TEST_GPRS && not defined TINY_GSM_MODEM_HAS_GPRS
#undef TINY_GSM_TEST_GPRS
#undef TINY_GSM_TEST_WIFI
#define TINY_GSM_TEST_GPRS false
#define TINY_GSM_TEST_WIFI true
#endif
#if TINY_GSM_TEST_WIFI && not defined TINY_GSM_MODEM_HAS_WIFI
#undef TINY_GSM_USE_GPRS
#undef TINY_GSM_USE_WIFI
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
#endif

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

void setup() {
  // Set console baud rate
  SerialMon.begin(115200);
  delay(10);

  // !!!!!!!!!!!
  // Set your reset, enable, power pins here
  pinMode(A5, OUTPUT);
  DBG("Pin HIGH");
  digitalWrite(A5, HIGH);
  delay(5000);
  DBG("Pin LOW");
  digitalWrite(A5, LOW);
  delay(1300);
  digitalWrite(A5, HIGH);
  DBG("Pin HIGH");
  // pinMode(20, OUTPUT);
  // digitalWrite(20, HIGH);
  // !!!!!!!!!!!

  DBG("Wait...");
  delay(6000);

  // Set GSM module baud rate
  TinyGsmAutoBaud(SerialAT, GSM_AUTOBAUD_MIN, GSM_AUTOBAUD_MAX);
  // SerialAT.begin(9600);
    SerialAT.begin(115200, SERIAL_8N1, 26, 27);

  // Set up the GSM module
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);

  digitalWrite(4, LOW);
  digitalWrite(5, HIGH);

  delay(2000);
}

void loop() {
  
  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  DBG("Initializing modem...");
  if (!modem.restart()) {
    // if (!modem.init()) {
    DBG("Failed to restart modem, delaying 10s and retrying");
    // restart autobaud in case GSM just rebooted
    // TinyGsmAutoBaud(SerialAT, GSM_AUTOBAUD_MIN, GSM_AUTOBAUD_MAX);
    return;
  }
...

### Scenario, steps to reproduce
I have an application that was working fine sending data in 2G (gprs) using module Lyligo.
After weeks, the board stops sending data to web server, I troubleshoot following way:
- Replace SIM card
- Test local gprs connect with same apn --> ok
- Replace Lyligo entire board including antenna
- Reload the TinyGSM library

During all troubleshooting, it happens rarely that modem initialise but cannot connect to web client. So I guessed it was hardware but finally the complete hardware is replaced, and the very rare success during initialisation let me think that the code is ok.

So I load your troubleshooting code like above and it's the same situation like you see here under. Same issue with different SIM card and Lyligo board. Connected in USB.

Do you have any advices about my next step for troubleshooting this, or at least some command or function with your library to test deeper why is not initialising the modem?

Thanks in advance.

### Expected result
I would expect an initialisation of the modem, and then a connection to apn and WebServer...

### Actual result
the "if (!modem.restart())" is always verified, means that modem cannot initialise.

### Debug and AT command log
10:33:15.289 -> [52372] Trying baud rate 14400 ...
10:33:25.320 -> [62382] Trying baud rate 28800 ...
10:33:37.318 -> [74392] Initializing modem...
10:33:47.513 -> [84592] Failed to restart modem, delaying 10s and retrying
10:33:47.513 -> [84592] Initializing modem...
10:33:57.738 -> [94792] Failed to restart modem, delaying 10s and retrying
10:33:57.738 -> [94792] Initializing modem...
10:34:07.922 -> [104992] Failed to restart modem, delaying 10s and retrying
10:34:07.922 -> [104992] Initializing modem...

SRGDamia1, 2021-02-03:
First try setting your baud rate (ie, 9600) rather than attempting to auto-baud.  First, use the `setBaud(rate)` command to set and save the baud rate for the modem serial communication.  Then in further programs, just use ATSerial.begin(theSameRate).  What you have written now first tries a whole bunch of baud rates (`TinyGsmAutoBaud(SerialAT, GSM_AUTOBAUD_MIN, GSM_AUTOBAUD_MAX);`) and then just two lines later you set the baud to 115200.  The SIM800 can auto-baud, but it's not as stable and you'd do much better to not attempt it in any production code.  The SIM800's default baud rate is 9600, but it can go as high as 460800.

You can also try adding a short yield in commands if they're ending up chopped up (`#define TINY_GSM_YIELD() { delay(2); }`).  That's sometimes needed if you're using a slow baud rate (like the default 9600 for the SIM800) with a fast processor (like the ESP32).

maxime400, 2021-02-04:
Thanks a lot for your quick responsenet the information I learned from you. I changed following your request but it doesn't solved the issue.
Afterward, I found the origin and it was my mistake ! The 1st SIM was empty of data (due to time limit from provider), and the second was blocked due to wrong manipulation from my side. 
Thanks a lot for your support.

Maxime