vshymanskyy/TinyGSM #496: Orange network error

guideon, 2021-02-23:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: 
Main processor board: 
TinyGSM version: 
Code: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### Debug and AT command log

guideon, 2021-02-23:
Hi you can maybe help this is what I have when using a ttgo t-call with a French sim card, looks like the Orange network does not like the board. Please help. 
AT+GSN

860389052454366

OK
[21758] IMEI: 860389052454366
AT+CPIN="0000"

OK
AT+CMGF=1

+CME ERROR: SIM busy
AT+CSCS="GSM"

+CME ERROR: SIM busy
AT+CMGS="+33648550214"

+CMS ERROR:[21878] SMS: Erreur
 invalid input value
[21893] ### Unhandled: invalid input value

+CPIN: READY
[22067] ### Unhandled: +CPIN: READY

Call Ready
[24658] ### Unhandled: Call Ready

SMS Ready
[25995] ### Unhandled: SMS Ready

+CTZV: +4,0
[38269] ### Network time zone updated.

*PSUTTZ: 2021,2,23,14,52,10,"+4",0
[38275] ### Network time and time zone updated.

DST: 0
[38276] ### Daylight savings time state updated.

+CIEV: 10,"20801","Orange F","Orange F", 0, 0
[38283] ### Unhandled: +CIEV: 10,"20801","Orange F","Orange F", 0, 0

SRGDamia1, 2021-02-24:
Looks like you didn't wait long enough for your sim/module to boot.