vshymanskyy/TinyGSM #498: The modem.localIP() request crashes the ESP32-CAM Al-Thinker with stand alone SIM800L and LilyGo TTGO T-Call ESP32 SIM800L

woodlist, 2021-02-28:
Here the dump of TTGO T-Call crash:
23:39:46.537 -> Initializing modem...
23:39:50.179 -> [6657] ### TinyGSM Version: 0.10.9
23:39:50.179 -> [6658] ### TinyGSM Compiled Module:  TinyGsmClientSIM800
23:39:50.488 -> [6972] ### Modem: SIMCOM SIM800L
23:39:50.488 -> [6973] ### Modem: SIMCOM SIM800L
23:39:51.508 -> 
23:39:51.508 -> Connecting to APN: internet.beeline.am
23:40:13.585 -> [30073] ### Network time and time zone updated.
23:40:13.585 -> [30074] ### Daylight savings time state updated.
23:40:13.890 -> Waiting for network... success
23:40:13.890 -> Network connected
23:40:13.957 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
23:40:13.957 -> Core 1 register dump:
23:40:13.957 -> PC      : 0x400014e8  PS      : 0x00060130  A0      : 0x800d3978  A1      : 0x3ffb1f00  
23:40:13.957 -> A2      : 0x99498242  A3      : 0x99498240  A4      : 0x000000ff  A5      : 0x0000ff00  
23:40:13.957 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1ed0  
23:40:13.957 -> A10     : 0x3ffb1f80  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x0000ff00  
23:40:13.957 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
23:40:13.991 -> EXCVADDR: 0x99498240  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
23:40:13.991 -> 
23:40:13.991 -> Backtrace: 0x400014e8:0x3ffb1f00 0x400d3975:0x3ffb1f10 0x400d3a01:0x3ffb1f30 0x400d297a:0x3ffb1f50 0x400d44a3:0x3ffb1fb0 0x400889d9:0x3ffb1fd0
23:40:13.991 -> 
23:40:13.991 -> Rebooting...
23:40:14.025 -> ets Jun  8 2016 00:22:57

woodlist, 2021-02-28:
The program code for TTGO T-Call:
// Your GPRS credentials (leave empty, if not needed)
const char apn[]      = "internet.beeline.am"; // APN use https://wiki.apnchanger.org
const char gprsUser[] = "internet"; // GPRS User
const char gprsPass[] = "internet"; // GPRS Password
// SIM card PIN (leave empty, if not defined)
const char simPIN[]   = "0000";

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22

// Set serial for debug console (to Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to SIM800 module)
#define SerialAT Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

#include <TinyGsmClient.h>
#include <Wire.h>

// Setup a oneWire instance to communicate with any OneWire devices
//OneWire oneWire(ONE_WIRE_BUS);

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif
// TinyGSM Client for Internet connection
TinyGsmClient client(modem);
#define IP5306_ADDR          0x75
#define IP5306_REG_SYS_CTL0  0x00
// I2C for SIM800 (to keep it running when powered from battery)
TwoWire I2CPower = TwoWire(0);
bool setPowerBoostKeepOn(int en) {
  I2CPower.beginTransmission(IP5306_ADDR);
  I2CPower.write(IP5306_REG_SYS_CTL0);
  if (en) {
    I2CPower.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
  } else {
    I2CPower.write(0x35); // 0x37 is default reg value
  }
  return I2CPower.endTransmission() == 0;
}
void setup() {
  // put your setup code here, to run once:
  SerialMon.begin(115200);
  // Start I2C communication
  I2CPower.begin(I2C_SDA, I2C_SCL, 400000);
  // Keep power when running from battery
  bool isOk = setPowerBoostKeepOn(1);
  SerialMon.println(String("IP5306 KeepOn ") + (isOk ? "OK" : "FAIL"));

  // Set modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);
  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);

  // Set GSM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
  delay(3000);

  // Restart SIM800 module, it takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  modem.restart();
  SerialMon.println();
  // use modem.init() if you don't need the complete restart

  // Unlock your SIM card with a PIN if needed
  if (strlen(simPIN) && modem.getSimStatus() != 3 ) {
    modem.simUnlock(simPIN);
  }

  SerialMon.print("Connecting to APN: ");
  SerialMon.println(apn);
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    SerialMon.println(" fail");
  }
  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork()) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" success");

  if (modem.isNetworkConnected()) {
    SerialMon.println("Network connected");
  }

  SerialMon.println("Local IP:" + modem.localIP());

  SerialMon.println("Signal quality:" + modem.getSignalQuality());
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1);
}

SRGDamia1, 2021-03-01:
Um, I'm pretty sure `SerialMon.println("Local IP:" + modem.localIP());` is not valid code.  You can't add a string and an IPAddress  object like that.  Use the `getLocalIP()` function which actually returns a string.

closed 2021-03-01 by SRGDamia1

woodlist, 2021-03-01:
Please, write the exact line, which will reveal the IP. By the way, It is not well manner, when you are closing the subject, having not got the acknowledgement from issue opener.

woodlist, 2021-03-01:
I do confirmation that the "String IP = modem.getLocalIP();" works fine. Now, the issue is really solved.