vshymanskyy/TinyGSM #500: I can't communicate XBee 3 Cellular with ESP32

wavewolf, 2021-03-04:
Hello everyone, I hope you are having a good day.

I recently purchased the "XBee 3 Cellular LTE-M/NB-IoT" module and wanted to communicate with it through my ESP32 module and failed to establish communication with the "AT_Debug" test program. It never communicates.

My question is the following, is it necessary that the Xbee module must be pre-configured before in the program of the manufacturer Digi (XCTU) so that it can communicate via UART with my ESP32?

I have checked the connections (the AT_Debug program works fine with other modules of mine like the SIM800 or Lte iot click(same SARA-R410M)) and the power supply is stable, I think everything is OK.

What can it be? Do you have any success case with this module?

Thanks for your time.

SRGDamia1, 2021-03-04:
I've used this module extensively, though not with an ESP32.  Are you sure you have the correct baud rate?  The default for XBee's is 9600 8N1.

wavewolf, 2021-03-04:
> I've used this module extensively, though not with an ESP32. Are you sure you have the correct baud rate? The default for XBee's is 9600 8N1.

Hi, Sara.
I have waited for it to reach 9600 in the autobaud of the AT_debug but it fails to establish communication.
I will try to run the test in arduino to rule out that it is because of ESP32.
The only difference with arduino is that I use the "HardwareSerial SerialAT(2)" to communicate with the xbee module.
But with this same configuration, I can communicate with SIM800 and SARA R4.

So I will try to find out what might be going on.
Thanks.

SRGDamia1, 2021-03-05:
Don't even run the auto-baud.  If it's a brand-new bee it should be set to 9600, so you should just set that.  The auto-baud is really finicky and I only recommend using it for initial testing when you have no idea what the baud rate is.

Are you sure you have the right pins?  Could your bee be sleeping?  If it's set up to sleep, you need to hold the DTR/SleepRq pin LOW (9 on the bee socket) to keep it awake to communicate.  You can also check the status pin (13 on the bee socket). It should be high if the bee is awake.

wavewolf, 2021-03-05:
> Don't even run the auto-baud. If it's a brand-new bee it should be set to 9600, so you should just set that. The auto-baud is really finicky and I only recommend using it for initial testing when [...]

Yesterday I was able to establish communication (in XCTU) with the xbee cellular module through a FTDI module I had. I was looking at the module configuration in the XCTU program and everything seems to be fine by default.

I have been checking what you have told me with ESP32 and XBEE and pin 13 on the xbee is high. And pin 9(DTR/Sleep) is also high.

I have set pin 9 low and set the code with "SerialAT.begin(9600);"
To make sure, I send the "AT" and "+++" commands separately but I get no response.

this is the code I use, just in case.

`/**************************************************************
 *
 * This script tries to auto-detect the baud rate
 * and allows direct AT commands access
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 **************************************************************/

// Select your modem:
// #define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
 #define TINY_GSM_MODEM_XBEE

// Set serial for debug console (to the Serial Monitor, speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
//#define SerialAT Serial1
HardwareSerial SerialAT(2);

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3); // RX, TX

#define TINY_GSM_DEBUG SerialMon

#include <TinyGsmClient.h>

// Module baud rate
uint32_t rate = 0; // Set to 0 for Auto-Detect

void setup() {
  // Set console baud rate
  SerialMon.begin(115200);
  delay(3000);
}

void loop() {

  SerialAT.begin(9600);

  // Access AT commands from Serial Monitor
  SerialMon.println(F("***********************************************************"));
  SerialMon.println(F(" You can now send AT commands"));
  SerialMon.println(F(" Enter \"AT\" (without quotes), and you should see \"OK\""));
  SerialMon.println(F(" If it doesn't work, select \"Both NL & CR\" in Serial Monitor"));
  SerialMon.println(F("***********************************************************"));

  while(true) {
    if (SerialAT.available()) {
      SerialMon.write(SerialAT.read());
    }
    if (SerialMon.available()) {
      SerialAT.write(SerialMon.read());
    }
    delay(0);
  }
}`

SRGDamia1, 2021-03-08:
To get into "command mode" where AT commands are accepted you have to hit the "+++" *really* (impossibly) fast and there cannot be a new line or carriage return after or before the "+++".  Some serial monitors send the CR/NL every time, which will screw it up, and at the default setting I can almost never manage to type the pluses fast enough.  Move the AT begin to the setup and try having your program send the pluses for you at the start; after `SerialAT.begin(9600);` add `delay(1000); SerialAT.print("+++"); while (SerialAT.available()) {SerialMon.write(SerialAT.read());}`. 

Have you tried any of the example programs other than this one?

wavewolf, 2021-03-08:
Hi Sara.
I think we can consider this issue solved.
I have followed your indications and I have tested the http example to see if the modules (ESP32 and Xbee) were really communicating and it worked by simply setting the AT begin to 9600.
Here is the log:

```
Wait...
Initializing modem...
[8074] ### Modem:  Digi XBee3 Cellular LTE-M
[10339] ### TinyGSM Version: 0.10.9
[10339] ### TinyGSM Compiled Module:  TinyGsmClientXBee
[14453] ### Modem:  Digi XBee
Modem Info: B08
Waiting for network...
```

I have also done what you told me about using the "+++" command through the programmed and it worked, I received an "OK" through the serial monitor.

So we could say that everything has always worked, except that I wanted to activate the "command mode" and there are some complications to do that, but it also works.

Thank you for everything.

SRGDamia1, 2021-03-08:
Great!  I'm glad it's working!  Please close the issue.

closed 2021-03-08 by wavewolf