vshymanskyy/TinyGSM #504: Getting the

FStefanni, 2021-03-08:
[ x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x ] Question or request for help

### What are you working with?

Modem: sim800l, sim7000g, ublox sara u201 
Main processor board: esp32 
TinyGSM version: 0.10.9 
Code: 

### Question

Hi,

I am trying to read some communication stats.
I have seen the `getSignalQuality()` method (and related issue), and it is fine for me.
I am only missing the kind of connection band now (lte, gprs, 2g, 3g, etc.).
Is there a way to get this information by calling a pre-defined method?
Or, if I would like to implement one by myself, do you have any hint?
Or could this be a missing feature, worth to be implemented directly inside TynyGSM?

Thank you for the support,
regards

SRGDamia1, 2021-03-08:
Yes, I think all three of the modules you listed can report their current connection type.  In the case of the SIM7000, you can already get that using the `getNetworkModes()` function for GSM vs LTE and the `getPreferredModes()` for LTE-M vs NB-IoT.   For the "standard" u-blox (inc U201) those functions don't exist, but I think you could copy them from the SIM7000 implementation, change to the appropriate AT commands from the u-blox manual (AT+URAT, I believe), and submit a PR once you've tested.  For the SIM800, I can tell you the answer will always be "GPRS" because that's all it supports.

SRGDamia1, 2021-03-08:
Also, for what it's worth, GPRS != 2G and LTE isn't the same 4G though people use the terms like they are.  The  modules will generally tell you what type of technology it is using (ie, GPRS, EDGE, LTE) not the technology "generation" (ie, 2G, 3G, 4G).

FStefanni, 2021-03-09:
Hi,

thank you for the quick reply.
These are good news, so in the end I am going to implement something, and I'll let you know.
Probably I'll implement something just for the SIM7000G, and I'll propose a pr.

Regards.

SRGDamia1, 2021-03-09:
It's already there for the SIM7000.

FStefanni, 2021-03-10:
Hi,

yes, I have seen, but:

* It is missing the get of current enstabilished connection technology (the current methods just uses the setting of "preferences" for the connections)
* The two "get" methods just return the "help" for the preference options, whilst I want thir current settings (the code uses the syntax "AT+<CMD>=?" instead of "AT+<CMD>?"). I do not know why someone could want to get the "help" instead of the actual setting, but I am going to let these methods to stay there, for backward compatibility.

Regards

PR #506 (2021-03-10): SIM7000: added methods to get some connection preferences
SIM7000: added methods to get some connection preferences, plus the current connection technology.
Fixes #504
closed 2021-03-10 by SRGDamia1