vshymanskyy/TinyGSM #509: Access SIM7600 SD card

Satzende, 2021-03-16:
- [x] I have read the Troubleshooting section of the ReadMe

- [x] Question or request for help

### What are you working with?

Modem: [SIMCOM SIM7600G-H](https://docs.m5stack.com/#/en/module/comx_lte?id=easyloader)

Main processor board: 

[ESP32-D0WDQ6-V3 | 240MHz dual core, 600 DMIPS, 520KB SRAM, Wi-Fi, dual mode Bluetooth](https://docs.m5stack.com/#/en/core/core2)

TinyGSM version: 0.10.9

Code: 

```

#define KILL
#ifdef KILL
//#include <SD.h>
//#include <M5Stack.h>
#include <M5Core2.h>

#define TINY_GSM_MODEM_SIM7600
#define SerialMon Serial
#define SerialAT Serial2
#define TINY_GSM_DEBUG SerialMon
#define GSM_PIN "5938"

#include <TinyGsmClient.h>
#define GSM_AUTOBAUD_MIN 115200
#define GSM_AUTOBAUD_MAX 115200

#if TINY_GSM_TEST_GPRS && not defined TINY_GSM_MODEM_HAS_GPRS
#undef TINY_GSM_TEST_GPRS
#undef TINY_GSM_TEST_WIFI
#define TINY_GSM_TEST_GPRS false
#define TINY_GSM_TEST_WIFI true
#endif
#if TINY_GSM_TEST_WIFI && not defined TINY_GSM_MODEM_HAS_WIFI
#undef TINY_GSM_USE_GPRS
#undef TINY_GSM_USE_WIFI
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
#endif

#define DUMP_AT_COMMANDS
#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

const int chipSelect = 4;

void init_sd(){
  //SD Card ini
  M5.Lcd.println(F("Initializing SD card..."));
  //SPI();
  if (!SD.begin(chipSelect))
  {
    M5.Lcd.fillScreen(RED);
    M5.Lcd.println(F(">>>>SD card ERROR"));
    M5.Lcd.println(F("Programm STOP!"));
    
    while (true);
  }

  File root = SD.open("/");
  if (!root || !root.isDirectory())
  {
    M5.Lcd.fillScreen(RED);
    M5.Lcd.println(F("Failed to open directory"));
    M5.Lcd.println(F("Programm STOP!"));
    
    while (true)
      ;
  }
  root.close();  
  M5.Lcd.println(F("DONE"));
  
}

// debug zweck

#define MY_LTE
//#define MY_SD

////////////////////////////////////////////////////////////////
void setup()
{
  M5.begin(1,1,1,1);
  M5.Lcd.fillScreen(WHITE);
  SerialMon.println("tester");
  M5.Lcd.setCursor(0, 10);
  M5.Lcd.setTextColor(BLACK);
  M5.Lcd.setTextSize(2);
  

  #ifdef MY_SD
  init_sd();
  #endif  
  
  #ifdef MY_LTE
    SerialMon.println("Init Modem");
  pinMode(5, OUTPUT);
  digitalWrite(5, HIGH);
  delay(5000);
  digitalWrite(5, LOW);
  delay(1300);
  digitalWrite(5, HIGH);

  delay(6000);

  TinyGsmAutoBaud(SerialAT, GSM_AUTOBAUD_MIN, GSM_AUTOBAUD_MAX);
  //M5.Lcd.println(F("Initializing modem..."));

  if (!modem.restart())
  {
    M5.Lcd.fillScreen(RED);
    SerialMon.println(F("Failed to Start Modem"));
    SerialMon.println(F("Programm STOP!"));
    
    while(true);
  }
  SerialMon.println("Done!");
  //LTE init ende
  #endif 
}

void loop()
{
    M5.update();
  if(M5.BtnA.wasReleased()){
    
    SerialMon.println(">>>>>>>>A<<<<<<<<");
    modem.sendAT("\r\n");
    modem.waitResponse();
    
    
  }

  if(M5.BtnC.wasReleased()){
    SerialMon.println(">>>>>>>>C<<<<<<<<");
    
    
    modem.sendAT(F("+FSCD?\r\n"));//wo bin ich
    modem.waitResponse();
    
    modem.sendAT(F("+FSCD=D:\r\n"));//geh zu D
    modem.waitResponse();

    modem.sendAT(F("+FSCD?\r\n"));//wo bin ich
    modem.waitResponse();
    
  }

}   

#endif
```

### Scenario, steps to reproduce
Edit -> TinyGsmClientSIM7600.h And another file. . .
I do not know exactly why. But certainly has nothing to do with the problem
[src.zip](https://github.com/vshymanskyy/TinyGSM/files/6149513/src.zip)

### Expected result
On the SD card is a soundfile, which is to be played the modem someone calls. The modem can not access the SD card.

### Actual result

### Debug and AT command log
```

AT+FSCD?

AT+FSCD?
+FSCD: C:/

OK
AT+FSCD=D:

AT+FSCD=D:
ERROR
AT+FSCD?

AT+FSCD?
+FSCD: C:/

OK
```

Satzende, 2021-03-16:
[I also asked in another forum 
](https://forum.arduino.cc/index.php?topic=732412.0)

Satzende, 2021-03-17:
Lösung : ich bin dumm.
Das LTE Modul hat keine Verbindung zur SD Karte, kann also auch nicht  verbunden werden.
https://static-cdn.m5stack.com/image/product_jpg/assets/img/product_pics/module/com.x_lte/com.x_lte_sch.jpg

closed 2021-03-17 by Satzende

miraconda, 2024-11-15:
Did u solve this?