vshymanskyy/TinyGSM #518: How to send simple AT commands using the library

krupis, 2021-04-09:
Hello. I am using this library to program lilygo ESP32 GSM development board. It works fine but I want more flexibility and I want to control what AT commands I want to send and what commands I do not need to send. Can someone explain to me how am I suppose to use sendAT() method correctly?
Few questions:

1. What is GF meaning when calling sendAT() function for example:

 modem.sendAT(GF("AT"));

2. How can I see the response from the command sendAT() command. 

For example in my loop I call these methods

void loop()
```
{
  String res;
  modem.sendAT(GF("AT"));
  modem.waitResponse();
  delay(3000);
}
```

And the serial monitor returns :
[image]

I simply want to know how can I write my own AT commands and retrieve the response

YogPanjarale, 2021-04-21:
You can try to read directly from `SerialAT` or `SerialMon` which was used to initialize the `modem`
example 
```c++
while(SerialAT.available())
  Serial.write(SerialAT.read());
}
```

SRGDamia1, 2021-04-22:
Use `sendAT` with `waitResponse`, just remember that the `sendAT` command already includes the "AT".  So `sendAT(GF("AT"))` actually sends "ATAT".  If you want to check signal/CSQ, you'd only include the "+CSQ" bit, not the "AT+CSQ":

```c++
{
  modem.sendAT(GF("+CSQ"));
  modem.waitResponse();
  delay(3000);
}
```

HazemBenAmmar, 2021-11-19:
Hi, and how can i get the response message for exemple i want to read a message:
modem.sendAT("+CMGR=10"); 
 thank you.

SRGDamia1, 2021-11-24:
@HazemBenAmmar - use the `waitResponse(...)` function.

If you're looking to keep the text returned, you can feed a string to wait response and it will populate it:
```cpp
String res1;
modem.sendAT("+CMGR=10");
modem.waitResponse(1000L, res1);
```