vshymanskyy/TinyGSM #524: Blocking function or infinite loop

brunocadet, 2021-04-23:
Hi,

I found this issue :
When testing .isNetworkConnected()  TinyGsm enters in an infinite loop if 'false'.

Thanks for help.

SRGDamia1, 2021-04-28:
It shouldn't.  There's no looping at all in that function.  If you're looking at inside of `waitForNetwork()` where there is a loop, there's a built in default of 60 seconds.  Awfully long, but not infinite.  You can shorten it to whatever you want by using `waitForNetwork(timeout_ms)`.

Otherwise, post the full AT dump showing the problem.

roysG, 2021-12-06:
Is there any replace method for waitForNetwork that no blocking the code, 
or another suggested way to check if the connection is backed, without blocking the loop?

SRGDamia1, 2021-12-06:
@roysG I'd take a pull request if you have a solution.