vshymanskyy/TinyGSM #527: Problems about return the GPRS coordinates

Hammerschmidt, 2021-04-26:
[X] I have read the Troubleshooting section of the ReadMe
[ ] Request to support a new module
[ ] Bug or problem compiling the library
[X] Bug or issue with library functionality (ie, sending data over TCP/IP)
[X] Question or request for help

### What are you working with?

Main processor board: ESP32 Dev Board
GSM Modem: SIM800L
TinyGSM version: 0.10.9

Hello, i have a doubt about "getGsmLocation()".

### Scenario, steps to reproduce
I placed in my code the following behavior.
`Serial.println(modem.getGsmLocation());`

### Expected result
I wanted to return of my chip localization.

### Actual result
I retuned the following coordinate: 49°15'06.0"S 25°25'08.0"W;
P.S.: My network and my GPRS connection are working.

Someone can help me about it?

SRGDamia1, 2021-04-28:
Are you in the middle of the South Atlantic?

Assuming you're not, did you make sure you are connected to the network before asking for your location?  Do you have decent signal quality?  I'm not sure I've ever gotten back bogus coordinates from it; usually just no coordinates at all if it doesn't have a good lock from towers.  Did you look at the whole AT command log to check if the raw response is ok, but is being parsed incorrectly?

Hammerschmidt, 2021-04-28:
I need to see the AT logs, but I see my signal quality returned using the command "getSignalQuality()" retuns for me 17 (I don't know if it is a good or bad quallity), and my things in MQTT I recive very well too. But I need to found what is happening in AT commands.
P.S. : I'm not in South Pacific Ocean.

SRGDamia1, 2021-04-28:
CSQ of 17 should be plenty.

Hammerschmidt, 2021-05-05:
I need to see better what happened here, I will close the topic for now.

Hammerschmidt, 2021-05-05:
I need to see better what happened here, I will close the topic for now, thanks.