vshymanskyy/TinyGSM #528: Cannot Connect to the BLYNK Server!!!

Manu-dewa, 2021-04-27:
Hello

I am using an esp32 with a SIM800C module and wired hardware serial lines according to the TTGO T-CALL board.

i am using example blynk code provided by this link - https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_BlynkClient

All the AT commands working without an issue.

The issues rises when i tried use the example blynk code (in the above link). i get an error. 
(Unhandled : CIPSTATUS) and repeates that error over and over.

Below link houses the error i am getting and it is screen capture of the my serial monitor

i am using tinygsm version - 0.10.9
                  blynk version - 0.60
                  arduino version - 1.8.12

[image]

SRGDamia1, 2021-04-28:
It looks like it's probably chopping up the `+CIPSTATUS` response into two responses of `+` and `CIPSTATUS` and then not recognizing it as the proper response.  Try adding a very small "yield" (delay) for TinyGSM:  add `#define TINY_GSM_YIELD_MS 2` to the top of your program.  That sometimes helps (or is needed) when you have a relatively fast processor (like the ESP32), especially if you're using the SIM800's default 9600 baud.

Manu-dewa, 2021-12-06:
@SRGDamia1 sorry that didnt work still the same issue