vshymanskyy/TinyGSM #531: Not success to get the request, even after successfully connection to internet

roysG, 2021-04-28:
Wait...
Initializing modem...
[12453] ### TinyGSM Version: 0.10.9
[12454] ### TinyGSM Compiled Module:  TinyGsmClientSIM7000
[12656] ### Unhandled: AR
jH
[12789] ### Modem: SIMCOM SIM7000G
Modem Info: SIM7000G R1529
[13828] ### Unhandled: 
[14834] ### Daylight savings time state updated.
[14840] ### Network time and time zone updated.
[15846] ### Unhandled: OK
Waiting for network...[16847] ### Unhandled: 
 success
Network connected
Connecting to  success
GPRS connected
Performing HTTP GET request... [18295] ### Daylight savings time state updated.
[19276] ### Unhandled: 
[21918] ### Unhandled: 
[23011] ### Unhandled: 
[24013] ### Unhandled: 
[28015] ### Unhandled: 
[29017] ### Unhandled:

roysG, 2021-04-28:
Code:

`/**************************************************************
 *
 * This sketch connects to a website and downloads a page.
 * It can be used to perform HTTP/RESTful API calls.
 *
 * For this example, you need to install ArduinoHttpClient library:
 *   https://github.com/arduino-libraries/ArduinoHttpClient
 *   or from http://librarymanager/all#ArduinoHttpClient
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 * For more HTTP API examples, see ArduinoHttpClient library
 *
 * NOTE: This example may NOT work with the XBee because the
 * HttpClient library does not empty to serial buffer fast enough
 * and the buffer overflow causes the HttpClient library to stall.
 * Boards with faster processors may work, 8MHz boards will not.
 **************************************************************/

// Select your modem:
//#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
 #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
//#define SerialAT Serial1

// or Software Serial on Uno, Nano
#include <SoftwareSerial.h>
SoftwareSerial SerialAT(10, 9); // RX, TX

// Increase RX buffer to capture the entire response
// Chips without internal buffering (A6/A7, ESP8266, M590)
// need enough space in the buffer for the entire response
// else data will be lost (and the http library will fail).
#if !defined(TINY_GSM_RX_BUFFER)
#define TINY_GSM_RX_BUFFER 650
#endif

// See all AT commands, if wanted
// #define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon
// #define LOGGING  // <- Logging is for the HTTP library

// Range to attempt to autobaud
[... 168 lines cut ...]
  SerialMon.println(F("Response:"));
  SerialMon.println(body);

  SerialMon.print(F("Body length is: "));
  SerialMon.println(body.length());

  // Shutdown

  http.stop();
  SerialMon.println(F("Server disconnected"));

#if TINY_GSM_USE_WIFI
    modem.networkDisconnect();
    SerialMon.println(F("WiFi disconnected"));
#endif
#if TINY_GSM_USE_GPRS
    modem.gprsDisconnect();
    SerialMon.println(F("GPRS disconnected"));
#endif

  // Do nothing forevermore
  while (true) {
    delay(1000);
  }
}`

roysG, 2021-04-28:
One more test result:

OK

+CFUN: 1

+CPIN: READY

SMS Ready
[13114] ### Modem: SIMCom SIM7000
[13240] ### Daylight savings time state updated.
[13246] ### Network time and time zone updated.
Modem Info: SIM7000G R1529
Waiting for network... success
Network connected
Connecting to [83170] ### Unhandled: SdUz-5
[84179] ### Unhandled: OK
 success
GPRS connected
Connecting to vsh.pp.ua
[85313] ### Unhandled: S*I=Izr
5
 success
Performing HTTP GET request...
[86688] ### Unhandled: ?

SRGDamia1, 2021-04-28:
This really looks like a baud rate/serial port accuracy error.  Try to use hardware serial if at all possible.  If you absolutely cannot use hardware serial, I strongly recommend using AltSoftSerial over SoftwareSerial for supported boards.