vshymanskyy/TinyGSM #532: PyCom GPy or FiPy with TinyGSM

ClemensGruber, 2021-04-29:
Request to support a new board **PyCom GPy** and/or **FiPy**.

The PyCom boards are quite compact ESP32 boards, a bit more expensive but in a nice form factor and not so power hungry in deep sleep as other boards. The FiPy has also an LoRa module onbaord so one device would fit for different use cases or areas. We had used the FiPy in an academic project and had a lot of trouble with the MicroPython firmware and libs. Ongoing bugs, crashes, less support, lot work to get it less power hungry ... so MicroPython and the PyCom software environment is not the way to go on. 

Both board working with the Sequans Monarch SQN3330 module, see https://alepycom.gitbooks.io/pycom-documentation/content/chapter/tutorials/lte/firmware.html

The module supports CAT-M1 oder CAT-NB1 (aka NB IoT) only! To change from M1 to NB1 you have to change the firmware of the modem!! So not a easy usable solution. 

There are some requests for runnig with TinyGSM under https://forum.pycom.io/topic/3134/using-pycom-boards-with-arduino-ide

On the TinyGSM main page here on github the Sequans Monarch is listed as 
> Sequans Monarch LTE Cat M1/NB1 (VZM20Q)

I don't know how different the VZM20Q in camparison to the SQN3330 is. Perhaps the AT base is similar or equal. 

Has someone tried a PyCom device with TinyGSM already or can share some code that is perhaps working in parts?

SRGDamia1, 2021-04-29:
Sequans website is pretty unhelpful in finding the AT commands (or even the modules) but the AT command manual that PyCom provides for the GPy and FiPy is the exact same document that I had for the VZM20Q.  I'm guessing that means all the Monarch modules use the same AT commands and you should be able to use what's already written without needing any changes.

pottendo, 2021-08-19:
hi, you may have a look here: https://github.com/vshymanskyy/TinyGSM/issues/579
bye, pottendo